Ralim/IronOS #1455: How to create own multilangual IronOS firmware?

Nachtara97, 2022-10-30:
Hey there,

i own a Pinecil v1 and want to create a "own" multilingual IronOS firmware. But i can't figure out how to do that. I know that multilangual firmware releases are exist, but i don't want all these languages, i only need two of them (EN and DE). (maybe i can also automate it when a new release was released)

Can anyone please tell me howm to do that? Thanks in advance!

discip, 2022-10-30:
@Nachtara97
Please see here: [Development.md](https://github.com/Ralim/IronOS/blob/dev/Documentation/Development.md)

>To build a custom compressed multi-language firmware including German, Spanish, and French for the TS100 (note if `model` is unspecified, it will default to `TS100`):
>
>```
>make -j8 custom_multi_langs="DE ES FR" firmware-multi_compressed_Custom
>```

discip, 2022-10-30:
@Nachtara97
Please see here: [Development.md](https://github.com/Ralim/IronOS/blob/dev/Documentation/Development.md)

>To build a custom compressed multi-language firmware including German, Spanish, and French for the TS100 (note if `model` is unspecified, it will default to `TS100`):
>
>```
>make -j8 custom_multi_langs="DE ES FR" firmware-multi_compressed_Custom
>```

In your case:
```
make -j8 model=Pinecil custom_multi_langs="DE EN" firmware-multi_compressed_Custom
```

Nachtara97, 2022-11-06:
> @Nachtara97 Please see here: [Development.md](https://github.com/Ralim/IronOS/blob/dev/Documentation/Development.md)
>
>
> In your case:
>
> ```
> make -j8 model=Pinecil custom_multi_langs="DE EN" [...]

Instructions are macOS only

Ralim, 2022-11-06:
That means you did not do a recursive git clone, as you are missing the submodules

discip, 2023-01-22:
@Ralim
This can be closed since #1520 is implemented.

closed 2023-01-23 by Ralim