Ralim/IronOS #1456: need help sourcing an Inductor 6.8uH Pinecil

River-b, 2022-10-30:
Both the  pinecil V1 and V2  use  an Inductor for L1  but I can not seem to find this part anywhere.  it's brown with 6R8 on it.
Appears to be  6.8uH.  do you have any leads on this, name of manufacturer?   I looked on LCSC.com,  for 6.8uH, but can not find anything that is also  3.3A and also 0.04ohm  or that is 4x4 size (I assume they mean 4x4mm  from the schematic 4x4)

[image]

from the schematic 
[image]

AdamKGoddard, 2022-12-09:
MPS MPL-SE4030-6R8 closest I could find. 4x4x3mm, 6.8uH, 3.3A Isat, 0.083ohm

River-b, 2022-12-09:
> 

thanks do you know which website you found it or a link ?

AdamKGoddard, 2022-12-12:
https://au.mouser.com/ProductDetail/Monolithic-Power-Systems-MPS/MPL-SE4030-6R8?qs=sPbYRqrBIVn8IOJ%252B8Jqo%252Bg%3D%3D

https://au.element14.com/monolithic-power-systems-mps/mpl-se4030-6r8/power-inductor-6-8uh-semishield/dp/3501880

River-b, 2022-12-12:
thanks.  this is better than nothing.  it might be something they have in Shenzhen but  not sold/listed online. it's the same with the DC barrel, I don't see it sold anywhere.

closed 2022-12-12 by River-b