Ralim/IronOS #1460: Pinecil v1 with cygnett powerbank

Avalonnw, 2022-10-31:
Hi,
I am experiencing some sort of a bug/weird behavior.
I have a Cygnett 19200mah Chargeup Advance powerbank (can't find it on their website though) that is supposedly 100w PD output. It can charge my laptop, phones, etc.
I also have a Pinecil v1 that runs off a wall PD 65w charger and works very well. When i try to pair those two, however, Pinecil reports 5v on 2.17 firmware and 16v on 2.19 and just plain resets when I start it to heat up. Same cable as usual, just power source is different. Is the problem with firmware, or powerbank, or somewhere in between? I tried different cables.

Avalonnw, 2022-11-01:
Sorry if I was not clear enough.
v.2.17 with wall brick - works at 20v
v.2.17 with powerbank - reports 5v
v.2.19 with wall brick - works at 20v
v.2.19 with powerbank - reports 16v and restarts at the press of "+" button
In all tests 2 cables were used with the same results.
Powerbank is fully charged and is 1 month old. It charged my 65w laptop with no issues.

Avalonnw, 2022-11-01:
Thank you. I'll definitely consider Anker, however I was under impression that Cygnett is a reputable brand also. On the back of the bank it has a list of the voltages it's supposed to provide and 20v@5A is amongst them. It charges my phone at full speed which suggests it goes into 20v mode. Unfortunately I do not have a debugger or a monitoring device to measure what's going on :( Maybe it's time to invest in one...

SolderEater, 2022-11-02:
I wonder if it could be negotiating a PPS mode at 16V, and then failing to  use it.  What does the  Pinecil PD debug menu show when you plug it in with the + button held down, then hit the + button to cycle through the supported power modes detected?

Avalonnw, 2022-11-02:
PD debug state 12 VBus
1   5v  3.0A
2   9v  3.0A
3  12v 3.0A
4  15v 3.0A
5  20v 3.0A
6  3-11v  3.0A
7  3-21v  3.0A

Ralim, 2022-11-03:
The fact that they are all capped at 3A shows you are not using a 5A rated cable.

However it should still negotiate fine from that enumeration 16V is a little odd for a value to negotiate for. It would have to be using PPS to reach that.

Any chance you could try the 2.20 beta builds from actions tab to check if your hitting the regression I found in the PD library?

If holding it on that menu it is stable, it makes it likely that PD is negotiating fine and its the charger end that's being odd. (Of course it could be a code bug, I'm sure there are more of them. But that's harder to prove without a tester).

Avalonnw, 2022-11-04:
2.2 is running, same result:
https://youtu.be/FekBI_Td-GE

Ralim, 2022-11-04:
To confirm, I assume you mean 2.20?

Have you checked the debug menu uptime count to make sure its actually resetting on load?

Im not super sure why its stuck at 16V, makes me wonder if the power bank is getting stuck in an odd mode. Do you have a 5A cable you can test with?

Avalonnw, 2022-11-04:
2.20.1FD6C60
Uptime resets.
I don't have a 5A cable unfortunately :(
It could be a powerbank, however I wonder why it works with other power-demanding devices. It there a way to monitor and debug PD bus on a windows PC or Android phone I wonder?

River-b, 2022-11-09:
> 2.20.1FD6C60 Uptime resets. I don't have a 5A cable unfortunately :( It could be a powerbank, however I wonder why it works with other power-demanding devices. It there a way to monitor and debug PD bus on a windows PC or Android phone I wonder?

1. Highly recommend  buying a PD3.1  usb-C  240W cable, they are backwards compatible with all lower  rated USB-C chargers. 
2. 100W cables are not worth buying now imho  since they were Deprecated by USB-IF in Dec  2021.  Only  240W cables are EPR class and PD3.1  - very future proofing for all future USB-C chargers and  all lower PD65w ones too.  240W cables are about the same price as 100w ones since they both have  EPR chip ($12-$15).  Although I did see a [$6.99 silicone 100W cable ](https://a.co/caAfRnW)  on amazon  recently which was a bargain.  still recommend the 240W PD3.1 cable though to be future proof.

4. Pinecil V2 needs minimum  3+ amps to operate   so if you have  15V,  2.33 amps for example then it will drop to lower volts that is too low to work.  I look for anything that is  at least  20V PD65W  3-3.25 Amps on at least one  usb-c port on the chargers. 
5. check out Power section on [Pinecil WIKI  page here](https://wiki.pine64.org/wiki/Pinecil#Power_Supplies_and_Watts)
6. join  live community chat for live feedback/troubleshooting. links also in the wiki for chat.

discip, 2023-02-15:
@Avalonnw
Have you managed to figure out how to solve your issue?
If so, please consider closing this. 😊

Thanks

Samuelrmlink, 2023-02-16:
This is now a fixed issue. Looking at your PDOs - the 7th PDO is PPS.
This bug has been fixed. See https://github.com/Ralim/usb-pd/issues/17

The latest version has not yet adopted my changes - although they are in the dev branch. 
I've attached my patched version. This should fix the issue. @Avalonnw 

https://drive.google.com/file/d/1NZA_fwtL-ZI2E7XqOl-ZTkRs7hUOjUTI/view?usp=share_link
Had to create a Google Drive link. Github would not allow me to update a hex file. 
You can either use the following command (if you have dfu-util installed) or the Pine64 updater if you're using Windows. 
Just make sure to download and use the hex file from my Google Drive since the fix has not been added to a tagged release yet. 

dfu-util -D <filename>

River-b, 2023-02-17:
if this beta bug patch from Samuel  fixes the issue please comment here and close the ticket thanks @Avalonnw

River-b, 2023-02-17:
> Github would not allow me to update a hex file.

@Samuelrmlink  if you make the file into a ZIP, then you can attach it in here, I attach Bin files in issue tickets by making them Zip first.

Samuelrmlink, 2023-02-17:
@River-b Good point. 
I'll attach the zip here. 
Try this @Avalonnw 
[Pinecil_EN.zip](https://github.com/Ralim/IronOS/files/10762951/Pinecil_EN.zip)

Avalonnw, 2023-02-17:
Thanks! I'll give it a try tonight!

discip, 2023-02-22:
@Avalonnw
Did it finally work?

Avalonnw, 2023-02-23:
Yes, sorry, had some other issues with powerbank :( It works perfectly now!

closed 2023-02-23 by Avalonnw