Ralim/IronOS #1461: MHP30 Temperature Stability

edrose, 2022-11-01:
**Describe the bug**
Temperature stability on the MHP30 is quite poor, it feels like the control algorithm needs adjusting. The HotPlate will consistently overshoot the target temperature by around 20°C, before falling back down to around 20°C below the target temperature and then hovering within a +/- 10°C range of the target temperature. If the setting is too hot then I sometimes also get a "Thermal Runaway" error, which I guess is a result of the overshoot. 

I suspect that the thermistor in the Hot Plate is not very close to the surface, leading to a delay in the readings. Stability on the stock firmware is not perfect, but it's much better then IronOS. The stock firmware will overshoot by about 10°C before recovering and staying within +/-2°C of the target temperature.

The behaviour is slightly different when a PCB is on top, since that sucks away a lot of the heat. On the stock firmware, the temperature does not overshoot, and it manages to regulate the temperature much better. However on IronOS it still overshoots by about 10°C and takes a while to stabilise.

When placing a PCB on top, there is a ~20°C fall in temperature on IronOS. On the stock firmware the fall is about 15°C and the recovery is much quicker.

My guess from observing the two firmwares is that there are two differences between the stock firmware and IronOS:

1) The stock firmware seems to poll the thermistor at a higher rate, allowing it to react more quickly to changes in temperature

2) The stock firmware seems to turn the heating element off earlier than IronOS, better accounting for the thermal inertia in the hot plate. 

**To Reproduce**
1) Install IronOS on the MHP30 preheater
2) Set a target temperature (e.g 190°C)
3) Observe the stability of the temperature on the readout
4) Place a PCB on the hot plate
5) Observe the stability of the temperature on the readout

**Expected behavior**
I expected the thermal regulation to be on par with the stock firmware. 

**Details of your device:**
Device: MHP30
Release: 2.19.A3BBABC

**Additional context**
I'm happy to do some experimentation if someone can point me to the places in the source code where the algorithm can be tuned. I'm a Software Engineer, so am well placed to contribute if that's helpful.

Ralim, 2022-11-06:
Hello 👋🏼 ,

This is interesting as I dont see this on the two units I have here. So if you feel like doing some tuning and testing, it would be super useful.

The function `getTipThermalMass` is the main suspect to be tuned/adjusted along with the constant `TIP_THERMAL_MASS`.
If you have time to test adjusting those it would be great.

The sampling time of the thermistor and the power drive to the hotplate are related, so the more we are dumping power, the less time we spend measuring. This is why the algorithm for control toggles between a faster and a slower rate (Which you may notice if you watch a heatup carefully).

What power source are you using btw? As performance just change a little with different source's. 

The MHP30 looks to have _two_ means of measuring hotplate temp, we are using one of them. This is based on reverse engineering the firmware (no sources at all released for MHP30). Its possible their firmware is doing some fancy combo of both.

discip, 2023-02-15:
@edrose
Are you still having this issue?
If not please don't forget to close this issue. 😊

closed 2023-05-16 by discip

tomasrojasc, 2024-08-01:
Sure, here's a revised version for better legibility:

I am having a similar problem as the OP.

I flashed the latest version of IronOS, and I consistently experienced overshoots in the temperature target. This issue improves when I limit the power, but that solution doesn't work for my application. I need to adjust the temperature in steps of ~10°C, ranging from approximately 35°C to 350°C, and I require a relatively stable temperature. Overshoots are unacceptable, and limiting the power prevents me from reaching high temperatures.

Does anyone have a possible solution for this? I also tried compiling IronOS myself. Although I managed to compile it, I'm not sure I did everything correctly. I experimented with changing the `TIP_THERMAL_MASS` value by increasing and decreasing it as @Ralim suggested, but it only worsened the problem.

Any help would be greatly appreciated.

Best regards,
Tomás

Ralim, 2024-08-01:
Moved to a new issue as this one is closed