Ralim/IronOS #1462: MHP30 Accelerometer not Recognised

edrose, 2022-11-01:
**Describe the bug**
The accelerometer on my MHP30 is not recognised by IronOS. When scrolling through the debug menu, "ACC None" appears. "Move" is always 0.

The MHP30 does have an accelerometer, as the stock firmware uses it for topple detection.

**To Reproduce**
1. Load IronOS onto an MHP30
2. Press and hold button B to enter debug menu
3. Scroll through the menu. "ACC None" will be displayed and "Move" will always be 0.

**Expected behavior**
I expected IronOS to recognise the accelerometer in the MHP30.

**Details of your device:**
Device: MHP30
Release: 2.19.A3BBABC
Accelerometer: "None"

**Additional context**

Ralim, 2022-11-06:
Could you try the 2.20 beta from the builds linked from [here](https://github.com/Ralim/IronOS/pull/1416).
Just in case this an accelerometer that is already enabled.

River-b, 2022-12-09:
@edrose   if this ticket is resolved with new firmware could you close the ticket?

discip, 2023-02-06:
@edrose
Are you still having issues with this?

docgalaxyblock, 2023-02-15:
The accelerometer works fine on v2.20
*since the issuer is not responding over a week now...I guess you can close this*

discip, 2023-02-15:
Closing this issue because the OP is not responding and no one else has reported comparable behavior.

closed 2023-02-15 by discip