Ralim/IronOS #1466: USB Mass storage to import/export settings

chatelao, 2022-11-08:
**Is your feature request related to a problem? Please describe.**
- Expose the Iron as a mass storage device on USB.
- Allow up-/downloading all settings as a textfile

discip, 2023-02-15:
@chatelao
Do you own A Pinecil of any kind?
Are you going to provide a PR for this FR, if not please consider closing this issue, due to the developers statement?

discip, 2023-02-15:
@chatelao
Do you own a Pinecil of any kind?
Will you do a PR for this FR, if not please consider closing this issue reasoned by the developer's statement?

ace-dent, 2023-03-29:
+1
New user with a TS80P ~ also missing the ability to edit settings, as per the stock firmware.

River-Mochi, 2023-03-30:
> New user with a TS80P ~ also missing the ability to edit settings, as per the stock firmware.
@ace-dent 

will you be submitting the PR  for this otherwise this ticket should maybe be closed as Ralim said above, not planned to implement as there is not enough space in flash for a USB stack.

Ralim, 2023-03-30:
On the Miniware devices there is less than 1k of flash space left, we dont have any option of doing a usb stack on the Miniware irons due to this.

On Pinecilv2 we are providing this option via BLE instead as that is far easier to implement.

closed 2023-03-30 by Ralim

ace-dent, 2023-03-30:
Thanks @Ralim for the info. I'm new to the project, so didn't realise you were so space constrained...
Is there a reason Miniware can flash larger firmware to the TS80P?
- Miniware Vanilla v1.36 hex ~136KB
- IronOS v2.20 hex ~122 KB

River-Mochi, 2023-03-30:
> On Pinecilv2 we are providing this option via BLE instead as that is far easier to implement.

what is the BLE option provided for Pinecilv2  as USB storage device?  
do you mean exporting data over BLE so that BLE apps like PineSAM can pick it up?