Ralim/IronOS #1467: 2.5mm plug for continuity testing

chatelao, 2022-11-08:
**Is your feature request related to a problem? Please describe.**
- Add a continuity/resistance/etc. testing functionality

- Over a 2.5mm Jack providing a probe
- Compared to ground / vcc of the USB-C

Check the awsome GM328 concepts for possible features:
- https://github.com/kubi48/TransistorTester-documentation/blob/main/pdftex/english/ttester.pdf

River-b, 2022-11-09:
> **Is your feature request related to a problem? Please describe.**
>
> * Add a continuity/resistance/etc. testing functionality
> * Over a 2.5mm Jack providing a probe
> * Compared to ground / vcc of the [...]

1. Which soldering Iron  is this request for?  not enough details here.
2. If this is a Pine64 Pinecil Iron,  have you  tried the  $4 official Pinecil break-out board  or other low cost USB-C  break-out boards on the market with Pinecil?  see pine64.com  for  details on break-out board.
3. Are you asking for a physical hardware  2.5mm  Jack port change on the Iron itself  or firmware support? much more details needed to what is being requests.
4. If it's hardware change, I'm not sure this is doable  here since this is firmware/software  support.   That said, there are people who are doing independent hardware modifications of their  Irons and then  ask  for specific firmware to expose it.   Example is adding LED light to  Pinecil V2,  request is to connect it to SPI  so we can control the on/off in the menu.   If you have a specific  mod  you did on the iron for this jack added,  please outline how you would like the firmware to support it in more details. 
5. if you do modifications of  the  Firmware code to support it,  you could also Submit as PR  to be included in future versions of  IronOS.

Ralim, 2022-11-20:
@chatelao 
If I understand you are asking about somehow hacking up the device to act as one of the low cost transistor testers.
This is not possible at all without fundamental hardware resdesigns for nearly all test modes.
Rather than spending $20-$30 on new hardware to make this sort of thing possible; just buy a tester for about half of that, which will work far far far better and be simpler.

closed 2022-11-20 by Ralim