Ralim/IronOS #1469: Create an temp offset setting, instead of just a self calibrate.

SolderEater, 2022-11-11:
I can set a temp offset on my $15 junk digital iron.  Also on my $39 900M station.  Also on my $39 KSGER T12 clone.  I want to calibrate the Pinecil to a reference like a 191 tip temp reader. Having the Pinecil calibrating to itself is not ideal.  You don't calibrate a set of scales to itself, or a multimeter to itself.  You calibrate to a separate known good reference.  I can ask Trump or Biden if they are honest, but without an external source, I have no source of truth.  And I sent you another 5 coffee's :)  This is just a pet peeve of mine.  I bought a 191 temp tester, but it is worthless with my Pinecil, except to show how far each tip is off.  I can't correct anything.

Ralim, 2022-11-12:
So there historically was temperature calibration back when the firmware assumed a single uV/C value. But the firmware moved towards having a lookup table to improve accuracy ad mid point temperatures.

This is set up so that error at 320-350C is minimised but is generally very close (within 3C worst case) across generally the entire range.

While this improved accuracy across all of the official tips, it does mean for some tips its still out.

So the question of how you would want to perform a manual calibration across 40 odd points of the scale?

=====================

The current calibration that is done is just cold junction offset, which is there to remove hardware bias. If it's this that's wrong you will have a consistent shift across the entire measurement range.

If it's not consistent, then its the above lookup table.

=======

Your original issue is missing the required information of device model and firmware revision. Please add this information back in that was in the template :) because that information is required before any further triage

SolderEater, 2022-11-13:
V2 20.5416783
06-11-22
All the other irons I see will calibrate at a single temp, then be exact at that temp, but drift as you become further from it.  Normally I calibrate at 300, and then test 250 and 350 to see how far those are off.  I'm trying to comprehend the 40 points scale thing.  Can you just bump the entire scale by a given degrees in C?  On other irons you never do get an exact setting across a full range of temps, but you can set the temp you use the most to be exact.  I began to struggle with Pinecil when I tested a non-OEM tip and it was 34c off, even after calibration.  Then I tried another of the same, and it was right on.  Then I tried a OEM tip and it was 34 off the other way.  I'll test about 10 tips across 3 irons (I think they are all running different firmware) tonight and see if this is really even a thing.  I've mostly been using these tips https://www.aliexpress.us/item/2251832671811959.html which has packaging that looks identical to the AE tips that River uses.  It is probably a very common TS100 tip manufacturer.

discip, 2023-02-15:
@SolderEater
Please provide the information of the device you are referring to.

River-Mochi, 2023-03-23:
> @SolderEater Please provide the information of the device you are referring to.

He has a Pinecil V1 and also a Pinecil V2.

River-Mochi, 2023-03-23:
@SolderEater   in the meantime you might like to read this ticket  which is about all the different ways people have suggested for  calibration. 
https://github.com/Ralim/IronOS/issues/303

River-Mochi, 2023-03-30:
@SolderEater  tbh  I think that one tip you got on aliexpress that was 34C  off from everything else might  be defective.

It is interesting,  how do other  irons do it, that allow you to put in a human readable number and the calibration method is more straightforward. 

  I've seen that type too,  where you can  use a temperature  sensor like Hakko 119.  and if you see  the hakko shows  330C  and your iron is 320C, then it allows you to go to calibration section and  type in 10C  to add.  

and although that might work for the  current temperature,   it might not be correct now  for other temperatures on the same iron.  like  what happens at 280C  or 350C?

discip, 2023-05-16:
Closing, as the OP seems to be gone.

closed 2023-05-16 by discip