Ralim/IronOS #1471: Stange temperature after overheat

YaBrick, 2022-11-14:
Hello, 
Sometime ago I've used TS100 at 450C° for about an hour, and after that solder started always showing 200C° at room temperature, without heating. It seems like some shift in temperature - when it show 320 - in reality it's about 120, 420 - 220, etc. And that's pretty annoying, because I pretty much can't solder anything that requires more than 200C°. Tried calibrating temperature, resetting settings, updating firmware - nothing worked. Tried with different voltage - still shows 200C°. Any ideas what's wrong?

River-b, 2022-11-16:
This is interesting, especially since loading the Miniware Original firmware fixes your issue. 
at first one might expect the Tip became defective, but since your tip works fine and the temperature displays normally using Miniware firmware, this is interesting.

side note: what is the maximum rating  that a Miniware TS100 Tip  can run at?   I can't find maximum specifications for the Miniware tips anywhere.

Ralim, 2022-11-16:
Miniware firmware has a whole bunch of mitigations for hardware failures that make sweeping assumptions. The most likely thing is you have an issue with your cold junction offset due to defective handle temperature. They do this to reduce returned units from failures, and instead assume most users wont notice the error in the temp readings.
Can you please list out the values in the debug menu for your device?

All of the tips are rated for 0-400C operation.

River-b, 2022-11-16:
> 

This is interesting Ralim, basically IronOS is showing people the truth/true values, and Miniware hides the information.  the temperature will still be way off  using either Miniware or IronOS, it's just the IronOS  displays more true values.

River-b, 2022-11-16:
@YaBrick  look at [Troubleshooting Document](https://ralim.github.io/IronOS/Troubleshooting/), you _may_  need to replace your Temperature sensor.  fairly easy to do on Pinecil as we have the Schematics and part numbers. I have no idea if ts100 is easy.  you may need other parts replaced too - need to look at Debug menu and report your readings (see below). 

@YaBrick   when you do your calibration, make sure you follow the correct technique.
1. Update to the newest IronOS firmware.
2. leave iron unplugged for a while  to make sure handle temperature inside the PCB is room temperature. I sometimes stick my iron in the Refridgerator for 5 minutes to speed this up and then  leave it in the room for 5 minutes more to get to room temperature.
3. Upon a fresh plug in bootup, right away  go to Advanced > calibrate temperature.  Do calibration,  then scroll back to the main screen. Unplug and plug back in the Iron.  depending on which firmware it is, you will see Offset number either right after the calibration is done  or when you reboot. (don't worry about what this offet number is but you can report it here).  
4. Follow the [Debug  document](https://ralim.github.io/IronOS/DebugMenu/) in Github  and enter the **hidden Debug menu**. 
5. Report the Handle temperature and Tip Resistance,  might be called   HAN -C   and Tip-R   or Chan and Rtip. (Ignore sections of the document discussing Pinecil Iron as it is very different from TS100). 
6. Report also what Firmware you used for  4-5.

YaBrick, 2022-11-16:
Calibrating as @River-b sujested hasn't helped, same result
Here's the photos of from debug menu, handle temperature - 1997,
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
 and tip resistance - 832. Firmware 2.19, English.

YaBrick, 2022-11-16:
Haven't found model of the sensor, so I just.... Straight up asked for it from Miniware support. Also haven't saw anything about sensor on the board

River-b, 2022-11-17:
**Handle Temperature** (Chan or Han C)  at  **1997**  is  way off.    Mine is usually  around 27C, it's supposed to be near room temerature or  just a little warmer.  if your CHAN = 1997   then that is why  you can not do any normal  calibration.  can see this explained in the Troubleshooting document too that I linked you before.

There were **several suggestions in Troubleshooting Document** for what to do with  a **Chan that is reading higher**.  Let us know  **which suggestions in the list you tried** and what happened after you tried each of those suggestions?  I recommend you do all the suggestions there first. 

Please Crop your photos first to just be handle if you can so it takes up less space,  I had to scroll through many many pages of email and they took up so much space on screen I missed the text.  we only really need Firmware, Chan, Rtip , Ctip. 

@Ralim  would they need to possibly get anything else besides the  Temperature sensor on the PCB for a C-HAN that is way off? 

Some of the Temperature sensors types were listed in the  [Troubleshooting document ](https://ralim.github.io/IronOS/Troubleshooting/) but TS100 is not there.  if you find out let us know. 

https://ralim.github.io/IronOS/search.html?q=han+C
```
[Han C](https://ralim.github.io/IronOS/DebugMenu/#han-c)
This is the handle temperature or more accurately the reading of the Cold Junction Compensation (CJC) temperature sensor. This is expressed in °C. Range of 20-40 °C is normal depending on how hot/cold
```
[image]

Ralim, 2022-11-17:
Yep this 100% looks like something hardware-wise is wrong and its just their firmware masking it and assuming your iron is always 25C. 

From the schematic: https://github.com/Ralim/IronOS/blob/dev/Development%20Resources/TS100%20V2.46%20Schematics%20V1.0.pdf

Part's suspect would be the temp sensor (TMP36GRTZ) and the R32 as the prime suspects.
Noting that because of the two board construction of the TS100, a loose connection between the two boards could also cause this.

River-b, 2022-11-17:
nice schematic.  welp. you have several  things  to Check / Resolder Per  Troubleshooting  Guide on Han C (Chan) 
and   also Possible  replacing the 2 parts Ralim mentioned.

[image]

YaBrick, 2022-11-17:
Ok, thank you, I checked everything writed in troubleshooting, also reflowing - welp, yeah, it's something with the R32 or the sensor

River-b, 2022-11-17:
both are pretty low cost items to replace if reflow/solder on them did not fix it.

this might be the TMP36 part  https://files.pine64.org/doc/datasheet/pinecil/TMP35_36_37.pdf
see if that looks similar.

River-b, 2022-11-29:
@YaBrick   did you also look at the connection Ralim mentioned?  The connection between the two boards?  I don't have a TS100, so I don't know where this is. but check that before buying parts.

`Noting that because of the two board construction of the TS100, a loose connection between the two boards could also cause this.`

River-b, 2022-11-29:
Could you give a photo  of the  two board connection in the TS100 iron?

YaBrick, 2022-11-29:
I'll send it, just want to return to fixing the solder a bit later

River-b, 2022-12-09:
@YaBrick   could you close this ticket if you problem is resolved.  Also  list out what  was the final Solution?

YaBrick, 2022-12-10:
Well, for now there's no final solution - I'll just try to change parts of the solder when I'll be able to. I think boards connected correctly, so it should be the problem that Ralim said. Until then I'd better close the problem, yes?

closed 2022-12-10 by YaBrick

River-b, 2022-12-11:
@YaBrick   I would still reflow there if you can, as it doesn't hurt and it might fix it.  it's hard to tell with eye sometimes if something is cold joint. 
Also good to clean up  any  excess solder left on PCB so it doesn't float around and cause a short.  Last photo,  near the  front copper contacts looks  dirty on PCB with flux and solder.