Ralim/IronOS #1472: Add feature to stop Iron if Tip fails and resistance/ohm is very low

River-b, 2022-11-16:
**Is your feature request related to a problem? Please describe.**
a user recently reported they had an older V1  pinecil fail in middle of soldering. They measured tip resistance to only **1 ohm**. 
 Regular  longer tips are expected to be ~ 8.0 ohm.   We suspect they just need to replace their mosfet and it failed when the Tip failed to 1 ohm resistance causing mosfet to overload.

Usually when we see tips fail, people measure tip resistance as  Infinity and this is a better failure of the tip because then it doesn't  murder Pinecil Iron when the tip fails.

**Describe the solution you'd like**
Is there a way to detect resistance of tips below  6 ohms, show a  no-tip detected  and shut off the  Iron?

Ralim, 2022-11-16:
Could be done on V2, no way on V1.

River-b, 2022-11-16:
this would be excellent enhancement then and another reason to get a V2.