Ralim/IronOS #1476: Add IronOS  website link to Firmware

River-b, 2022-11-17:
**Is your feature request related to a problem? Please describe.**
1. Add a Link to Github IronOS  to inform people:

- Origin of the OS, then they know where to get updates, support
- if they are curious and visit, they find the helpful documents
- inside IronOS is a great place to find this link
- People complained that Pine64 does not give a link to Github IronOS  on the V2  webpage. Pine Store was contacted about this but no change was ever made to the webpage.

2. Miniware people most likely know where Github IronOS is because they have to purposely come here to get the firmware to overwrite the default Miniware OS. 
3. Other devices like Pine64  already comes with IronOS so people do not necessarily know where to get support or Updates. Other software/firmware include a link to get updates/support, this is a common practice. Some people reported they never updated Pinecil because it already came with  IronOS. We still find people recently who have version 2.14 firmware, having the weblink might encourage more updates which will enhance the user experience. 

**Describe the solution you'd like**
OPTIONS
1. 
Advanced Settings >  IronOS Support   
  scroll sentence:    `Go to https://ralim.github.io/IronOS/`

2.
Advanced Settings >  About IronOS
  scroll sentence:    `Go to https://ralim.github.io/IronOS/ for Updates & Support`

Or when they select  "IronOS  Support"  or "About IronOS"  words  then they see  sentence also.

Ralim, 2022-11-18:
We dont have the space to add this for the TS80 series of irons and the settings menu is fairly shared.
Would be better to ask this to be in the manual with the Pinecil?

River-b, 2022-11-18:
I will try again. I have already in past already with no success:

1. sent email to Pine Store asking for IronOS  and/or Pinecil Wiki   links to Pinecil sale page (no change happened).
2. sent message to Gamiee  for User Manual change .  or if they don't have time, just  give me access to the User Manual document and  I could make changes to it. I want to add large font weblink and a QR  code on front page. 

too bad there is no room.  worldwide  havign the support link so easy to use.

Ralim, 2022-11-18:
I would prefer the link be held elsewhere rather than adding junk to the firmware; if its not going to be useful to a majority of users it would be far better for it to be held elsewhere. I also think this is not even strictly an issue of customers not knowing of IronOS  (Google makes the finding it easy, and IronOS is mentioned in the sales docs and media around the Pinecil), but more likely an effect of just "people dont update until they have to"

closed 2022-11-18 by River-b