Ralim/IronOS #1477: Feature Discussion: Locked-Down Mode

Electro707, 2022-11-19:
This Issue is to discuss a potential feature: Having an option to have an iron go into "Lockdown Mode" where the temperature and settings are set, and can only be exited from Lockdown Mode with a pin for example.

As an example, at a classroom or student lab where soldering is involved, it is sometimes preferred to have the irons locked down so students cannot mess with the temperature for example.

I don't mind working on it myself, but I wanted to get the community's ideas and feedback before I spend the work working on it.

DjordjeMandic, 2022-11-19:
Not a bad idea but I see it a bit difficult to enter pin with 2 buttons only. But one button can be used for incrementing digit and another for switching to next digit/confirming input. Holding that another button could be interpreted as exit/back

Ralim, 2022-11-20:
Hmm; how you enter exit could be interesting. Not against the idea, but maybe pen out rough ideas here before coding; as I suspect the UI/UX will be the controversial part 😁

River-b, 2022-11-29:
@Electro707   perhaps, since Pinecil  V2  is gettings some BLE development now.  it might be possible to do  a PIN  Lock using Android phone or a PC browser . 

if you have  V2,  you could test current  BLE webpage now  and talk to Joric to see if he can help you make that. it  would need to be  Something you can click on Lock on Bluetooth page, but then the Pinecil V2 firmware would also need to accept that message as Total lock down on any setting changes or messing with calibration .  and they can only hit  (+)  and (-) .  I'm not sure what abilities you want to keep for students. 

I notice  another Iron  puts PIN  lock out on the  BLE bluetooth app,  since theIron itself doesn't have a way.  this is an example of another BLE  iron.
[image]

If you want to try out  the Tester BLE API  app for Pinecil V2 right now,  there[ are instructions here](https://www.reddit.com/r/PINE64official/comments/z5ix9a/bluetooth_ble_testing_pinecil_v2/)
you just need Android phone with Chrome,  or  Windows with Chrome,  but Chrome on Linux also works if you enable bluetooth flag.   See instructions in this reddit post for all Browser options. And contact  Joric in his  Github  see if he can work with you to add Lock out Pin, and Lock Icon to the BLE app  he is currently working on. 

I've tested it on several browers.  Honestly if we had   Heat On/Off  working, I would fully use my Phone on a stand in color  over the tiny screen. 

Only takes 2 minutes to set up your V2 to start looking at the Potential of the BLE Web app.  (not possible in iphone/ipad).

River-b, 2023-02-20:
Perhaps see the brand new **PineSAM  BLE  app** for Pinecil V2 which has BLE bluetooth capabilities.
PineSAM app is a locally run script which opens a browser on your local PC/laptop  Windows/Mac/Linux  and allows you to change all V2- Pinecil settings.
I can also type in the temperature I want like 307 °C  or I can use a slider to move it. 

https://github.com/builder555/PineSAM/issues/22

Perhaps between this BLE bluetooth app for V2  and  a lock feature in IronOS, it could at least be done for Pinecil V2.   people could activate  Lock Mode on pinecil, but would need to use BLE app to set it up and do the Pin?  
And if people got messed up and locked out, just Flash whole firmware again would fix it.
This would just help as a preventative measure in student settings, but if someone wanted to they just simply flash whole thing again to get rid of the lock mode?

discip, 2023-09-23:
Closing as OP doesn't seem to be interested anymore.

phillipthelen, 2025-03-21:
I am considering suggesting Pinecilv2 irons for my makerspace. Having an option like this would be useful to prevent people from messing with the settings and changing how things work for subsequent users. We'd still need temperature setting enabled, to allow for different use cases, but other than that it would be good if the whole settings menu is just disabled.

Using the BLE api for this would be a very good use case imho, since as already mentioned here, the UX of inputting a pin with just two buttons is… tricky (more code maybe? :P ). Using the USB connection would of course also be useful, but would require something like #837 first. And then clients that actually use that.
 I'd have to do some digging into the codebase to get familiar enough, but I'd be interested in working on this. Since currently the BLE connection can be established without any passcode or anything, a simple boolean `enableLockMode` wouldn't really work, since anybody could just disable it again. I'm not too familiar with Bluetooth yet, but from looking at the documentation in the wiki, maybe the following could work:

1. Two new settings (internal, not exposed through the on-screen display): "lock mode enabled" and "lock mode passcode"
2. One (or two) new characteristics on the settings service for enabling/disabling lock mode. Similar to the save settings service. When locked, the configured passcode has to be written to the service to unlock. To lock again, either a separate service is used or the same service that locks the device if `0x0001` is written?
3. when in lock mode, any updates to the settings service characteristics (apart from the unlock one obviously) don't do anything.

Feedback on if this is a sensible idea/approach and something that would be merged in when done would be appreciated!

ia, 2025-03-23:
This is very interesting feature request, yet pretty specific for a particular use case.

As a _straight forward "brute force"_ approach, the easiest, simplest & fastest solution - to make a custom build(s) for your set of irons disabling two-three lines of code in a few places:
- disabling trigger to go to Settings menu;
- disabling soldering temperature change (optionally, since as I can see, there is no clear agreement on that yet about should it be changeable or not as well :)

Not sure if it's exactly what has been requested, but for the use cases described above, I think that having a custom build of firmware with read-only settings (some sort of _"kiosk"_/_"demo mode"_) is probably the most perfect solution. _**So let us know what do you think about this?**_

Approach to add it as a part of a setting to a _generic supported iron_ raises a lot of questions how better to implement it, for the reasons mentioned above by others:
- hard to enter any pin/password with two buttons only;
- plus, overcomplicating code base with very specific feature.

Now, for _Pinecil V2_ this is slightly different deal, and the way how it could be reached is something like:
- user set locking mode;
- _IronOS_ on _Pinecil_ generate pseudorandom 4 digit pin;
- user must enter 4 digit pin in a _smartphone app_ to unlock any changes.

However, I'm not sure how realistic it is, since I'm not familiar with _BLE_ stack on _Pinecil_ myself yet.

Ralim, 2025-03-23:
@phillipthelen  & @ia 

Yeah I think this is a good use case to do. Especially now things are more stable. Avoiding custom builds would be ideal long term, and this isn't crazy overhead. I think more work will be on the apps to support this :(

This will require the BLE capable apps to implement support for this as well, so having only able to be setup by them might work to prevent locking yourself out as easily being possible.

However, I've been planning to make up some tooling to allow creating a flashable file that pre-loads settings and can also be used to re-set settings.

I might work on that sooner as I see that as a pre-requisite for this to ensure users have an easy way to get back out of this mode. (And for irons with no BLE, then only way in/out would be to flash one of these files?)

I think we would want a few modes:
1. No locking (default)
2. Partial-Lock
3. Full-Lock

### No Locking
Exactly as it is now

### Partial-Lock

Allows user to change the temperature when in the soldering mode.

### Full lock

Disable all temperature adjustment and lock out the entire menu

ia, 2025-03-24:
> However, I've been planning to make up some tooling to allow creating a flashable file that pre-loads settings and can also be used to re-set settings.
>
> I might work on that sooner as I see that as a pre-requisite for this to ensure users have an easy way to get back out of this mode.

I just was going to start to work on that myself, since I'm getting tired to have my favorable daily settings wiped every time when I test some builds... and was going to ask for your help with this, because I have some idea how it could be implemented from the point of source code, but I'm not familiar with `DFU`/`HEX` format to make `Settings` structure compatible with flash process in binary form.

> (And for irons with no BLE, then only way in/out would be to flash one of these files?)

Yeah, I guess... we will see.

Superminaren, 2026-04-05:
I second this. I help running a soldering workshop and being able to lock settings and button presses would be a blessing due to:
- People accidentally turning off idle mode
- Setting temp too high for the tip
- Changing other settings accidentally