Ralim/IronOS #1479: Pinecil V2 crashes with iron OS v2.20 during heatup whith EPR USB-C charger

asap2Go, 2022-11-24:
To reproduce:
Flash Pinecil V2 to ironOS 2.20 Release.
Power Pinecil V2 via USB-C with and an Anker 717 Charger (140W with USB-PD 3.1 EPR).
Pinecil starts up and detects 28V supply as shown in the menu.
Change into solder mode.
Pinecil heats up for about 1 second then crashes and reboots.

I tried changing settings to limit the power draw to 30W, but the problem still occurs.

Power supply and Anker cable (rated 140W) work fine as tested on a Laptop. 
Pinecil also works fine with a barrel jack and 19V power supply.

Is it possible to add a menu option to entirely disable EPR and only use SPR?

River-b, 2022-11-27:
which Exact  2.20  did you use?
there is no stable release yet only beta versions.
Looks to me like you are not using the  2.20  beta that has the EPR  in it.
I'm using a beta now that works great with all my EPR chargers.

River-b, 2022-11-29:
Phiten  in discord owns an Anker 717.  and Ralim made changes in the code a couple months ago to fix one messaging issue with it.  he reported that it worked.
Usually changing to 30w and such does not work with the EPR  chargers.  reboot has to do with the PD  negotiation failing because Pinecil and Charger do not agree one some specific term.  EPR chargers also need Chunk messages sent very quickly which is not true for older  USB-C  3.0 chargers. 

If this still doesn't work with link I provided for  the beta, then could go to discord and ping Phiten directly to see  what he is doing.  Phiten is one of the discord pine64 moderators so you can find his name at the top of the members list.  
Pinecil Wiki page has  direct links to all the chats. 
https://wiki.pine64.org/wiki/Pinecil#Community_Chat_links

River-b, 2022-12-09:
@asap2Go   did you use the firmware I linked? 
This ticket is still left open.   if EPR charger works with the new firmware, then  you can **close** the ticket. 

what you mean you went into debug with (+)   and  **"Selecting  28V 5amps it then works fine "**?   
The firmware does not allow selecting of any  PD usb-C voltage  in the debug menu.  The debug menu is simply  a display  of  things. 
 if you are in  the PD debug (+) held down before plugging in,  this is  separate from regular  debug menu access but either way 
 there is no user capable selector in those.   PD debug is simply  a display  of the   voltages that the Charger  is  offering to give to Pinecil , that you can scroll through the messages  using the (+)  to see all the different  things the Charger is broadcasting it will offer to Pinecil.

    Pinecil will  tell the Charger which one voltage it prefers and it usually  selects the highest one,  for example  28V,  if a EPR cable is used ( and the Charger  basically just offers Pinecil a List of available options).  All of this  with a PD3.0 or a PD3.1  USB-C charger happens Automatically,  no selector is available in Pinecil  as it is not  set up  as a triggering device like a usb-C special  trigger/decoy tester with trigger buttons.  Pinecil Simply (with no user intervention)  tells a PD charger what it can handle up to, and the charger  Automatically sends the offers, then they agree silently and automatically which one to use. 
    
you can use (+)  to view PD menu  and (+)  to scroll through to see all negotiated messages  but not to select anything.  Part of the advantage of using a PD usb-C charger is that it is a Smart charger and all  "negotiation"  is done automatically between  the  Charger and the  Device, in this case a Pinecil.    QC 3.0 chargers on the other hand are not as good  because in QC you have to specifify  in Pinecil menu under power the exact  voltage you want, which is  cumbersome.

There is no  PD  selector in IronOS  because the PD  negotiation is part of the protocol and happening automatically betwen  Pinecil  and the Charger (using  IronOS  as the brains to do the talking).    What you see when you scroll with (+)   is just a List of what they discussed in the negotiation,  the  Charger is Sending Pinecil a List of everything it can do.    There is no user selectible option in the PD Debug menu. 

Details on Debug menu here,  and at the very bottom of the page is a separate section on the separate  PD Debug menu
https://ralim.github.io/IronOS/DebugMenu/

Could you confirm?  Just  start the pinecil Normally with the new firmware,  you should see 28V  in the right of the Screen.  this means  PD3.1  smart  negotiation  is happening Automatically and normally.  there should be  no selector  needed as all PD negotiation is automatic between the  device and the charger,  using an EMark  chip in the case of an EPR cable.

asap2Go, 2022-12-20:
Hello,
sorry for taking so long to answer. 
The action you linked has already expired, so I cannot test this specific release.
I tried the newly release, stable 2.20 but the issue is still the same.

You are right about the debug menu. 
It appears that heatup works sometimes and it seems to have been a coincidence that it worked after started via debug menu.
Probably my Pinecil is just a little buggy.
I'll close this issue and use the barrel jack or another iron.

Thanks for your detailed explanations and support!

closed 2022-12-20 by asap2Go

Ralim, 2022-12-21:
I don't think it's your unit, and it's worth keeping this open to look into.

One idea I have, is how fast are you starting the heat up, ice seen once or twice where if you start heat up while firmware is negotiating to change into epr / change voltage it trips out.

But I've struggled to reproduce so not certain of it.

If you can reproduce it quickly, a video so I can measure timings would help (on 2.20).

asap2Go, 2022-12-26:
Hey, 
I changed the cable to an Anker Powerline III Flow (100W, 180cm)
and now the Pinecil works very solidly at 20V, 5A.
I assume the previous Anker 765 cable (140W, 180cm) was causing the issue.
Heatup speed was in standard setting and appeared to be quite fast just like in the YouTube Video that compares the different power modes of the pinecil V2.
It kept rebooting after getting to about 60-65°C. 
No screen issues at all during heatup.
After reboot it briefly goes into 5V fallback for less than a second and afterwards renegotiates 28V.
I can try and refund the Anker 765 cable and test whether the issue persists with a new cable. Then I can make a Video for you to check the timings.

River-b, 2022-12-28:
> I assume the previous Anker 765 cable (140W, 180cm) was causing the issue.

I have already  seen complaints about that  "anker 140W"  cable.  First, know that this cable is NOT following any USB-IF   specification. there is no  such thing as a  140W  cable to USB-IF  standards org, it  is something weird that Anker is selling  and  including  with the Anker 737 Battery  (don't use the cable).  I'm just shocked they are selling this weird  140w cable for $20  when you can get a Legit  240W cable that follows the International  USB-C  specs for half that price. 

USB-IF  only  has  two Categories of cables  as of  Dec 2021.  They only  recognize  SPR (standard Power range)  these are  the normal  cheap  PD60 cables  that have  No emark chip in them.   and ALL   EPR  cables (extended power range)   are capable of   240Watts 48volts, 5amps.   You can get a certified PD3.1  EPR cable for about  $12-$15.  and non-certified ones are  about the same price currently.  if you use a Real   240W, 48V, 5amp  EPR  USB-C to C cable, then  you will  see full 28V   on Pinecil screen.  

otherwise  honestly,  anker  is again lieing to people saying that  140w cable is PD3.1 when it is not (it will not do more than 20V on all the people who reported about this weird cable).  Anker /Euphy  is going  down tubes lately when their marketing people either publishing no real  data on cables/equipment or outright lieing to people in specs, especially at the super high price they are charging for that non-standard spec cable. 

The Chargers on the other hand  can  be PD3.1 and 28V, 140W.  We have tested at least  8 different   140w-28V chargers now and they ALL  show  28V  on Pinecil Screen  if  you have  2.20 firmware or the BLE  firmware.  
Check Pinecil Wiki,  there is a Link to the Bluetooth Firmware   in the Wiki page if you are curious about it. search  Wiki here for   Bluetooth  and you will  get a link to where that beta is archived.   https://wiki.pine64.org/wiki/Pinecil#Firmware

River-b, 2022-12-28:
@asap2Go    do you want a link to a cable that is known to work?  there are also some in  Pinecil Wiki page. 
https://wiki.pine64.org/wiki/Pinecil_Power_Supplies#Usb-C_Cables

I have the Club 3D  Orange cable that is  240w,   it works  at 28V on pinecil.  it's ok and it's certified,  I'm waiting for new  Silicone ones to come out in a couple months.   the Ugreen  240w cable also works but  again both these cables  are not silicone.  so I personally  only bought one of them as a place holder.