Ralim/IronOS #1496: when Pinecil  V2 goes to  99.9W,  it does not  increase to 100+ Watts, but resets to  Zero Watts

River-b, 2022-12-12:
**Describe the bug**
While testing Pinecil,  I use boost  several times, and on External  USB-C tester  show upwards  of  ~107 watts.
however,  Pinecil Screen  never goes past  **99.9 W**  and then  mysteriously  resets to  zero Watts, and then starts climbing again. 
Does 2.20  not allow  for  over  99.9 Watts on Detailed screen?  With EPR chargers we need this even if it's not a floating point to be whole number  105W  etc.

**To Reproduce**
See  23 second video,  play youtube in Slow.  See 13-15 seconds,  it goes to 99.9W  then resets to zero.  This is repeatable for all tests, it just resets at 99.9 to zero watts. 
**https://www.youtube.com/watch?v=TOl6z5yhGz8**

TEST if you want to Reproducts:   not recommended test,  as it could ruin  Pinecil, and definitely shortens life of the Tip. this was done on a spare 2nd Pinecil just to show that with EPR it could indeed go over  100W.
- This happens on both 2.20 BLE  and 2.20 Release version
-  small glass of water, set temperature to  1180C or 190 C  to protect pinecil.   it's sitting in water so it's a lot of effort to maintain temperature constantly and the handle / C han will rise  high, near ~50C   sometimes.
- Then  hold down  BOOST [+]  until it gets over 100-107watts on USB-C tester then release and it goes back to normal watts and maintaining the 180C or 190C  that is Set Temp.
- when boost is held down,  External USB- tester will show 100-105Watts.   Then I release button and  it  goes to lower watts just to maintain the Set temp. 
- In all videos  I made showing Real Pinecil screen,  at 99.9W  , it then resets to ZERO watts.

**Expected behavior**
Show  the 3 digit  watts numbers 100-126 Watts   and don't  reset  watts to  zero  when it hits  99.9 Watts

**Details of your device:**
- Pinecil V2
- Any  2.20 release.   I have to find  old 2.18, but I think 2.18  showed the 110 W.
 - Power adapter being used: all EPR  charges ,  28V, 140W, EPR cable
 
Also see attatch video of different test.  Did 7 tests, this happens every time. 
https://user-images.githubusercontent.com/97197236/206949430-dd37a5e2-2b5f-47b0-a9af-5049030ad4da.mp4

Ralim, 2022-12-12:
Yes at the moment it's limited to 2.1 format when displaying the wattage.

This is a big, will try and get in 2.21

closed 2023-01-24 by Ralim

River-b, 2023-01-24:
it works. I got it to go to  106W  on the screen.  it no longer resets  at 99.9 watts to zero.

River-b, 2023-01-24:
https://user-images.githubusercontent.com/97197236/214387593-ed300347-9f04-4719-aab5-c92d90ea26fb.mov

for historic proof.

PR #1536 (2023-01-24): Drop decimal when > 99.9W

* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [x] There are no breaking changes

* **What kind of change does this PR introduce?**
When the estimated wattage is > 99.9W we drop the decimal and just print 100W

* **What is the current behavior?**
#1496

* **Other information**:
Closes #1496 when merged