Ralim/IronOS #1501: MHP30 Flashing bug

cronald-sys, 2022-12-15:
**Describe the bug**
Attempting to update my MHP30 from v2.19.A3BBABC to the latest release results in a .NOT file, not .RDY or .ERR

**To Reproduce**
1) Start with v2.19.A3BBABC
2) Initiate DFU mode
2) Copy MHP30_EN.HEX onto MHP30

**Expected behavior**
A .RDY or .ERR file

**Details of your device:**
 - Device: MHP30
 - Release: v2.19.A3BBABC
 - Power adapter being used: Verified USB A - C data cable (used moments before to update a Pinecil V2)
[image]

emrahkaracam, 2022-12-19:
Same problem for TS100. I accidentally loaded the bin file instead of the hex file. Device only boots in dfu mode(3.45). 

Is there a way to fix it?

Ralim, 2022-12-19:
As to the hex not working, it worked for me here.
What bootloader is on your mhp?

emrahkaracam, 2022-12-20:
I fixed the ts100. Download linux version booted from usb. Try the firmware update on linux.

discip, 2023-01-07:
@cronald-sys
Did you manage to solve your issue?
If so, please don't forget to close this. 😊

closed 2023-01-22 by Ralim