Ralim/IronOS #1508: Pinecil V2 reboots at target temp

LeasyBen, 2022-12-29:
I am using shargeek 100w charger and anker flow usb c cable. The iron heats up quick but when it reaches 350C it reboots. Sometimes it doesn't reboot and works for a while but then reboots later. This behavior also happened on an Anker 65w charger.

Ralim, 2022-12-29:
Can you confirm what firmware you are running?

LeasyBen, 2022-12-29:
Happens both on the stock Pinecil v2 firmware and after I upgraded to 2.20

Ralim, 2022-12-31:
Rebooting after reaching temperature is possibly the charger/cable not being stable when the power draw drops quickly.
You could trial setting a wattage limiter (will constrain average power usage, so its less of a drop)

LeasyBen, 2023-01-01:
> Rebooting after reaching temperature is possibly the charger/cable not being stable when the power draw drops quickly.
> You could trial setting a wattage limiter (will constrain average power usage, so its less of a drop)

Thanks I’ll try it, are you aware of any chargers that are well tested to not have this issue at all? I am willing to return the charger for one that works with no modifications

Ralim, 2023-01-22:
The Pinecil wiki should have some details afaik; you can also ask in the community chats and get a lot of info quickly usually.

River-b, 2023-01-25:
> 

LeasyBen  Pinecil Wiki page has links to the live Pinecil  Community  Chat  in  Discord where volunteers are.  the discord  also has a bridge to  telegram and matrix if you have those  and you can still see the chat but not have all the discord  features.

River-b, 2023-02-02:
@LeasyBen   has this issue been resolved?  if yes, please close this  ticket or let us know.  have you joined the live Pinecil community chat channel?

esmith13, 2023-02-02:
@LeasyBen are you using a Pine64 desktop power supply? If so these do not have a proper grounding setup and cause this issue on both V1 and V2 Pinecils regardless of firmware version. If so, a quick fix it to run a ground wire to the ground pin on any of the USB-A ports on the power supply. In my case I cut up an old USB-A cable and connected the black/ground wire to the ground pin of the same outlet I am powering the the power supply with and this completely resolved the issue for both my Pinecil V1 and V2. I found the suggestion on the PINE64 community forum about a year ago.
Hope this helps you.

River-b, 2023-02-02:
the  newer Pinepower Desktop  has 3 prong Grounded  cord... they started selling upgraded model sometime in  2022.  but  older one or  other  2-prong USB-C chargers can cause  some grounding issues.  
 I personally don't use a ground cable on the back of the Pinecil Screw and don't have issue.  but I do  always  plug in any 2- prong  usb-c chargers first to a  surge protector strip  first and then that has a 3 prong  ground plug  to  wall.   I've not had any strange reboot issues  on any chargers  except  perhaps an initial  reboot in the first  5  seconds from "some" of the newest  EPR PD3.1  140W ones (i.e. Apple but apple is a little weird).

River-b, 2023-02-23:
@LeasyBen  could you try the beta firmware that is linked as a zip in this ticket. see if it's a PPS issue. If you have PPS on that charger and it is rebooting, then perhaps this fix in this link will work.
https://github.com/Ralim/IronOS/issues/1460#event-8590732152

if it works, let us know and pls close the ticket.

LeasyBen, 2023-02-23:
> @LeasyBen could you try the beta firmware that is linked as a zip in this ticket. see if it's a PPS issue. If you have PPS on that charger and it is rebooting, then perhaps this fix in this link will work. [#1460 (comment)](https://github.com/Ralim/IronOS/issues/1460#event-8590732152)
> 
> if it works, let us know and pls close the ticket.

Thanks, will try asap

LeasyBen, 2023-02-23:
Alright, so it does not even power on with that beta firmware. I have v2 iron btw. Flashed it back to 2.20 and it works but still have the behavior mentioned earlier where it powers off randomly when using shargeek 100w charger. Note that this did not happen with a random 90w laptop charger I found, but I prefer to use my shargeek 100w charger.

River-b, 2023-02-24:
@Samuelrmlink  do you have a version of the PPS fix beta for Pinecil V2 so that @LeasyBen  could try it?

Samuelrmlink, 2023-02-24:
@LeasyBen @River-b I've got the same Shargeek Storm 2 power bank.
It's a pretty awesome looking unit - I'll do some further testing over the next few days to try to figure out what may be wrong here. 
I don't yet have the Pinecil V2 - it is on order. 

In the meantime - try flashing this firmware, then untick the "PD VPDO Enabled" option in the Power settings. 

Make sure to use the right files for your device. In this case - you'll want to flash the Pinecilv2 firmware. 

[Pinecilv1_PPSfix.zip](https://github.com/Ralim/IronOS/files/10820252/Pinecilv1_PPSfix.zip)
[Pinecilv2_PPSfix.zip](https://github.com/Ralim/IronOS/files/10820253/Pinecilv2_PPSfix.zip)

Samuelrmlink, 2023-02-24:
Just a clarification for my own curiosity - are you talking about a Shargeek Power bank, or is this a wall charger? I know they make both.

Samuelrmlink, 2023-02-24:
Also keep in mind that you may want to wait a few seconds after changing the PD VPDO Enabled setting before unplugging the iron. 

Sometimes it takes a few seconds to save the settings.

River-b, 2023-02-24:
> Sometimes it takes a few seconds to save the settings.

to save settings on pinecil, you have to make all changes, and then scroll all the way out of the submenu back to the main screen. and then it is saved to flash. then can reboot.

ironos doesn't save settings one at a time as you do them inside a sub-menu and this is to preserve flash since you have a limited number of flash changes (however most people will not see this limit).
 Idea is if you change 3 different User interface settings for example, it's only when  you scroll completely out of that back to the main screen for example, that pinecil will flash so it's all done one time, one flash.  then unplug and plug back in and go to the menu setting to check if it's correct.

Samuelrmlink, 2023-02-24:
> ironos doesn't save settings one at a time as you do them inside a sub-menu and this is to preserve flash since you have a limited number of flash changes (however most people will not see this limit).
That makes sense.

LeasyBen, 2023-02-24:
> 

It is the 100w wall brick. I will test the new firmware out soon. Hopefully this fixes it

LeasyBen, 2023-02-24:
> @LeasyBen @River-b I've got the same Shargeek Storm 2 power bank. It's a pretty awesome looking unit - I'll do some further testing over the next few days to try to figure out what may be wrong [...]

I am still getting problems rebooting with that new firmware and the option unchecked

River-b, 2023-02-24:
One more thing to try, go to a completely different room in the house that is on a different circuit and has a 3-prong grounded wall outlet.

@Samuelrmlink  one more thing to note is that they report that this same behavior of reboot when reaching 350 C  on two different chargers, shargeek 100w charger and  Anker 65w charger.
if it's happening with both chargers, the other variables I would check is  change to a completely different  USB-C to C cable. and if that doesn't resolve it on both chargers, try  other wall outlets around the house not in the same room or same circuit. 
Also try plugging the charger first into  a surge power strip and then plugging the 3-prong power strip into a wall outlet.

If it happens to 2 different chargers, and a cable is also swapped out and it still happens, then something maybe going on with Pinecil. in that case I would try the Ground wire from Ground screw at back of Pinecil to a good earth ground.   and then if it still doesn't work , need to look at the pinecil itself or firmware.

I have not had this experience or heard of this happening yet where it reboots  when it hits  the set temperature.

raduc, 2023-03-24:
I am getting the same type of reboot with this 65W PD adapter:
https://www.amazon.com/gp/product/B09F8WMW9L

The Pinecil gets to the target temp, then it reboots right away. If I restart it it works for a while then it reboots periodically again. I tried the beta firmware but it does not fix it. I also tried a different cable and plugging it directly into the wall outlet.

Thanks.

River-Mochi, 2023-03-24:
> The Pinecil gets to the target temp, then it reboots right away. I
@raduc    have you  tried putting a ground wire on the ground screw?  some  usb-c  chargers are ok and will work fine , but because they are  2-prong only and do not have a 3rd earth ground prong there is no were for any small current leakage in the charger to go and this  would cause pinecil to reboot. 

1. Try a different charger
2.  try wall outlets in a different room, or in surge protector that has 3 prong cord.
3.   Ultimately -  you might indeed need to try  putting a ground cable from ground screw on Pinecil to a known good ground point.  That is what the screw is for.  It's good to check if  a ground cable fixes it, then the charger you have has a little current leakage. Class II  chargers  can have different levels of current leakage, some more than others - even 3  of same model and only one causes reboots.  they don't have a 3rd earth pin to take that away.

One way to do it is  if you have a  PC  and it's plugged into the wall with a 3-prong  cord and  grounded, then you could run a cable from the  GND screw on Pinecil  and then clip other end to the metal on back of the PC like near the rear power supply fan.   or  another known earth ground point. 
basically Dancho in Pinecil pine64 chat  he made a wire using a round flat clamp that would fit under the screw,  and then ran that to a huge metal plate on his desk. and from there another ground wire goes to his  earth ground.

None of my 2-prong  USB-C chargers cause reboots but  it does happen with some people and it depends on the house electric situation too.   I plug all of my  2- prong usb-C  into a surge  protector strip first, and then  that has a 3-prong plug which plugs into my wall, and I've not had any reboots. 

I also have a USB-C  charger that has a fully  earth grounded  3-prong cord. all the USB-C  metal shields tie into that earth  prong (they have continuity with it). that is the one i use the most.  but only Pine Store sells this one. 

Also for an idea how to do it,  Google  TS100  Ground Wire, since the ground wire thing is done same on TS100 and Pinecil and there are some Youtube examples.

River-Mochi, 2023-03-24:
[image]
[image]

River-Mochi, 2023-03-24:
I would first try to use one of the USB-C chargers that other people got that  they know doesn't cause reboots. if you don't want to do ground wire.  there is a list of several on the Pinecil Wiki page and even more  kinds if you ask in the live pine64  Pinecil  Chat.

raduc, 2023-03-24:
Yes, it works with other chargers, only that particular one fails with it. Even so, I can use it but have to press the plus button to restart it every once in a while.

River-Mochi, 2023-03-24:
I suspect this is what is happening with the Original poster also.   @LeasyBen    per  photos above,  try add a ground cable to  back of your soldering iron, there is a special screw for it.  and then run to earth ground. 

If  this is only happening  on some charger,  and not the others, then  that charger simply has current leakags as many  Class II  usb-C chargers do and since they only have 2-prongs,  there is no 3rd earth  prong to take away the current leakage and this tiny amount can cause Pinecil to reboot. 
It might be acceptablle  for charging a phone battery as no one is wathcing the phone charge to see if it goes in and out randomly on the charging icon.   Sometimes these same  2-prong  chargers also can cause  Kitchen GFCI  wall plugs to flip too. 

There is a list of known  USB-C chargers on the Pinecil wiki page  you could try that do not  cause reboots for people.  or join the live volunteer  Pinecil chat,  and  people could help you pick one out.   could also just try ground cable too as that might give a path for the current leakage from the charger.

River-Mochi, 2023-03-24:
> Thanks I’ll try it, are you aware of any chargers that are well tested to not have this issue at all? I am willing to return the charger for one that works with no modifications

LIST here of some, and  Pinecil live chat people can help you  with others if you want to ask about other brands. 
https://wiki.pine64.org/wiki/Pinecil_Power_Supplies#USB-C_Charger

Vsono, 2023-04-02:
I just got a Pinecil v2 and I'm having a similar issue.

Sometimes it reaches the target temp, sometimes it reboots as soon as I click the [+] button.
I'm using an [EP-T6530 65W charger](https://www.samsung.com/us/mobile/mobile-accessories/phones/65w-trio-adapter-ep-t6530nbegus/) from samsung which has 3 output ports.
It works well on the 25w and 15w port but not on the 65w port.

I've tried several solutions I could find, but none of them worked out.

1. Putting a ground cable
2. Set the power limit to 20w or 40w
3. Using the latest firmware on the actions page
4. Using a firmware Samuelrmlink uploaded ([Pinecilv2_PPSfix.zip](https://github.com/Ralim/IronOS/files/10820253/Pinecilv2_PPSfix.zip)) and unchecking the PD VPDO Enabled option.
5. try other USB-C cables

PD Debug says:
<pre><code>State 12 NO VBus
1 5V    3.00A
2 9V    3.00A
3 15V   3.00A
4 20V   3.00A
5 5-11V 3.00A
6 5-16V 3.00A
7 5-21V 2.25A
</pre></code>

Vsono, 2023-04-02:
I've uploaded a video https://www.youtube.com/watch?v=h19CtuH-emg
Target temperature is set to 200 degrees c.

River-Mochi, 2023-04-03:
@Vsono   and others,  what I see  seems like there is an issue with the charger.
First your pd output, it says  20V 3amps which means on one  USB-C port you would have max of  60W, so I dont know how they are saying 65W ?  the math doesn't work. 
based on what you are showing in PD  debug that charger has PPS. what was the reason you unchecked PD VPDO?
Reset settings to Default and try again, don't limit the power.  If it doesn't work I would return that charger and get a normal   USB-C  charger that has at least one single port that supports  20V, 3.25amps, PD65w.
I assume you already tried to change the cable also. 

- See these 2 examples here in this other ticket with videos.  https://github.com/Ralim/IronOS/issues/1644
- One video example was  reboots,  and another was  unstable temperatures when they reached set temperature.  
-  **Solution**:  both  problems were immediately resolved when they got new chargers that were  good  usb-C 20V 3.25amps, (PD65W).  It's important when getting a charger  to look in specs or photos people post of the bottom of the charger to see if it has on PD  port at least that says  20V, 3.25amps. 
- One was using an old  DC barrel 24V, 2amp charger he had  from old TS100 iron that is too weak for Pinecil V2 because it did not have the minimum 3 amps.   
- other person was using a very weak 30w  low amp phone charger so whenever he got to set temperature it would drop up and down because it didn't have enough amps to keep up a stable  soldering temperature.

See the Pinecil Wiki article on power supplies,  and recommendations there. https://wiki.pine64.org/wiki/Pinecil_Power_Supplies

Vsono, 2023-04-04:
@River-Mochi 
It could be the charger but it had no problem with any other devices so I can't really tell.
I just noticed that when I use the [usb c cable from Pine store](https://pine64.com/product/usb-type-c-to-usb-type-c-silicone-power-charging-cable-1-5-meter-length/), PD Debug says lower values than the charger is capable of. thought the cable could handle 65W, I apologize.

With the 5A cable, PD Debug says:
<pre><code>State 12 NO VBus
1 5V    3.00A
2 9V    3.00A
3 15V   3.00A
4 20V   3.25A
5 5-11V 5.00A
6 5-16V 3.00A
7 5-21V 2.25A</pre></code>

I changed VPDO options along with the others because there were not many options I could try.
unfortunately, pinecil still reboots with default settings.

I might use a dc power supply for now.

River-Mochi, 2023-04-04:
> cable could handle 65W

the cheaper $3 cable from pine store is a basic  usb-C cable and basic SPR cables are only rated for PD60W since they have no emark. I've not heard of or purchased any usb-c  cable that is rated PD65w.  if you search on internet, the common ones are PD60, PD100 and PD240W.   if you want to get the full 65w  reading, then a PD3.1 cable 240W  or a PD100w cable is needed (as 240w and 100w are same price currently, people are getting the 240w ones).  if the usb-c to c cable is not labeled then it's default of pd60w.  without getting into too many details here, I recommend searching Internet for  "USB-IF  SPR  and EPR  usb-c cable"  and that will explain details about the cable spec.    

 1. when having issues likes this,  the first step is to change to different cables, especially if you are using a low cost cable.  try 2 or 3 other cables first to see what happens. Also try flipping the cable over, and then also flipping it the other way. 
 
2.  I see now you were using a regular PD60  usb-C  cable  and that is why you only see  20V 3A on the PD debug. would need a higher rated cable if you  want to see 20V 3.25 amp on PD menu, but the 5w difference on the screen  means little in reality, in my testing Pinecils heated at the same speed whether I used a higher cable or a PD60 cable with a standard good quality usb-c charger.

3.  recommend trying a different cable and then different usb-C charger.   We have come across situations where some chargers simply  cause pinecils to reboot.  have seen people who own  3  of exact same brand/model of charger and 2 work fine, and 1  causes pinecil reboot.   if they first test with several USB-C cables and it doesn't fix it, then  often getting a new  USB-C charger  does.  I find that the normal ones without PPS seem to cause me less issues.  I also have a PPS charger that reboots pinecil ,  my solution is I just don't use it.  I found several  usb-C  20V 3.25amp PD65w chargers on amazon and other sites for $15-$25  and those ones with no PPS on them  all work fine for me versus this very expensive $75 charger I have that reboots pinecil and now sits at the bottom of a box. 

4.  Once you try a new USB-C  charger,  if it works let us know. 

5. if you really want to investigate deeper for science  then could  get a PD analyzer like a Chargerlabs KM002 or KM003  (newer model), connect it to PC,  with the  charger in-line, then connect pinecil also,  and  run the PD analyzer saving the logs.      Ralim  has the ability to look at the logs because he also has a KM002.  from the logs is where there are all the details of exact transactions between the charger and the pinecil can give some insight.

discip, 2023-09-23:
@LeasyBen
Is this issue solved by now?
Have you tried the latest release?
If this is solved please consider closing this issue. 😊

LeasyBen, 2023-09-24:
> @LeasyBen Is this issue solved by now? Have you tried the latest release? If this is solved please consider closing this issue. 😊

This is still somewhat of an issue, the iron still doesn’t work on some chargers. If it is possible that the iron regulates its temperature with PPS instead of PWM some chargers that misbehave may work properly

closed 2023-09-24 by LeasyBen