Ralim/IronOS #1510: SC7A20 Accelerometer Problem on 3 brand new TS80P

ceut2k4, 2022-12-29:
**Describe the bug**
Hello,
Sorry to open this SC7A20 bug report again..
I have bought 3 TS80P from Eleshop at work, and flashed them without any problem on the 2.20 IronOS.
None of the 3 goes to standby mode on their stands.
So I checked the Accelerometer, and it is this one too: the "SC7A20" sensor IC.

**To Reproduce**
1. Flash the 2.20 FW
2.  Activate "detailed soldering mode"
3. Set at "1" the "Motion Sensitivity"
4. Set at 30s the "Sleep Timeout"
5. Put the TS80P on the stand after power it on
6. Wait for the timer to decrease: it never decreases, only blink between "30s" and "29s".
Sometimes it decreases until 27s but return quickly to 30s, without any movement.

**Expected behavior**
On the stand: the TS80P must decrease the timer.
When I shake it: the timer must restart at 30s

**Details of your device:**
 - Device: brand new 3x TS80P
 - Release: 2.20.14DAF70
 - Power adapter being used: Official QC Power Supply from the black box
 - If this is an accelerometer related issue, please include its model number here: SC7A20
 
**Additional context**
I have tried the 2.18 version with the "Fixes for SC7A20 in TS80(P)", and still have the same strange thing happening.

So is there something I do wrong ? Or a setting I have not seen ?

I'm a long time user of your great firmware, at home with my TS100 I have since many years, and also an old TS80P with the MSA301 and no problem with this IC.)

Thank you for your help !

Ralim, 2023-01-22:
Hmmmm; this IC is _cursed_ 😓 
Apologies for delay.
Will probably need to figure out a good way to debug this.

ceut2k4, 2023-01-27:
Hello,
Yes it seems this IC is weird :-/
Also, I don't know how I can help ...
But, on the 3 we have at work, I can do something on one without any problem, if needed and if you would like me to do something on it.
So, don't hesitate to ask me.

Thanks again for all, and for your great work ! :-)

oermenz, 2023-02-10:
Just tried this on Pinecilv2 with SC7A20 and it also seems to freak out. Turning off the oled even. Then again I wonder why someone would want to set motion sensitivity to "1" but I guess a bug is a bug.