Ralim/IronOS #1514: Cooldown flashing on detailed idle screen faster than on regular screen

discip, 2022-12-31:
It's not a real bug, more like an inconsistency.

**To Reproduce**

1. Activate **`Cooldown flashing`**.
2. Activate **`Detailed idle screen`**.
3. Heat up the device to **~60°C**.
4. Stop heating.
5. *Memorize* the **rate** at which the temperature display flashes.
6. Deactivate **`Detailed idle screen`**.
7. Repeat 3. - 4.
8. Note that the flash-**rate** is now much *higher*.

**Expected behavior**
The flash-rate should be the same in any case.

**Details of your device:**
iron  |  Firmware Version
-|-
**TS80P**|**https://github.com/Ralim/IronOS/tree/3e2da605ab9ee3807bcde0921aca760cbddf8d7b**

**Additional info**
I did try to find the corresponding part of code, but did not succeed. 😅

Firebie, 2022-12-31:
It is in source\Core\Threads\OperatingModes\HomeScreen.cpp:
```        if (!(getSettingValue(SettingsOptions::CoolingTempBlink) && (tipTemp > 55) && (xTaskGetTickCount() % 1000 < 300)))
          // Blink temp if setting enable and temp < 55°
          // 1000 tick/sec
          // OFF 300ms ON 700ms
          gui_drawTipTemp(true, FontStyle::LARGE); // draw in the temp
```
and
```          if (!(getSettingValue(SettingsOptions::CoolingTempBlink) && (xTaskGetTickCount() % 260 < 160)))
            gui_drawTipTemp(false, FontStyle::LARGE); // draw in the temp
```

discip, 2022-12-31:
@Firebie
# Thank you! 👍🏻

PR #1515 (2022-12-31): fixes cooldown flashing
resolves #1514
closed 2022-12-31 by Ralim