Ralim/IronOS #1519: Pinecil V2 Windows 11 Issue

ckravig, 2023-01-04:
I am running Windows 11 on my desktop and when I go to plug the Pinecil V2 in to update the firmware I get the following error(s) 

**Windows:** 

[image]

**Device Manager shows:** 

[image]

**Powershell using blisp shows:**

[image]

Ralim, 2023-01-04:
I would be suspect of usb cable/hub combination first.
USB port is used to program it at the factory so the port should be fine; but you can try both flips of the USB-C to be sure.

ckravig, 2023-01-04:
Okay, I will try flipping the USB-C first, and then some combinations of other cables and ports afterward. I am using the red 
USB-C to C cable from Pine64 plugged directly into a USB-C port on the back of my motherboard.

ckravig, 2023-01-04:
I just tried flipping the cable, using another 2 different cables (C and A) along with trying a different computer that is running Windows 10. All of that resulted in the same error.

Any other advice?

Ralim, 2023-01-04:
Try a different computer 🙃 
USB port would have been used at the factory; so _unlikely_ the port is damaged (hopefully).

ckravig, 2023-01-05:
I did try another computer and it gave the same results. The Pinecil has no issues working as an iron as far as receiving power from the usb.

Ralim, 2023-01-22:
You are making sure you are holding the rear button for the first ~10 seconds while you connect it (and screen should stay off/blank) ?

ckravig, 2023-01-23:
> You are making sure you are holding the rear button for the first ~10 seconds while you connect it (and screen should stay off/blank) ?

That was it, I see in the wiki where I missed that step to hold down the [ - ] button. 

Thank you so much for the help!

closed 2023-01-23 by ckravig