Ralim/IronOS #1526: Soldering information/control data transfer via USB

charles1489, 2023-01-07:
With Bluetooth we have soldering information/control data transfer between soldering iron and another device. Could we do the same with USB, so that the same thing would be possible via cable as well? This would allow a similar function for those soldering irons that haven't Bluetooth but have USB. I know the soldering iron requires lot of power, but for example Pinecil V1 also barrel jack. And there are USB hubs with auxiliary power.

Ralim, 2023-01-07:
DO NOT EVER CONNECT THE DC JACK AND USB AT THR SAME TIME! Doing this will cause damage to whatever you connect to the USB port!!!

As to why there is no USB communication:

On Miniware devices, it's as simple as we don't have space for the code.

On the Pinecils, it's as simple as no one has written the code for it.

Also it gets a little messy with QC as well since it also uses the USB data pins.

elamn1, 2023-01-08:
> DO NOT EVER CONNECT THE DC JACK AND USB AT THE SAME TIME! Doing this will cause damage to whatever you connect to the USB port!!!
> 

Isn't this hardware specific? Some do that, some don't, right?
Really, Pinecil V1 and V2 do that? What happens in practice then? It sounds a design flaw that not been taken into account user's possibility to put something both ports at the same time. For example, if you plug two USB-C chargers to MacBook (or any device I can think of) at the same time, it won't break the laptop. The device understand use only one of them.

> As to why there is no USB communication:

In these you mean why it is not possible to use USB-C cable with auxiliary power? (because in the previous you said that USB and DC Jack cannot be used at the same time)

> On Miniware devices, it's as simple as we don't have space for the code.

Don't they have any space left for new features/codes? Or do you assume this feature is very big to implement?

> On the Pinecils, it's as simple as no one has written the code for it.

But in all Pinecils is it possible to do this?

> Also it gets a little messy with QC as well since it also uses the USB data pins.

I understand, but I personally think that it shouldn't be a problem. It's great that we support QC, but I'd rather encourage the use of USB PD.