Ralim/IronOS #1532: What about GVDA GD300?

Kabron287, 2023-01-21:
https://www.gvda-instrument.com/soldering-station/usb-charger-65w-electric-soldering-iron-kit.html

Ralim, 2023-01-21:
Are schematics open or reverse engineered already ?
What MCU and PD interface is used?

Kabron287, 2023-01-21:
No info at the moment, but device looks very good.

Ralim, 2023-01-21:
I'm not overly impressed, it looks like it's the old style tips and no EPR support. If schematics are available and it's an easy MCU to develop for I can have a poke but can't promise anything.

Titanium-0neByte, 2023-01-23:
It would be great to see IronOS on this soldering iron 😁

filimonic, 2023-01-25:
Agree, I'd be great to see IronOS there

filimonic, 2023-01-25:
[image]
This is main MCU

Two other ICs are USB PD and mosfet
[image]

[image]

Ralim, 2023-01-25:
Yeah I have no idea what that main MCU is. Doesnt seem to have any manufacturer markings either.
Do these devices show up on usb when connected to a PC?

Titanium0B, 2023-01-26:
Apparently, it is not displayed when connected to a PC (I don’t know because it hasn’t arrived to me yet), but the microcontroller can be STM8S103 or ATtiny2313 (SOP-20) can be different, but these are more likely

BF3000, 2023-02-20:
Not a chance.
Gvda gd300 doesn't support firmware update.
There is another soldering iron for the same t65 tips called fnirsi hs-01, that one at least have the ability to update firmware.

closed 2023-05-16 by discip