Ralim/IronOS #1546: Sequre S60 Support

bradanlane, 2023-01-26:
**Support for Sequre S60**

im not sure if it is possible or not to support IronOS on the Sequre S60.

I recently added this soldering pencil to my bench as a companion to a TS80.

The S60 is shorter than the TS80/p, heats up faster, and uses JBC tips. It comes with three tips especially design for very fine soldering tasks.

From a UX perspective, it has the same OLED and 2-button interface.

What I do not know is if uses a compatible microcontroller.

If this is an unfamiliar device, I am happy to attempt a disassembly.

River-b, 2023-02-02:
do you have a link to the Squre60?  that is interesting if it does JBC  245 style tips - they have different ohms resistance. 
It would also help  if  Sequre also  provided a **schematic** for their iron.
if possible,  get them to give the current  source code to at least  Ralim  even if they don't want to share the code publicly. 
I imagine with the different  ohms on 245 tip cartridges, they will have a different table of heating profiles too.

Ralim, 2023-02-06:
To keep things open here I have emailed Sequre about the S60, still talking back and forth to see what will come of this but they are interested at the least.

Will update here when I have notable news to share.

River-b, 2023-02-06:
I just realized it uses  C115/C210 JBC tips and not the 245, according to the  Sequre  website.

Ralim, 2023-02-11:
Small update:
Sounds like we have a go-ahead from Sequre; going to be sent a sample unit and some documentation so will try and get it up and running.

>one drawaback is it seems to be 12V maximum power.

I'm guessing thats a tip rating but will be investigating. 12V@5A isnt suuper common, but I'm sure something can be figured out.

> However, there is probably some hack that could go from USB-C cable to DC battery power

Some of their other products have this exact "hack" cable you mention. I dont encourage them due to damage >_<

bradanlane, 2023-02-11:
**Re: “initial go ahead”**

This is great news. I struggle with their current firmware.

**Re: “12V”**

This does not appear to be an issue.  The significantly lower thermal mass of the tips results in the S60 heating extremely fast. The low thermal mass does have 2 downsides. (1) PID is a challenge as it is constantly trying to adapt to soldering. (2) it struggles to maintain temperature which is likely interrelated with PID.

I can not use the S60 as my “daily driver” because of the PID issues. The upside is that I happy to test alpha and unstable code.

River-b, 2023-02-21:
Maybe if Sequre could send you sample and schematics of the **S1012** or **S1012 Pro** (not sure of the difference of the Pro).  
Reason is the S1012 uses 3 kinds of tips:  both Short and long TS100 tips, and also the long T12 tip. They allow people to move the contacts to a different spot in the rear of the PCB for T12 users, and move contacts to front of the PCB for ts100 tips.

Sequire calls the  Pine64 Short Tip (ST)  an  "Si" tip.
https://www.youtube.com/watch?v=Q9HEc6_F7hU

Pine64 short tip came out in Nov 2022, and the Sequire "Si" tip appears to be similar/same ust came out Feb 15, 2023. 

The S1012 supports PD3.0  and DC barrel.  not sure why it doesn't also support PD3.1

bradanlane, 2023-02-21:
FYI: I looked at what difference - if any - exist between the “C210 compatible” tips which ship with the S60 and official JBC tips. The former are copper while the latter are stainless steel.

I expect most users will only use the tips shipped with the S60 given a single JCB tip costs more than an entire S60 kit.

That said, I’ve order a JCB tip for testing and comparison. 

> Small update: Sounds like we have a go-ahead from Sequre; going to be sent a sample unit and some documentation so will try and get it up and running.

River-b, 2023-02-21:
Photos of the inside of Sequire S60 and the PCB. the way the JBC tip is held inside is interesting.

https://community.element14.com/technologies/test-and-measurement/b/blog/posts/usb-c-soldering-iron-quick-review-sequre-s60

EgorOA, 2023-02-28:
> Sequire calls the Pine64 Short Tip (ST) an "Si" tip.

Thanx!
They sell those tips there (probably somebody needs it):
https://sequremall.com/products/si-soldering-iron-tip-antioxidant-maximum-power-100w-for-si012-si012-pro-d60-pro
https://aliexpress.com/item/1005005222272163.html

River-Mochi, 2023-02-28:
> They sell those tips there (probably somebody needs it):

Sequire charges twice as much for them as Pine Store for the Short tip.
 can get  4 short tips 26 + 12 shipping = $38  from Pine store which less than cost of a 2 sequre copy Si tips. Sequire appear to have copied the  Pine64 design but, no one has tested them yet afaik in Pine64 community. 
 we assume they might work but don't know. 
Sequire Si copies - they want nearly $19 for one tip (tip + shipping) from the Aliexpress link. their official store is harder to gauge since shipping cost is per country.

For comparison, people can check prices of short tips on official Pine Store also:
https://pine64.com/product/pinecil-soldering-short-tip-set-fine/
https://pine64.com/product/pinecil-soldering-short-tip-set-gross/
Pine64.com 4 Short tip pack + shipping  is a lot cheaper than just 2 tips from Sequire (which are not yet confirmed to work in Pinecil V2).

River-Mochi, 2023-02-28:
> They sell those tips there (probably somebody needs it):

If someone needs Short tips I recommend getting them from official Pine Store since they are cheaper and officially work in Pinecil V2.  The Sequire Si tip copy of the Pine64 short design has not been verified or tested to work in V2 by Pine64 community yet afaik.

- https://pine64.com/product/pinecil-soldering-short-tip-set-fine/
- https://pine64.com/product/pinecil-soldering-short-tip-set-gross/
Pine 4-pack short tips + shipping  26+12= $38

Sequire Si ,  just one tip from their aliexpress store + ship is ~$19.
Just two "Si" copy tips from Sequire cost more than full 4-pack Short tips from Pine Store. 
"Si" tips are not officially guaranteed by Sequire to work in Pinecil V2 nor tested yet, so be aware.

If they were $7 total price I might try to test one, but no point at $19 for just one tip, I'll stick with Pine Store Short packs.

Ralim, 2023-03-01:
@bradanlane 

Heads up that the sample S60 has arrived.
I'll start looking into support for it now; and should get something together shortly.
Sadly I'll be away for internet for the next few weeks though so if I'm silent that is why :)

Ralim, 2023-03-08:
I have spent a bit of time with the sample device.
Few teething issues with the i2c hardware in the MCU used; so I'm using software i2c for now. And will come back to hardware I2C later on.

I don't foresee any issues with the port; though the screen is a larger resolution so will need to plan for handling that. For now I've just windowed to use a subset of the screen.

Need to setup tip control and then when I'm back home in 2 weeks I'll figure out a rough temperature curve and get it somewhat near calibration.

blakadder, 2023-03-19:
I have an SI012 Pro. It does work with a 20V USB charger, I'm using it with an HP laptop charger. It can swap TS and T12 tips out of the box. I also have two of the SI tips but weren't impressed by their performance.

Soldering iron runs on an CKS32F (one of the good chinese STM32 clones) [datasheet link](https://datasheet.lcsc.com/lcsc/2005271105_CKS-CKS32F103CBT6_C556577.pdf)

River-Mochi, 2023-03-20:
> I also have two of the SI tips but weren't impressed by their performance.

Could you please measure the resistance ohms on the Si tips you got  is it between  6.1-6.5 ohms?
measure at the  2 electric contacts on the white side please after clean/wipe with alcohol.

[image]

blakadder, 2023-03-25:
I get 6-6.6 ohms

I have K (89.7mm) and I (85.3mm)

Ralim, 2023-04-25:
@bradanlane 

Very first draft test firmware (basic hardware support only).
Still has some very notable issues:

* Need to add movement detection
* Buzzer is not used
* Calibration isnt perfect
* OLED needs rework (for now its a cropped view, as have not yet updated IronOS to support dual screen layouts

[S60_EN.zip](https://github.com/Ralim/IronOS/files/11315721/S60_EN.zip)

Duckle29, 2023-05-11:
Just got my S60 in the mail, and then found this issue. Saying I'm excited would be an understatement :3

Duckle29, 2023-05-11:
@Ralim Since you're in contact with Sequre, do you think it'd be possible to offer a factory default firmware for the curious who'd like to be able to return to factory defaults?

Ralim, 2023-05-28:
I shall ask, didn't realise it wasnt on their website; apologies.

Taudris, 2023-07-23:
Any update on the stock S60 firmware? I'd love to try IronOS on mine, but I need a way to revert if it ends up not working for me.

Ralim, 2023-07-23:
Have not heard anything back from them; in the past emails where it was asked they ignored the question. I'm thinking the answer is going to be a "no" at this point I'm afraid.

damex, 2023-07-26:
could we get some preliminary builds for s60 available so everyone gets the same firmware?
just got s60 to use with ironos and need to figure out builds system >_>

Ralim, 2023-07-26:
Just go to actions tab, and look at any recent build for the Dev branch. All firmware is built on every commit and attached. :)

damex, 2023-07-26:
> Just go to actions tab, and look at any recent build for the Dev branch. All firmware is built on every commit and attached. :)

thanks, i have just tried both building myself and most recent build for dev branch.

IronOS starts but when you try to heat it up - you see temperature climb for a second (lets say up to 70-something) and then reboot.

is it supposed to be able to heat up or that behaviour with reboot is expected?

damex, 2023-07-26:
i did further testing - if it is powered by 5v - there is no heating attempt done. i can see 5v or 6v listed on main screen.

qc09 aka stock ts80p qc charger gets shipped with results in one second attempt to charge and reboot afterward. i can see 13v listed on main screen.

attempt to use xiaomi ad652g make it work just fine. it heats up and keeps temperature just fine. i can see 12v on main screen.

do i need to do further testing with different power supplies? i am happy that xiaomi ad652g works though :)

swapping various usb-c to usb-c cables make no difference here.

ia, 2023-07-27:
Hello.

> if it is powered by 5v - there is no heating attempt done.

5v is less than enough to provide enough of power for [proper & comfortable] soldering using those modern smart soldering irons. So that is ok.

> Using qc09 aka ts80p qc charger it ships from ali results in one second attempt to heat up and reboot afterward

Now that is weird! According to the specs of `QC09`, it should be suitable for `S60`. As of for reboot, it _probably_ _could be_ some power negotiation problem between hardware of the charger (due to its design) & hardware of the iron. These are nasty problems hard to debug _usually_.

> do i need to do further testing with different power supplies? i am happy that xiaomi ad652g works though :)

Since you have a working solution, I would just stick to that.

bradanlane, 2023-07-27:
> 5v is less than enough to provide enough of power for [proper & comfortable] soldering using those modern smart soldering irons. So that is ok.
> 

As a point of reference, the S60 factory firmware supports 5V and is quite usable. 25C to 300C takes less than 8 seconds at 5V.

5V support is very handy when all that is available is 5V.

ia, 2023-07-27:
> As a point of reference, the S60 factory firmware supports 5V and is quite usable.

I forgot about amps. So if it's with a proper charger up to 3 amps of current then yes, it could work. But as far as my humble understanding is, it's still _better_ (i.e., _less_ _"stressful"_ for hardware & wires) to have more volts & less amps to get more wattage than vice versa.

> 25C to 300C takes less than 8 seconds at 5V.

When it comes to proper & comfortable soldering, it's not about how much seconds it takes to temperature T but how well temp T can be sustained when the iron giving the heat away to solder for melting & to soldering joint for _sticking_, so the key is _proper_ heat dispensing under _good_ power control, not marketing numbers ;)
I don't say that the numbers is not true, but usually they used more like for marketing by manufacturers.

Don't get me wrong, in impatient situations I really like myself how fast modern irons can be pre-heated but that's not the main factor for _enjoyable_ soldering. I don't try to be nerdy, rude or stubborn, just sharing my though on the topic although it's getting a bit off-topic, sorry. :)

bradanlane, 2023-07-27:
My comment is from actual usage _(not marketing material)_. I have the S60. (I also have a TS80 and TS80P). 

I have switched to the S60 almost exclusively. It is completely usable at 5V. Yes, it is better at 12V, but it has no problem working at 5V with a generic 12W USB battery. This usecase is extremely convenient.

Danimal4326, 2023-08-04:
S60P is now available with 21V and PPS power supply support:
https://sequremall.com/collections/soldering-irons/products/sequer-s60p-upgraded-version-electric-soldering-iron-supports-pd-qc-dc-pps-power-supply-compatible-with-c210-soldering-iron-tip-precision-electronic-mobile-phone-repair-tool-anti-static-soldering-pen-60w?variant=42851400679612

damex, 2023-08-04:
I wonder if s60p gonna get same treatment as s60

they also started heavily discounting original s60

mik1011984, 2023-08-21:
I'll have a S60P tomorrow for testing. Fingers crossed that it works reasonably well as I've sold my SI02. It was either the S60P using C210 tips or a SI012 Max. I hope I made the right choice. 
They've also since announced the S99 with C245 style tips. The products just keep coming.

Yuriy-RU, 2023-09-01:
https://sequremall.com/blogs/sequre-product-firmware-upgrade

Taudris, 2023-09-13:
I have some interesting news. Sequre's updated firmware (1.15.1 and 1.15.2 - they seem to be the same except for the supported languages) allows my S60 to negotiate 20V via USB-PD. Sometimes the iron will work fine, but other times it will immediately restart after heating the tip a very small amount (never enough to be uncomfortable to the touch). Whether it works seems to vary by tip and power limit.

Lenovo 45W, Chisel (K) tip
* Auto: Works
* 100%: Restart
* 70%: Restart
* 65%: Works sometimes (possibly based on starting temp)
* 60%: Works sometimes (possibly based on starting temp)
* 50%: Works
* 20%: Works

Lenovo 45W, Pointy angled (IS) tip
* Auto: Restart
* 20%: Restart

Baseus 60W, Chisel (K) tip
* Auto: Restart
* 20%: Restart

Baseus 60W, Pointy angled (IS) tip
* Auto: Restart
* 20%: Restart

damex, 2023-11-24:
any idea how identical s60 and s60p ? i wonder if their firmwares are interchangeable

discip, 2023-11-24:
@damex
Maybe this could help: https://github.com/Ralim/IronOS/discussions/1806