Ralim/IronOS #1547: screen is "messed up" with the detailed soldering screen

TomW1605, 2023-01-27:
**Describe the bug**
when the Pinecil goes to sleep after the timeout the sleeping screen is "messed up" and still shows the temp
[image]

**To Reproduce**
let the pinecil go to sleep while soldering with the detailed soldering screen

**Expected behavior**

should just say "Sleeping"

**Details of your device:**
 
 - Device: Pinecil V2
 - Release: 2.20.07D85C8
 - Power adapter being used: Anker Nano 2 with pine64 c-c cable

**Additional context**

for me this has reliably happened every time it sleeps. i have tried with the sleep timeout of 10 to 60secs. it goes away when you move the pinecil to wake it up

possibly related to #1417 

[image]

TomW1605, 2023-01-27:
didnt think to test this but it is not an issue when using the standard soldering screen

discip, 2023-01-27:
Thank you for pointing that out.
Will fix. 😊

PR #1548 (2023-01-30): fixing languages & reverting unwanted weblate changes
fixes #1547
closes #1544
closed 2023-01-30 by Ralim