Ralim/IronOS #1554: Docker development setup is not working

builder555, 2023-02-03:
**Describe the bug**
When starting the docker container, the following error is displayed:

```
#0 3.273 Installing collected packages: mypy-extensions, bdflib, platformdirs, pathspec, packaging, click, black
#0 3.377   Attempting uninstall: packaging
#0 3.379     Found existing installation: packaging 21.3
#0 3.387 ERROR: Cannot uninstall 'packaging'. It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall.
------
failed to solve: executor failed running [/bin/sh -c python3 -m pip install bdflib black]: exit code: 1
```

**To Reproduce**
1. Clone repo
2. run the `start_dev.sh` file

**Expected behavior**
Container starts

Ralim, 2023-02-03:
Eugh why is python a mess.
Having a look.

builder555, 2023-02-03:
It seems `black` library doesn't want to be installed outside of venv on alpine. Just tried using `debian` base image - it works no problem. @Ralim

Ralim, 2023-02-03:
Yeah moving black to the alpine package now.
`debian` moved so slowly that it didnt have riscv compilers for far too long so wasnt an option at the time.

commit e19e1a06 (2023-02-03): Use alpine packaged black

Closes #1554
closed 2023-02-03 by Ralim