Ralim/IronOS #1558: Using 6 ohm tips with Pinecil v1?

EgorOA, 2023-02-05:
Trying to use v2 short 6 omh tips with pinecil v2.
1.  Some 65W PD sources are rebooting. It's pitty need to use another one.
2. Temperature is changing +-10 degrees. It seems that PID parameters for temp regulation is more suitable for 8 ohm tips than 6 ohm.

Understanding, that v1 cant measure tip resistance, can be done tip type selection in menu?

River-b, 2023-02-14:
a manual menu selection for V1 older pinecil is being worked on.  this would potentialy allow people to manually select if they are using normal longer 8.0 ohm tips or newer  shorter  6.2 ohm tips - which then changes which profile it will use for power and temperature.

the V2 newer iron has different hardware and does not need the manual selection, it is capable of autodetecting tips that are closer to 6 ohms or closer to 8 ohms on each reboot (hence always unplug before swapping tips). 

I am currently kind of able to test short tips on my V1 even without the update but need to be sure to use a 20V 3.25amp 65W  or higher charger. Weird issues happen if trying it with too low a power supply since V1 is not aware there is a 6.2 ohm tip inserted. It is currently best to just use short tips in the V2 newer iron if  werid behavior is happening since the V1 things all tips are 8.0 ohms it is working based on that profile.

See this post for some details on V1 versus V2 and use of Short tips.  https://www.reddit.com/r/PINE64official/comments/wpbla4/pinecil_firmware_upgrades_new_short_tips/

EgorOA, 2023-02-15:
> the V2 newer iron has different hardware and does not need the manual selection, it is capable of autodetecting tips that are closer to 6 ohms or closer to 8 ohms on each reboot (hence always unplug before swapping tips).

It's very interesting but I can't see any difference in tip binder circuits in schematics V1 and V2 
https://wiki.pine64.org/wiki/Pinecil#Pinecil_V1_mainboard_schematic
https://wiki.pine64.org/wiki/Pinecil#Pinecil_V2_mainboard_schematic:

And I can't understand how V2 can autodetect tip type. Help me guess it, please

River-b, 2023-02-16:
have to ask @Ralim  where he implemented it in V2 code for autodetect and select correct profile for 6.2 ohms versus 8.0 ohm tips.   the V2 uses a Bouffalo BL706  MCU that is part of the difference with  the older V1. 

Ralim mentioned before that  the resistance of the tip on boot-up is very noisy.  Therefore, the firmware is just acting as a toggle in V2.  If it's closer to 6.2 ohms, then it's picking Short tip profile. If it's closer to 8.0 ohms, then it's picking the 8.0 profile. it's not giving an exact profile based on every gradation of tip.  Example some of my tips are 7.8ohm or 8.2 ohms - they will all get the 8.0 profile.

EgorOA, 2023-02-16:
> have to ask @Ralim where he implemented it in V2 code for autodetect
I have caught the difference - a little circuit (addon with 2 resistors and diod) that allows to pull up tip by GPIO24 and measure tip resistance. Pitty but V1 has no such circuit so we have to select tip type manually(

River-Mochi, 2023-03-23:
@EgorOA   I am curious, these same  PD65  20V  chargers,   on the V1,  do they work fine with no rebooting if you use the longer normal cartridge/tips?

EgorOA, 2023-03-24:
> do they work fine with no rebooting if you use the longer normal cartridge/tips?

Yes

jayjr1105, 2023-05-19:
@Ralim I would be willing to be an early beta tester for 6ohm tips on V1 hardware.  LMK!  thanks!

jakobi, 2023-08-17:
I finally received a near-unobtainium "ts200 style tip" via banggood with a cold resistance of 4.2ohm (unobtainium at least outside the US...).

Seeing no difference in a pinecil v2 between that tip and a pinecil short with 6.5 ohm, I got curious:

The below are results of using the tips in a pinecil v2 fw 1.21. As I removed to tip to measure it,  multimeter results of resistance  are a rough estimate only due to tip cooling, as well as the pinecil's own max power during heatup.

| tip style         | 24v     | PD    | 25-40c | ~200-300c | ~400c |
| ---                    | ---       | ---      | --- | --- | --- |
| hakko            |  60W  | ~45   |8.2 | 10 | 13 |
| pinecilshort  |  77W  | 55W  |6.4 | 9  | 11  |                  
| ts200              | 72W  | 55W  |4.5 | 6  | 9.5 |

Still, given the resistance values of that ts200 tip, I expected to see higher peak power. pinecil v2 thinks it to be a 6.2 ohm tip. Some kind of (PD negotiation and) PWM throttling based on the assumed tip resistance?

On hot insertion of tips with 24V, the pinecil refused to heat (or it tried with 0W), I had to unplug it on tip changing. Nice surprise, when trying the ts200 tip for the first time in the pinecil at 24V.

Retinning afterwards was done in a pinecil v1 @ 240c on usb PD, and even the ts200 played nice there.

At 450c in a v2, the ts200 tip temperature display was way more jumpy than with the short pinecil tips, and tended to list 20w power use. 

As for being able to use hakko tips for this test: I use the rubber grips from a hakko station and hold that rather than the pinecil itself, so tip distance from hand is normal. Inside the pinecil I  duct-taped a mold and filled it with epoxy to make a depth stop.

Ralim, 2023-08-25:
>Still, given the resistance values of that pts200 tip, I expected to see higher peak power. pinecil v2 thinks it to be a 6.2 ohm tip. Some kind of (PD negotiation and) PWM throttling based on the assumed tip resistance?

So long as your power supply is capable of what a 6.2 ohm tip would pull, then there is no PWM limiting going on.
So, so long as you used a >=60W charger there isnt any limiter going on.

>On hot insertion of tips with 24V, the pinecil refused to heat (or it tried with 0W): I had to unplug it on tip changing. Nice surprise, when trying the pts200 tip for the first time in the pinecil at 24V. I've never seen that behaviour before (at least not on PD).

The code will block until it can reliably determine the resistance of the tip by checking the delta via the pullup resistor.
When you are hot sometimes it will take a while for that to measure correctly. usually cycling off, letting it cool a bit and starting again fixes this. Since we are trying to measure a really small signal; once the tip or body is hot it gets dramatically harder to use. This is why the official advise is not to hot swap tips, or if you must, only to do it in the same series (i.e. only 6 ohms, or only 8 ohms).

pwnept, 2024-03-05:
As of firmware V2.22, is there a toggle for Pinecil V1 to select between long and short tips?

discip, 2024-03-05:
As far as I know, there isn't.

mrambossek, 2024-07-18:
any updates on this?

Jojo-1000, 2024-11-13:
It seems that there is an implementation merged in the development branch recently, so hopefully the next release will support it:
#1977

If someone tests a dev build, please update on how well it works. I am undecided on which tips I'm going to get for my v1 in the future.

Ralim, 2024-12-07:
This should be fixed with latest dev builds, I'm hopeful for some testing thats not me 😁 
Performance should be slightly better thermally, but mostly they improve ergonomics.

jayjr1105, 2024-12-07:
What are the requirements to use the 6ohm tips with a V1 other than the latest dev build?  PD power supply minimum or barrel jack minimum?  I tried the dev build shortly and still had some instability with the iron stopping heating with a short tip installed.

EnekoT2001, 2024-12-08:
> This should be fixed with latest dev builds, I'm hopeful for some testing thats not me 😁 Performance should be slightly better thermally, but mostly they improve ergonomics.

just tested it on mine, seems to work fine so far, thank you!

propell-ant, 2024-12-10:
> What are the requirements to use the 6ohm tips with a V1 other than the latest dev build? PD power supply minimum or barrel jack minimum? I tried the dev build shortly and still had some instability with the iron stopping heating with a short tip installed.

@Ralim, may the instability come from the different tip thermal characteristics assumed?
Pinecil v2 (which uses the short 6.5 Ohms ones) has `getTipThermalMass() { return 120; }` [here](https://github.com/Ralim/IronOS/blob/19eb1c8a4625a440b568646dfee6b1585f106396/source/Core/BSP/Pinecilv2/BSP.cpp#L163),
whilst the Pinecil has `getTipThermalMass() { return TIP_THERMAL_MASS; }` and `TIP_THERMAL_MASS 65` [defined here](https://github.com/Ralim/IronOS/blob/19eb1c8a4625a440b568646dfee6b1585f106396/source/Core/BSP/Pinecil/configuration.h#L168).
E.i. short tip may have much slower response time for the given Joules applied, which is unexpected by PID regulator configured for normal tip.

Ralim, 2025-03-04:
Yeah the Pinecilv2 uses PID mode control, while the V1 uses thermal mass control.

If someone is keen to do some testing I can throw together a test build for the V1 using the V2's tuning?

propell-ant, 2025-03-28:
Well, I can perform the test with V1, but need some time to order 6.5 Ohms tip. It'll take a couple of weeks to arrive.

propell-ant, 2025-04-15:
Pinecil v1 using firmware v.2.22 seems to work well with the 7 Ohms tip I've bought. Probably the tip "isn't counterfeit enough" to expose the unstable behaviour.
I'm ready to perform the tests now.