Ralim/IronOS #1559: une notify for real time BLE data

TomW1605, 2023-02-05:
**Describe the solution you'd like**

curently you need to poll data from BLE however BLE provids functions for the server to notify the client of changes in data. this would prevent spaming data requests and provide more realtime data

**Additional context**

i have started working on this and got a basic proof of concept working using the bulk data polling callback to trigger a notify on the temp. this worked however given it still relies on poling dosnt really help. unfortunatly i dont know/understand enough about the program flow of the code to work out where best to put the notify callback.

here is what i have so far
https://github.com/TomW1605/IronOS/commit/330761cbde81234337f49f435f186d8e60f1f8a5

i do intend to keep working on this however i think it will either take a long time or i will need help from someone with a better understanding of the code

TomW1605, 2023-02-05:
there is one option that uses notify to make more eficent use of data while still triggering with polling. if you have a spesific BLE characteristic that is polled it can trigger a notify for any values that have changed while not sending ones that havnt. im not sure how much this would help to make it worth it for the possible issues of storing the last value but it might help.

Particpant, 2024-01-27:
Oh, this would be really helpful.

Ralim, 2024-12-07:
@TomW1605 
Hia, and sorry for this being punted down the road a few times.
Where did you get to with this on your fork?

TomW1605, 2024-12-07:
honestly havnt looked at it in months, life got busy and i never got back to it. not sure where i got up to and my fork is so out of date at this point im not sure how useful what i did do will be. unfortuantly life is still pretty busy so i dont really have time to try and update things to work again but if someone else wants to take over i have made sure all my changes are in my fork and will happaly answer any questions as best i can