Ralim/IronOS #1563: TS80P Shutdown (crash)

ecard0, 2023-02-06:
**Describe the bug**

I upgraded from 2.16.

With the latest version, when the TS80P is in standby mode (idle), it will shutdown after a couple of seconds.  It happens only when I run it from a power bank 22W. 

The problem is gone after I reverted back to my older version 2.16

**To Reproduce**

**Expected behavior**

**Details of your device:**
 
 - Device: TS80P
 - Release: 2.20
 - Power adapter being used:  Power bank USB-C 9V 22W
 - If this is an accelerometer related issue, please include its model number here:

**Additional context**

Ralim, 2023-02-06:
Is the device using PD or QC protocol when talking to the power bank? (Check the debug menu)
If its using QC you may need to increase the settings for the keep-alive power pulse to stop the bank sleeping.

ecard0, 2023-02-08:
I played with QC settings and not working.  I reverted back to 2.16

ecard0, 2023-02-08:
Hi Ben, it's in QC. Do you have settings to recommend me to test?

Ralim, 2023-02-08:
I would suggest:

Power Pulse 3
Delay 3
Duration 1

This should mean that every 4 seconds; for one second it will pulse around 3W into the tip.

If that still doesnt keep it awake try 5/2/1 possibly

ecard0, 2023-02-08:
I tried 3/3/1, 5/2/1...  same issue :(

Ralim, 2023-02-08:
I'm very unsure then; and would really want to see what is going on with the voltages on the usb link during this.
Suspect its just a quirk of the power bank though.

ecard0, 2023-02-08:
I'm also getting the same results with another power bank (totally different brand and size).

But, everything is A1 with 2.16.  Should I test versions between 2.16 and 2.20

ecard0, 2023-02-09:
I did a quick test, I installed 2.19 and it's running perfectly.  Must be a change that was introduce in 2.20.

discip, 2023-02-15:
@ecard0
Are you running the latest official release?

Please try this: [TS80P.zip](https://github.com/Ralim/IronOS/suites/10924657369/artifacts/552314240)

ecard0, 2023-02-18:
Is this version different than the official 2.20 ?   It has the same problem.

Back to 2.19

Ralim, 2023-02-19:
Can you capture what is occurring on the voltages?
2.19 vs 2.20 had a lot of changes; so would be ideal if we could _see_ the issue.

ace-dent, 2023-03-30:
I was also having issues with my power bank(s) switching off unexpectedly...
Here's some test results in case it's useful:

---

Versions tested:
v2.18.70565DE 18-04-22 on TS80P
v2.19.A3BBABC 13-07-22 on TS80P
v2.20.14DAF70 13-12-22 on TS80P

Advanced settings (as default):
- Power pulse 0.5W
- Power pulse delay 4 x2.5s
- Power pulse duration 1 x250ms

---

(1a) USB-C to USB-C cable > into power bank A > USB-C socket.
  &emsp; 2.18: Vin 9.1V ’PWR PD No VBus’ — Time 3000+ √
  &emsp; 2.19: Vin 9.1V ’PWR PD No VBus’ — Time 3000+ √
  &emsp; 2.20: Vin 9.1V ’PWR PD No VBus’ — Time 3000+ √

(1b) USB-C to USB-C cable > into power bank B > USB-C socket.
  &emsp; 2.18: Vin 9.2V ’PWR PD No VBus’ — Time 3000+ √
  &emsp; 2.19: Vin 9.2V ’PWR PD No VBus’ — Time 3000+ √
  &emsp; **2.20: Vin 10.1/ 10.0V ’PWR PD No VBus’ — FAIL: Times out. X**

(2a) USB-C to USB-A cable > USB-A to USB-C adaptor > into power bank A > USB-C socket.
  &emsp; 2.18: Vin 8.9V (negotiated from 5.1V initially) ’PWR PD No VBus’ / flashes ‘PWR QC’ — Time 3000+ √
  &emsp; 2.19: Vin 8.9V (negotiated from 5.1V initially) ’PWR PD No VBus’ / flashes ‘PWR QC’ — Time 3000+ √
  &emsp; 2.20: Vin 9.1V (negotiated from 5.1V initially) ’PWR PD No VBus’ / flashes ‘PWR QC’ — Time 3000+ √

(2b) USB-C to USB-A cable > USB-A to USB-C adaptor > into power bank B > USB-C socket.
  &emsp; 2.18: Vin 9.0V (negotiated from 5.1V initially) ’PWR PD No VBus’ / flashes ‘PWR QC’ — FAIL: Times out. X
  &emsp; 2.19: Vin 9.0V (negotiated from 5.1V initially) ’PWR PD No VBus’ / flashes ‘PWR QC’ — FAIL: Times out. X
  &emsp; 2.20: Vin 9.0V (negotiated from 5.1V initially) ’PWR PD No VBus’ / flashes ‘PWR QC’ — FAIL: Times out. X

(3a) USB-C to USB-A cable > into power bank A > standard USB-A socket.
  &emsp; 2.18: Vin 5.1V ‘PWR PD No VBus’ / flashes ‘PWR QC’ — Time 3000+ √
  &emsp; **2.19: Vin 5.1V ‘PWR PD No VBus’ / flashes ‘PWR QC’ — FAIL: Times out. X**
  &emsp; **2.20: Vin 5.1V ‘PWR PD No VBus’ / flashes ‘PWR QC’ — FAIL: Times out. X**

(3b) USB-C to USB-A cable > into power bank B > standard USB-A socket.
  &emsp; 2.18: Vin 5.1V ‘PWR PD No VBus’ / flashes ‘PWR QC’ — FAIL: Times out. X
  &emsp; 2.19: Vin 5.1V ‘PWR PD No VBus’ / flashes ‘PWR QC’ — FAIL: Times out. X
  &emsp; 2.20: Vin 5.1V ‘PWR PD No VBus’ / flashes ‘PWR QC’ — FAIL: Times out. X

(4a) USB-C to USB-A cable > into power bank A > QC3.0 USB-A socket.
  &emsp; 2.18: Vin 8.9V (negotiated from 5.1V initially) ’PWR PD No VBus’ / flashes ‘PWR QC’ — FAIL: Times out. X
  &emsp; 2.19: Vin 8.9V (negotiated from 5.1V initially) ’PWR PD No VBus’ / flashes ‘PWR QC’ — FAIL: Times out. X
  &emsp; **2.20: Vin 8.9V (negotiated from 5.1V initially) ’PWR PD No VBus’ / flashes ‘PWR QC’ — Time 3000+ √**

(4b) USB-C to USB-A cable > into power bank B > QC3.0 USB-A socket.
  &emsp; 2.18: Vin 9.0V (negotiated from 5.1V initially) ’PWR PD No VBus’ / flashes ‘PWR QC — FAIL: Times out. X
  &emsp; 2.19: Vin 9.0V (negotiated from 5.1V initially) ’PWR PD No VBus’ / flashes ‘PWR QC — FAIL: Times out. X
  &emsp; 2.20: Vin 9.0V (negotiated from 5.1V initially) ’PWR PD No VBus’ / flashes ‘PWR QC — FAIL: Times out. X

discip, 2023-06-29:
@Ralim
Have you looked into this yet?