Ralim/IronOS #1564: Is there any way to pull more than 30W on the TS80P?

404-Disbanded, 2023-02-06:
The soldering iron should be capable of holding at least 40W (25% above limit). I'm just wondering what component (either hardware or software) is enforcing the limit. I set the power limit to unlimited however there was no change and no matter what I do I can't get it to pull more than that. I need at least a few more watts in order to properly be able to solder on this pcb with a heat spreader on the back. It just doesn't get hot enough to melt the factory solder and the TS80P is the only soldering iron I have... Any ideas?

Ralim, 2023-02-06:
Your power is limited by the tip resistance.
Two fundamental electronics equations of `Volts = Amps * Resistance` and `Power = Voltage * Current`

The TS80P is rated to 12V; and uses a 4.5 ohm tip.
Therefore 12V/4.5 Ohms = 2.66A; 12V * 2.66A = 32W.
The maximum power the TS80P's tip is delivering is 32W.

Realistically since we need to measure the temperature it averages out around 30W ish. (You can either heat tip or read temperature; not both at once).

The only way you could get more power into it is by having it negotiate up at 15V; but that will both increase voltage (and may break things) it will also increase your current to 3.3A which may also break things.

Generally it is better to use the right tool for the job sadly.

404-Disbanded, 2023-02-07:
Quick addendum here, my iron defaults to QC even when on a PD brick that supports both however it never negotiates a voltage that isn't 12.2 v even when the QC voltage is limited to 9 which is rather odd. I was under the impression that the setting forced a certain QC voltage and if it was capable of 14 V (Which is the setting max). At 13.6 volts under the current resistance we yield an amount at 3.02 amps or about 40.02 watts. Theoretically QC 3 maxes out at 36 watts but it can be pushed further as the limit isn't solid and either way 36 is better than 30-32.

If I could somehow force PD negotiation to 15v I agree that 3.3A would be rather concerning so by dropping a of 1 ohm resistance in series with the element it should be possible to drop the current down to 2.72Amps if I'm not mistaken which should be safer than leaving the current at such high numbers.

I determined that 40 watts should be okay within the context of this soldering iron by determining the waste heat generated through systemic loss from input to output and then scaling that by the increase in amperes but I realize my oversight here in terms of frying the chip. At 13.6 volts the systems (end to end) should be fine even compensating for the increase in amps however I can't get to that point through settings and I'm not entirely sure where to start with direct DC injection.

builder555, 2023-02-07:
@404-Disbanded did you try changing PD Timeout setting? I have a QC3 charger that delivers 12V, with PD Timeout at 20 (2s) i can limit it to 9V no problem, with lower settings sometimes it fails and jumps to 12.

Ralim, 2023-02-08:
>Quick addendum here, my iron defaults to QC even when on a PD brick that supports both however it never negotiates a voltage that isn't 12.2 v even when the QC voltage is limited to 9 which is rather odd.

If its always going to 12.2V; are you sure its not using PD (Checking debug menu or watching the data transfer)?

The QC voltage sets what voltage it will attempt to seek to (via QC3.0); if you do not have QC3.0 it wont be able to seek the voltage.

The PD timeout can be set longer to allow more time for slow PD stacks, or very small to force skipping PD.

> I determined that 40 watts should be okay within the context of this soldering iron by determining the waste heat generated through systemic loss from input to output and then scaling that by the [...]

I would go through the TS80 schematic (since its basically same as TS80P) and validate all components can take current and voltage. I dont have ratings for the tips or their connector though.

For direct DC, just inject it down the power pins on the USB-C. So long as injected power is >=9V it will not do any negotiation.

404-Disbanded, 2023-02-08:
Here check this out

https://user-images.githubusercontent.com/124635745/217413636-d785af55-de35-4337-acd7-b4e02af71692.mp4

sorry for the resolution and banding... only way I could get it under 10 mb

404-Disbanded, 2023-02-08:
On a slightly separate note any clue how the resistance of the circuit varies with temperature, I couldn't find a spec-sheet and my readings are rather erratic in addition to not being able to trust my voltage reading as outlined above.

River-b, 2023-02-14:
@404-Disbanded   Enter the PD  debug menu by holding down the front button before plugging in the cable. keep holding the button another 10 seconds then release. (front button is closer to the tip).  then use this same front button to scroll through all PD messages. this tells if it is truly using QC3 or PD. show video of  all the messages in PD  debug menu.

note that IronOS only supports QC 3.0  and not QC 2.0 so if the charger is qc2.0 it will only be using the PD  of it. 
If you see message  STATE 12  - this is a PD connection.

404-Disbanded, 2023-03-23:
Right so it definitely doesn't say STATE 12 and on a separate note it took 14v with no problem at all except it dropped a series of warnings and then turned itself off.

404-Disbanded, 2023-03-23:
sorry about the open close I misclicked

River-Mochi, 2023-03-23:
> Right so it definitely doesn't say STATE 12 and on a separate note it took 14v with no problem at all except it dropped a series of warnings and then turned itself off. So thanks for all the help.

if it doesn't say  state 12  then you are not connected to a PD  type charger and it's using lower QC  voltage which you have to set manually.
since there is no issue here, you  could close  this ticket or move it to Discussions  since it's not actually a problem with the device.

closed 2023-05-16 by discip