Ralim/IronOS #1566: Expose   "Reset All Defaults"  for BLE usage

River-b, 2023-02-07:
**Is your feature request related to a problem? Please describe.**
For Builder555  effort  for BLE  settings changer for Pinecil V2  we  can not currently  use the BLE  to  select "Reset All to Defaults"   as builder555 notes in the below ticket,  that this is not exposed or they can't find it if it is.

https://github.com/builder555/pinecil-v2/issues/3#issuecomment-1419979263
**Describe the solution you'd like**
Please expose  this  from the Advanced Menu > reset all to defaults.

builder555, 2023-02-10:
@Ralim can you let me know the ID of the characteristic(setting) for resetting to factory, and can I assume it'll require a `1` written to it? 
i'll add it to a branch, so it'll be ready when you release 2.21

Ralim, 2023-02-11:
@builder555 

Put up my first ideas in #1567 if you want to have a look.

builder555, 2023-02-11:
@Ralim i see all the settings shifted by 1, that was an exciting 15 seconds 😅

Ralim, 2023-02-12:
I did note it in the PR that I set them back to being `0` based.
I think thats a better way to do it though; since then setting `0` on the firmware is also setting `0` on the BLE interface.

closed 2023-02-12 by Ralim

builder555, 2023-02-12:
definitely better for consistency. i made a provision for backwards compatibility anyway, so it won't be an issue

Ralim, 2023-02-12:
Fantastic; thank you :)

PR #1567 (2023-02-12): BLE Settings rename + Unique Name

* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [] There are no breaking changes

* **What kind of change does this PR introduce?**
* Device will advertise as Pinecil-XXXXX where XXXX is random number.
* Adds option to reset settings
* Reshuffles UUID's for settings so that they make more sense for future

* **What is the current behavior?**
Would close #1566 
Would close  #1545

* **What is the new behavior (if this is a feature change)?**
This would break the current webapp for settings editing as it cha [... cut]