Ralim/IronOS #1570: MHP30 - Idle bug

docgalaxyblock, 2023-02-15:
*A video shows more then 100 words can describe*
https://user-images.githubusercontent.com/21227087/219048616-b247806f-5609-4847-b620-61f5d1a5854c.mp4

two issues:
- **A)** display goes of, starts flashing green, no heating
- **B)** at the first time it did not let me into heating mode again

**To Reproduce**
Just leave it on
  - idle at the menu (I guess it is an idle bug since it mostly happens after exactly 1min here)
  - heating *(for the video test at 40/50°C)* for a while
sometimes it happens after eg 2min sometimes after eg 13min

- MHP30 - 2.20.14DAF70
- Power adapter: QC92 *(comes with the NovelLive full kit on Ali)*
- *Not sure if this is an accelerometer related issue:* SC7A20

Tried both DE and EN with reset

Ralim, 2023-02-18:
You need to use a PD power adaptor for full functionality; heating on 5V just doesnt work and causes brownouts that cause mis-behaviour of the unit.

docgalaxyblock, 2023-02-18:
@Ralim I used 5V only for the video

> Power adapter: QC92 (comes with the NovelLive full kit on Ali)
That is the PSU I have and it happend while I want to solder a pcb

closed 2023-04-25 by discip