Ralim/IronOS #1573: MHP30 doesn't support a battery?

violavon, 2023-02-17:
MHP30 doesn't support a battery? It's USB device, how come it doesn't support a battery? What is the difference to others who support a battery?

What is EPR? Why doesn't it support it?

From README: 

Miniware MHP30
|   Device   | DC  | QC  | PD  | EPR | BLE | Battery | Recommended |
| :--------: | :-: | :-: | :-: | :-: | :-: | :-----: | :---------: |
|   MHP30    | ❌  | ❌  | ✔️  | ❌  | ❌  |   ❌    |     ✔️      |

Others
|   Device   | DC  | QC  | PD  | EPR | BLE | Battery | Recommended |
| :--------: | :-: | :-: | :-: | :-: | :-: | :-----: | :---------: |
| Pinecil V1 | ✔️  | ✔️  | ✔️  | ❌  | ❌  |   ✔️    |     ✔️      |
| Pinecil V2 | ✔️  | ✔️  | ✔️  | ✔️  | ✔️  |   ✔️    |     ✔️      |
|   TS80P    | ❌  | ✔️  | ✔️  | ❌  | ❌  |   ✔️    |     ✔️      |
|   TS100    | ✔️  | ❌  | ❌  | ❌  | ❌  |   ✔️    |     ❌      |
|    TS80    | ❌  | ✔️  | ❌  | ❌  | ❌  |   ✔️    |     ❌      |

Ralim, 2023-02-18:
Battery support means that settings for directly connecting batteries  & powerbanks are enabled.
These are not added in the MHP30 builds as they hardware is designed only for USB-PD.

EPR is USB-PD EPR; which stands for Extended Power Range. This is the extra mode that allows > 20V on USB-C for up to 240W of power. (Noting none of the devices can actually do 240W; the Pinecil V2 goes up to 140W max rating).

violavon, 2023-02-18:
Directly connect? I know MHP30 doesn't have DC jack, only USB-C PD. But it's the same with TS80/TS80P, which also have only USB-C PD/PD+QC and no DC jack. So what is the difference?

And MHP30/TS80/TS80P support USB-C PD (or QC) powerbanks, right?

River-b, 2023-02-20:
Technically from USB-IF   PD3.1 is EPR too (extended power delivery). and it's easier to say EPR than pd3.1

 it would be   PD3.0  for the older 20V limited USB-C chargers and  PD3.1 for the new 28V USB-C 140W chargers.
Google  USB-C  PD3.1 EPR  (tons of articles on this). 
Chargers that support it  only came out in 2022.  Apple was the first to market with one as the mackbook 2021 uses it.  Although the 2022  EPR chargers are 28V, 140W,  the Protocol PD3.1 goes up to 48V - 240W.
And  the Cables that are PD3.1  already are future proof and meet the 240W - 48V standard.
As  long as you don't get a fake cable,  there are USB-C cables  available now that are PD3.1  capable of up to 240W-48V-5Amp.   I own  at least 6-8 of them already.
usb-c 240w cable basically  replaces  PD100w cables.  they both come with a special Emark chip in them too.  
This is a very long and detailed topic so highly recommend   google  PD3.1 EPR and USB-IF. 

It's also a confusing topic because of the bad  naming that the USB-IF chose for this new protocol.  I think we just say  EPR because it's less confusing than saying PD3.1  which looks too close  to the old 20V PD3.0.

River-b, 2023-02-20:
if the Powerbank supports PD connection then it's under category of PD since it's simulating the same as a PD  electric charger.
Battery would be like a Lipo 5S battery or an 18V Ryobi Tool battery (or other brands in this voltage range).  
If you look here for example the Pinecil iron can connect directly to a tool battery and work very fast and efficiently on the  18-21V tool battery in the photo. 
https://wiki.pine64.org/wiki/Pinecil_Power_Supplies#DC_Barrel_Power

violavon, 2023-02-20:
Off-topic: what USB cables you use?

@River-b wrote:
> I own at least 6-8 of them already.

Ralim, 2023-02-20:
MHP30 needs 65W power bank for it to work correctly.
A 45W output one sort of works but it can barely heat the board.

violavon, 2023-02-20:
Do you mean 60W? So the requirements are exactly the same as for power bricks, great.

Can you answer the rest of the questions?

Ralim, 2023-02-20:
I suggest 65W wherever possible, 60W is okay but you are running reduced. And for this application the 60W heating element is really barely enough with non trivial boards.

The main difference is that the MHP30 is not recommended to be used on battery packs; and its firmware only supports PD, it doesn't have QC or any other protocol. 

As such it is fairly aligned to be used with wall connected power adaptora

River-b, 2023-02-20:
@violavon I will post something later in Discussions about USB-C cables when I have time.
Perhaps these kind of esoteric questions could be started in **Discussions** and not Issue ticket  or move over to  Discussion.

River-b, 2023-02-20:
> Do you mean 60W

typically the USB-C power banks I see on amazon say PD65W.  search for "laptop power bank USB-C PD 65w"  will find several.  
many PD3.0  chargers have the rating  20V, **3.25** amps, 65W  - which is how they get 65W, but in actuality you don't get full 65.

violavon, 2023-02-20:
Ok, so you confirm my point that the TS80 or TS80P don't support directly connecting batteries? Then README doesn't make sense. TS80 only supports USB-C/QC. What would make it better with batteries than the MHP30 with USB-C/PD? MHP30/PD supports more batteries (i.e. powerbanks) than the TS80/QC!

violavon, 2023-02-20:
@River-b Yeah, the USB-C cable question was a bit off-topic.

River-b, 2023-02-20:
ralim, maybe there is a mistake in the README documents for the TS80 that only has a USB-C connector? 
I don't have a TS80 or TS80P so I only can confirm  what I know about other irons I own. 
Pinecil V2 is the most versatile of all of them and takes all the categories including the newest EPR and can run at 28V usb-C.

River-b, 2023-02-20:
perhaps person who did check box for TS80 on battery it was in error as PD battery is not the same as  Lipo or tool battery direct connection with 2 wires.

violavon, 2023-02-20:
I suggest we remove the whole battery row. It's completely dependent on the existence of a DC connector (because a battery isn't a USB powerbank, it's a "raw" battery with two terminals).

Although Ralim said: "Battery support means that settings for directly connecting batteries & powerbanks are enabled." What do you mean by enabled? Can you link to that part of the source code?

River-b, 2023-02-20:
we should not remove the battery row.
it's specifically one of the things that TS100 and Pinecils can do that is unique. in the field connecting to Lipo  and tool batteries on the go outside working on drones or in the back yard.   this is established knowledge by people who own all of these devices.

violavon, 2023-02-20:
I can't see the difference between the two.
 
Battery= Lipo, Lithium Ion Tool batteries (within the volts/amps your iron is rated for)
DC= DC laptop chargers (within the volts/amps your iron is rated for)

River-b, 2023-02-20:
a DC laptop brick like  a DELL/HP/Toshiba  that plugs into a wall electric outlet is not a Lipo  battery that does not need a wall outlet.  please do some reasearch on how a lipo battery is different from electric plugged in laptop charger.

violavon, 2023-02-20:
> we should not remove the battery row.
it's specifically one of the things that TS100 and Pinecils can do that is unique. in the field connecting to Lipo and tool batteries while outside working on drones or in the back yard. this is established knowledge by people who own all of these devices.

I agree, but isn't it just that they have a traditional DC jack? Nothing special.

River-b, 2023-02-20:
Start a Discussions section, this is off-topic and not really for Issue Ticket.

violavon, 2023-02-20:
Of course, but transformers gives the same direct current as batteries. The only difference is that transformers always gives (almost) the same current/voltage, while batteries voltage can drop (and rise a bit).
I think that anything that supports direct power from a transformer (e.g. DC jack) also support the batteries that provide the corresponding current/current. Right?

And anyway atm all soldering irons with DC jack in the README support battery and others don't. TS100, Pinecil V1/2 support DC/battery and MHP30, TS80/P don't, right?