Ralim/IronOS #1579: Cooperation with STM32 Soldering Station Firmware project

violavon, 2023-02-18:
Could we cooperate with [STM32 Soldering Station Firmware project](https://github.com/deividAlfa/stm32_soldering_iron_controller) and bring support for Quicko T12-072, Quicko T12-103, KSGER v1.5, KSGER v2, JCD T12, T12-955, Handskit, KSGER v3, T12-958 and other STM32 soldering irons?

Ralim, 2023-02-19:
This would be a fairly large project to undertake mostly due to UI differences.
Certainly doable and any PR's are more than welcome.

Would be open to cooperation absolutely.
But I do not currently have time and funds to buy all these different units and add support for them myself (as of present).
If someone who owned them was willing to do a lot of the testing I would be happy to work out with them.
Happy to build out a scaffold for these devices and help someone add support as well.

violavon, 2023-02-20:
I agree the amount of work in the user interfaces. We should take the best ideas from both, create a mix of them and then probably make soldering iron and station variations, since they are quite different.

It's awesome that both projects are open to this! I think the next step could be that you, @deividAlfa and other contributors look at each other's projects a bit and discuss what would be the ideal, the goal. Maybe it's that we bring compatibility to the drivers, maybe we strive to develop good UI practices, maybe we make one mutual firmware, or all of these. The main thing is that we work closely together, because open-source soldering iron firmwares have proven their power and necessity.

violavon, 2023-02-21:
It seems that STM32 Soldering Station Firmware project is going to discontinue. I think it would be best then if we could bring support for the devices it supports. All the devices use STM32 MCU so this should be easy. There are also some new settings that could be brought to IronOS. The biggest problem would be to decide, how we handle different display sizes? Do we make responsive UI or two separate ones?

I think the first step would be to port the drivers.

What do you think @Ralim?

closed 2023-02-23 by violavon