Ralim/IronOS #1581: Support for Youyue 858D+ hot air gun

violavon, 2023-02-18:
Could we bring support for Youyue 858D+ hot air gun? @madworm's [Youyue-858D-plus fimware project](https://github.com/madworm/Youyue-858D-plus) could be great help for this.

Ralim, 2023-02-19:
It looks like this hardware does not meet minimum requirements of
- Arm Cortex M0 or faster
- 64k of flash or higher
- 16k of ram or higher

Ralim, 2023-02-19:
If someone wanted to make a PR I would be very highly likely to merge it in; but I have zero desire to work with or support atmel parts these days.

violavon, 2023-02-20:
Why ARM is essential? I understand that maybe ATMEGA is too unpowered, but I checked the source code of this repo and there is ARM and RISC-V.

Ralim, 2023-02-20:
ARM is not essential; but running FreeRTOS well _is_. So RISCV or ARM is best bet in the space. (Not atmega cores).
I just quickly skimmed the readme and only noticed the atmega references.

violavon, 2023-02-20:
"Arm Cortex M0 or faster" 

Ok so also RISC-V (nothing else?), how powerful does it need to be?

Ralim, 2023-02-20:
Any core that can run freertos with low overhead, which is generally any of the newer arm cores; all the embedded riscv cores.

Mostly it can't be the older 8/16 or proprietary cores like the microchip ones (though for that one I know you can do it but it's messy).

The amount of power depends how good your hardware is, the more we can offload to hardware the less the code does.
So for most devices we offload scheduling the tip control to hardware but on Pinecilv2 we don't. That consumes a few percent more CPU utilisation on that device; but it's got an absurdly overkill the processor so it doesn't matter.

violavon, 2023-02-20:
> proprietary cores like the microchip ones (though for that one I know you can do it but it's messy)

You mean MCUs by Atmel/Microchip company? What's the problem with them? I think, they aren't more proprietary than ARM. And ATmega is used e.g. in Arduinos so there is enough documentation of them.

What is the benefit of not offloading in some soldering ironies, such as Pinecilv2?

Youyue 858D+/Quick 861DW firmwares I linked contain many of the same features as IronOS. Why does IronOS require so much more powerful MCU?

Ralim, 2023-02-20:
>You mean MCUs by Atmel/Microchip company? What's the problem with them?

The AVR core isnt super great for running FreeRTOS on; and the PIC series I tend to avoid as not being as easy or friendly for debugging last I worked with them. Documentation isnt the limit; but more trying to get FreeRTOS running on them smoothly with their different architecture can be messy. Also all the "newer" cores such as ARM & RISCV have a much more friendly memory layout for debugging and keeps the code a lot more similar across devices so that fixes can be rolled out with less bugs.

>What is the benefit of not offloading in some soldering ironies, such as Pinecilv2?
None; the PinecilV2 cant offload due to limitations in the timer hardware. As such the gap must be done in software with IRQ handlers.

>Youyue 858D+/Quick 861DW firmwares I linked contain many of the same features as IronOS. Why does IronOS require so much more powerful MCU?

Because I dont have time to sit and set it up for other MCU's. As I have said before, if someone wants to submit a PR and sign up for maintaining that implementation I would gladly accept it and be happy to look into further support. But every device we add increases my unpaid work on this project by at _least_ an hour a week 😓  So I lean towards supporting devices that do not increase that burden significantly. Adding a hot air tool will require rework of a few bits and pieces that are fundamental to the firmware; which is non-trivial effort. I also need to buy samples for testing. Then I need to do the firmware port and sign up for supporting it for years. Given IronOS is a hobby project and not at all an income source; that is a lot of hours of outlay for a product that already has a working open-source implementation out there. My general view is that IronOS should be relatively focused on a neat; clean implementation for the devices it supports. Rather than a half-assed one for lots of devices.

violavon, 2023-02-20:
> Given IronOS is a hobby project and not at all an income source

How do you have enough free time for this?

> that is a lot of hours of outlay for a product that already has a working open-source implementation out there.

But [this](https://github.com/madworm/Youyue-858D-plus) is a bit unfinished/hacky.

Okay, so by this "Arm Cortex M0, 64k of flash, 16k of ram" you didn't mean that we need so much computing power or memory, but support can be done in a reasonable way only for fairly new ARM/RISC-V MCUs?

Admittedly, the user interface (e.g. display) of 858D+ and 861DW is quite restrictive and different from what IronOS is made for.

violavon, 2023-02-20:
Do you know any hot air station with suitable MCU, ports (preferably USB-C for programming) and better display? I could possibly spend a lot of my time developing support and UI for one.

GPSBabelDeveloper, 2023-02-24:
Lifetime free support for a $50 device is hard to come by when you're trying to get someone else to do the work. Once a project has set its minimum requirements, it's hard to go backward substantially in CPU/RAM/ROM requirements AND keep adding new hardware. Retrofitting code designed for a 32-bit base onto an 8-bit micro is rarely trivial. Since you've offered to "possibly spend a lot of time developing support", if you're willing to carry that long-term maintenance, perhaps taking Ralim up on his offer of integrating your work is the way to go.

FWIW, I skimmed many pages attempting to "fix" this device (hint: there are at least 4 different PCBs with that model #) including 30+ pages on eevblog, hackaday, youtube, github and more and didn't see a clear "winner" when it came to fixing whatever is wrong with that model. http://blog.spitzenpfeil.org/wordpress/projects/youyue-858d-hotair-station-modifications/ seems to be about the best lead. It seems like there were a lot of unhappy people with that model, so good luck!

closed 2023-05-16 by discip