Ralim/IronOS #1586: Why TS80 isn't "recommended"?

violavon, 2023-02-20:
Why TS80 isn't "recommended"?

Ralim, 2023-02-20:
Completely replaced by the TS80P and discontinued

violavon, 2023-02-20:
Discontinued? https://www.miniware.com.cn/product/ts80-soldering-iron-more/

Ralim, 2023-02-20:
Yes it may come as a shock, but production ramped down a long time ago and it's just existing stock clearing the system.

I marked it not recommended being optimistic that people might pause and buy the far superior TS80P instead

violavon, 2023-02-20:
Okay, but does IronOS work as well with TS80 as with TS80P? Nothing like with TS100 and STM32 clones?

Ralim, 2023-02-20:
To the best of my knowledge yes, but that doesn't change it from being a no-longer-reccomended device.

violavon, 2023-02-20:
Hmm, then I feel we shouldn't be recommending anything on IronOS. I understand if there is something like TS100 and STM32 clones, but otherwise no.

closed 2023-02-20 by violavon

Ralim, 2023-02-20:
I don't follow your logic sadly.

The recommendation is for buying, not for using ironos.

violavon, 2023-02-20:
In that context, it affects more the status of the support.

Ralim, 2023-02-20:
Would you prefer the Reccomended column be renamed to something else to make it clearer?
Do you have suggestions?

DjordjeMandic, 2023-02-21:
Maybe something like #1588 ?

Ralim, 2023-02-21:
Sounds good