Ralim/IronOS #1594: Refactor display settings handling in BLE callback

NeilHanlon, 2023-02-23:
https://github.com/Ralim/IronOS/blob/68d18a4fb0c468cb276ff89a37461f169cb9e695/source/Core/BSP/Pinecilv2/ble_handlers.cpp#L246-L255

There is probably a better way to handle this than three if statements in a row :)

Open to suggestions

GPSBabelDeveloper, 2023-02-24:
Here's three different ways to make that a little less grubby. I prefer the style of the first one, all other things being equal.

switch (uuid_value) {
  case SettingsOptions::OLEDInversion: 
   OLED::setInverseDisplay(getSettingValue(SettingsOptions::OLEDInversion)); 
   break;
 case SettingsOptions::OLEDBrightness:
   OLED::setBrightness(getSettingValue(SettingsOptions::OLEDBrightness)); 
   break;
 case SettingsOptions::OrientationMode:
   OLED::setRotation(getSettingValue(SettingsOptions::OrientationMode) & 1); 
   break;
 default:
   BT_WARN("Unhandled uuid_value in %s. Got 0x%x", __PRETTY_FUNCTION__, uuid_value);
   break;
}

Or keep it like this, but don't force common subexpression elimination to determine the mutual exclusivity of the paths:

 if (uuid_value == SettingsOptions::OLEDInversion) { 
   OLED::setInverseDisplay(getSettingValue(SettingsOptions::OLEDInversion)); 
 } else  if (uuid_value == SettingsOptions::OLEDBrightness) { 
   OLED::setBrightness(getSettingValue(SettingsOptions::OLEDBrightness)); 
 }  else  if (uuid_value == SettingsOptions::OrientationMode) { 
   OLED::setRotation(getSettingValue(SettingsOptions::OrientationMode) & 1); 
 } 

You could possibly shave a few clock cycles off making PMFs to the three methods, masking the bottom two bits from uuid_value and jumping through a function dispatch table  with the constant arguments andtesting explictly for the last one and doing something similar after a mask. A self-respecting optimizer MAY do this for you if it can determine these are pure functions (i.e. they're marked const and can be proven to not modify larger scope. If you need clock cycles this badly, you're probably living wrong, but I"m all about giving choices.

Consider marking the functions const:

static void    setBrightness(uint8_t contrast) const;
static void    setInverseDisplay(bool inverted) const ;
Same for others

... if you can. If you're not modifying any state of the OLED class (e.g. no globals, no honking on the invisible this*) that even further clarifies the code and exposes additional optimizationopportunities.

discip, 2023-03-10:
@NeilHanlon
So... are you planning to implement any of the proposed solutions? Or do you expect this to be done for you?
I could try to, if you want.

River-Mochi, 2023-03-11:
@NeilHanlon   of the new options for the code that GPSBabel suggested to make it better, which one are you going to Submit?  it should be submitted  asap though so that it can go into the 2.21 release yes?

NeilHanlon, 2023-03-12:
Yep, any of these would be fine for implementing. Was planning to get back to this soon, but thank you for taking it @discip !

River-Mochi, 2023-03-14:
@NeilHanlon  could you close this ticket since it's implemented in code now for  2.21  upcoming release  👍

discip, 2023-03-14:
No, this is not implemented yet.
This will automatically close once merged.
It doesn't have to be done manually.

River-Mochi, 2023-03-14:
thanks @discip

PR #1616 (2023-04-01): implementing #1594
This closes #1594.

@NeilHanlon
@River-Mochi
Please let me know if this is working as expected.

@GPSBabelDeveloper
Unfortunately I had no idea what to put in **`default`**.
Any suggestions?
closed 2023-04-01 by discip