Ralim/IronOS #1599: PD negotiation issues with Baseus CCGAN10CUS

kantoniak, 2023-02-27:
I have a Baseus charger and PD negotiation with Pinecil V2 sometimes works and sometimes doesn't and stays at 5V. I'd like to report it since the charger works perfectly with other hardware that I have and the issue may be connected to IronOS' stack. I've used Pinecil with other chargers (Lenovo laptop charger, MacBook charger) without problems.

I did some testing with PD debug mode (holding the `+` sign). The hardware:
* Baseus GaN2 Pro Quick Charger CCGAN10CUS (100W) (the charger in question)
* Baseus USB-C cables for Baseus charger (both 2 meters long): 65W (slightly used) and 100W (almost new)

I get different results based on the order I connect things together:
| Cable |           Connection order           |                       Effect                      |
|:-----:|:------------------------------------:|:-------------------------------------------------:|
| 65W   | To outlet, Baseus side, pinecil side | Always works (`12 no VBus`)                         |
|       | To outlet, Pinecil side, Baseus side | works about 1/2 of time (`12 no Vbus` vs `3 no VBus`) |
|       | Baseus side, pinecil side, to outlet | works about 1/3 of time (`12 no Vbus` vs `3 no VBus`) |
| 100W  | To outlet, Baseus side, pinecil side | works about 1/2 of time (`12 no Vbus` vs `3 no VBus`) |
|       | To outlet, Pinecil side, Baseus side | works about 1/2 of time (`12 no Vbus` vs `3 no VBus`) |
|       | Baseus side, pinecil side, to outlet | works about 1/3 of time (`12 no Vbus` vs `3 no VBus`) |

I don't have PD testers which could log details of the negotiation, unfortunately. Can it be Pinecil side issue? Any suggestions on how to improve the reliability of connection?

Samuelrmlink, 2023-02-27:
Try this firmware. 
[Pinecilv2_PPSfix (1).zip](https://github.com/Ralim/IronOS/files/10835852/Pinecilv2_PPSfix.1.zip)

Basically there was a firmware issue that was causing PPS PDOs that happened to also be the 7th PDO to have a disabled PPStimer - which would result in the charger cutting power to the Iron after about 10-30 seconds. (See PR #1572)

You can use the either the Pinecil updater on Windows or dfu-util on Linux to flash these update files to the device over USB. 
Make sure to use the attach files because this fix has not been shipped on the current release of IronOS. (V1.20)

Let me know whether this fixes the issue.

Samuelrmlink, 2023-02-27:
What PDOs are listed when you hold down the + button on your Pinecil?
Click through the different PDOs using the + button.

River-Mochi, 2023-03-01:
> What PDOs are listed when you hold down the + button on your Pinecil?

if you could make a short video and drop it in here of you moving through the PD+ menu that wouldl help or just type every PD sceeen you see exactly as it appears.  (hold down +  _before_  plugging in cable, then hold the + for another 10 seconds,  to make sure everything is settled down on negotiation, then use same + to scroll through each PD menu screen).

River-Mochi, 2023-03-01:
Ralim - Samuelrmlink's comment with Pinecil Beta fix  is being marked  _off-topic_ even though it is  relevant - is this some automation bug?
I had to click on it to see it since it's marked "off-topic" ?

[image]

Ralim, 2023-03-01:
No-one marked it as a offtopic; that is likely an automation from Github because of the double-post.

River-Mochi, 2023-03-01:
> I believe I've manually marked it to always show now.

thanks

kantoniak, 2023-03-01:
Thank you for the responses. I haven't been able to test latest firmware or get PD analyzer (this may take a while) yet, other info below.

> 1. what exact firmware are you running on the pinecil V2? Long hold (-) button.
```
v2.20.14DAF70
13-12-22
```
> 2. if you could make a short video and drop it in here of you moving through the PD+ menu that wouldl help or just type every PD sceeen you see exactly as it appears.  (hold down +  _before_  plugging in cable, then hold the + for another 10 seconds,  to make sure everything is settled down on negotiation, then use same + to scroll through each PD menu screen).

PD Debug, only the bottom line changes:
```
PD Debug
State 12 No VBus
 1  5V  3. 0A
 2  9V  3. 0A
 3 12V  3. 0A
 4 15V  3. 0A
 5 20V  3. 0A
 6  3-21V  60W
```

When connection doesn't work out, I only get:
```
PD Debug
State  3 No VBus
```

> 3. Try to use other cables if possible. also Flip the cable over.  _not just baseus cables._

Also tried MacBook charger cable and some unbranded 15-centimeter cable apart from the previous ones, checked all the flip configurations (😄) and it doesn't seem to affect results.

> 5. could  you post a photo here of the back side of the Baseus charger that is giving trouble- this is side that lists all the specs on it.

Image here: [Baseus-CCGAN10CUS](https://user-images.githubusercontent.com/17510686/222081790-15fafb28-a06d-49c5-9e0b-ff979709eab5.jpg)

River-Mochi, 2023-03-01:
@kantoniak yes if you could  flash the beta firmware that samuel linked to you  above in the first comment, that would be good to try.  you can download it right from inside his comment.
instructions for updating firmware are on pinecil  wiki page.

kantoniak, 2023-03-04:
Hello, I tried the linked firmware:
```
v2.20.95C4CAF
24-02-23
```
I'd say that even less negotiations are successfull. I went back to stable firmware for now and I will report back when I get a PD logger.

River-Mochi, 2023-03-05:
> even less negotiations are successfull.

what did you see when you were in the PD debug, which  state lines?

kantoniak, 2023-03-06:
PD Debug is exactly the same as before, I've updated the previous comment.

River-Mochi, 2023-04-03:
@kantoniak   I'm starting to think it's just this one charger you have that is not performing right. since all your other chargers work  perfectly fine with Pinecil V2.  do you have a  PD  analyzer like the KM002 or KM003 ?  you could capture  PD analyzer log on your PC  which would have better data than the minimal info in PD debug.  send logs in in and  perhaps Ralim could look at it since it will tell exactly what is going on. 

This happened rarely but on one of my other chargers,  sometimes only show as 5V, then I unplug and plug in again and get 20V.  those chargers I give away to people or they are in bottom of box.  PD analyzer info sent to Ralim is best, but I didn't want to bother at the time since it did work if I just replugged it in.

Specific chargers do cause issues with V2, and it's recommended to not use them. sometimes they are not made to USB-C spec,  or for example could be 3 of same model and 2 work and one causes pinecil issues.   basically the PD debug info is not going to be enough but sending PD analyzer  log might help since it shows exactly the negotiation between the charger and Pinecil.
a good negotiation between a Usb-C charger and  the Pinecil will show as State 12 (all is good).  State 6 happens if people plug into a QC 3.0 charger (which means it's waiting for comms, which will never happen since it's not PD).   I don't know what State 3 is. 

see here for some examples of other bad chargers: https://github.com/Ralim/IronOS/issues/1644

kantoniak, 2023-04-05:
Thanks for checking in, I just noticed the reply. I do own KM003, but I didn't know it can log PD packets. (I was hoping to sniff out CC lines with a breakout board and an oscilloscope and put that in sigrok or so, but I'm not comfortable with such surgery when connected to mains). Is there any tutorials how to log PD negotiation? Afaik official KM002/003 docs don't cover Power-Z app.

River-Mochi, 2023-04-05:
- with the KM003,  open Software on windows PC,  go to PD  analyzer  and  record   a short session between  the pinecil and the Charger. 

- Zip it and attach it here  and ask for Ralim to look at it since he will understand what it means. 

- One USB cable goes from the KM003  to the PC, and one cable from the KM003  to the Pinecil.  

- This video  I'm using the software on the right side of the screen  side by  side with Joric's BLE  chart on the same PC. 
- https://www.youtube.com/watch?v=7i5oPPaM0K4

- Photo here,  the black cable is going to my charger  which 
- and Pink cable going to the PC windows for the software from chargerlabs.
- [image]

River-Mochi, 2023-04-05:
I don't have direct links to the free software  so if you can't find it on  Chargerlabs website one of the  discord pine64 chat members shared it iir.
They post the software free because if you don't have a KM002/3  to plug into PC, then software does nothing.

kantoniak, 2023-04-06:
Thank you for elaborate explanation, I managed to read everything what's needed.

@Ralim I attach two logs for connections which end in State 3 and State 12: [pd-logs.zip](https://github.com/Ralim/IronOS/files/11165165/pd-logs.zip). Logs have Vendor Defined messages which in my understanding is cable identification. This part was sometimes missing but did not affect whether negotiation is successful. From my testing it looks like Pinecil sometimes does not respond to capability advertising and Baseus repetitively advertises for 10 secs and then stops.

discip, 2023-09-23:
@kantoniak
Is your initial issue solved?
If so please close. 😊

kantoniak, 2023-09-24:
Thanks for bringing this up. Could someone take a look at the logs please? If it seems to be a faulty device/model, I think the issue can be closed.

> I attach two logs for connections which end in State 3 and State 12: [pd-logs.zip](https://github.com/Ralim/IronOS/files/11165165/pd-logs.zip). Logs have Vendor Defined messages which in my [...]

vladimir-poleh, 2024-10-18:
I have similar problem with Wotobeus 240W charger. Tested with Pinecil V1 and Pinecil V2 and different USB Type-C cables. Both Pinecils are using firmware v2.22E.C308FE8C.

Pinecil V1 is working fine, I have following logs in Fnirsi FNB38 USB tester:
```
01 03A3 ACC ->
02 0241 CRC <-
03 05A6 RDY ->
04 0441 CRC <-
```
No available modes are displayed in the FNB38 PD listener UI, just actual 20V.

For Pinecil V2 most of the connections are not working. PD Debug is showing `State 3 No VBus`, `-` button returning to the start screen with blinking 496C temperature and 0.0V indication.
Logs from FNB38 for this case are 80 equal records:
```
01 61A1 CAP ->
02 61A1 CAP ->
  ...
80 61A1 CAP ->
```
Extended info for there log records:
```
01/80
0x61A1
0801912C
0002D12C
0003C12C
0004B12C
000641F4
C1A42164
```
Also I see following modes on the FNB38 screen:
```
1.  5.00V 3.00A
2.  9.00V 3.00A
3. 12.00V 3.00A
4. 15.00V 3.00A
5. 20.00V 500A
6. 3.30-21.00V 5.00A
```
No modes are selected. `PD3.0 105W` is displayed in the status area.

If connection of the Pinecil V2 was successful, I have following records in FNB38 logs:
```
01 61A1 CAP ->
02 61A1 CAP ->
  ...
12 61A1 CAP ->
13 0041 CRC <-
14 1082 REQ <-
15 0121 CRC ->
16 03A3 ACC ->
17 0241 CRC <-
18 05A6 RDY ->
19 0441 CRC <-
```
Same available modes are displayed as for unsuccessful connections but `5. 20.00V 5.00A` is highlighted.
Pinecil is showing correct temperature and 20.02V.