Ralim/IronOS #1601: Pinecil v2 bug or Defect

Wolf65, 2023-03-01:
**Describe error**.
When cooling down between 220-150 GOP of houses, the "no stinger" icon flashes for 1 frame on the screen

**For reproduction**
In the settings, the normal soldering interface and waiting (not advanced) is selected
**Expected behavior**
Cooling down with no stinger indication

**Details of your device:**.
 - Device: Pinecil v2
 - Release: 2.20.14DAF70 13-12-22 rus lang
 - Power adapter used: dc 24v 3a

Wolf65, 2023-03-01:
also when holding the temperature also for 1 frame the temperature 400+ is displayed

discip, 2023-03-01:
This could be a bad tip.

Did you try other tips? Did they behave similarly?

Wolf65, 2023-03-02:
Yes I tried to use iron tip from ts100 exactly the same behavior. 2 other people have the same soldering iron behavior.

River-Mochi, 2023-03-15:
Do you have a V2 ?   after cleaning it,  you could also  test it  
see the photo in here  on how to check new tips. 
V1 short tips are usually about  6.2-6.5 ohms.  (hint: use the black foam that came with pinecil to hold the tip while you test with meter). 
(clean the probe tips too ).

lmoe, 2023-03-16:
Wow thanks for the thorough explanation!

It was indeed the cleaning of the tip. 

I read parts of the wiki but not enough to do the right thing as I was doing 5 things at once...

With "tip" I understood the top part you use to solder. So I just cleaned the top and ignored the bottom that you stick into the iron.  

Gave it a complete IPA cleansing again and now the temperature does not flicker anymore nor is there any warning flashing. I'm using the short tip that gets shipped with the iron and it's at 6.2 ohms now. The Tip-R was always flickering in a 100+- range before, now it's more stable.  

Thanks again!

discip, 2023-03-16:
@Wolf65
Have you tried @River-Mochi 's advice, yet?

discip, 2023-03-16:
@Wolf65
Please try @River-Mochi 's advice to hopefully solve your issue.

Wolf65, 2023-03-17:
Yes, it's definitely better now the temperature "flashes" less often. I need to clean the iron tip even better. isopropal alcohol ran out at the wrong time.

River-Mochi, 2023-03-17:
- some people also opend Pinecil  see  "Easy Trick to Open"  https://www.youtube.com/watch?v=aK01V5DrrVk&t=1s
- then also wiped all contact pads too. Dry and reassemble.

River-Mochi, 2023-03-17:
> With "tip" I understood the top part you use to solder.

could you tell me where this is  or screen shot and I will try to make it more clear which end of the Cartridge to clean.

River-Mochi, 2023-03-17:
> With "tip" I understood the top part you use to solder.
in English this is a confusing word.   Tip = soldering cartridge 
and Tip  = the top of something like Tip of a  pen or a mountain. 

In this case we want to clean  the White  End or cold end where it makes electric contact but yes Wipe whole thing is good. I have no idea what kind of oil or residue the factory is leaving on the tips but it is interferring with voltage readings.

[image]

River-Mochi, 2023-03-17:
> . The Tip-R was always flickering in a 100+- range before, now it's more stable.

Raw - Tip   is normally  ~700-1100 when i  check Debug while iron is on but cold.  Short tip is ~690 and long tip about ~1000 for me.

River-Mochi, 2023-03-17:
@Wolf65  if alchohol cleaning +  heat it up a few times to 350C for a couple minutes resolves this,
 please close ticket 😄

closed 2023-03-17 by Wolf65