Ralim/IronOS #1609: BLE - need serial number for authenticity checker

River-Mochi, 2023-03-07:
**Is your feature request related to a problem? Please describe.**
There is an Authenticity checker for people who own pinecil V2 and want to check if it's legit.
Currently, we have to read and transfer the long and tiny numbers from ID and Serial number from a tiny pinecil screen to the checker webpage. 
**BLE already sends ID number, just need Serial number too.**  

**Describe the solution you'd like**
- [ ]  BLE send serial number with the ID number that is already being sent.
- Step 1, PineSAM could show the ID number and serial number in larger font to make checker process easier.
- Step 2, PineSAM roadmap to integrate Authenticity checker for conveinience. 
- For details see ticket: https://github.com/builder555/PineSAM/issues/83#issue-1614110090
- @gamelaster knows of this proposal.

As soon as PineSAM starts getting the serial number on BLE, work can be started on how to  integrate it to add the Checker feature for fun and convenience.

ew1abz, 2023-03-24:
What is a Serial number, and what is ID? 
According to https://pinecil.pine64.org/, the Serial number is the second, long string from the Debug menu. 
If so, the FW already sends it via BLE
https://github.com/Ralim/IronOS/blob/ccbee2187269fb9c286c7d1f6ae03f58a5cb1a05/source/Core/BSP/Pinecilv2/ble_handlers.cpp#L183 

Although, in the FW we use the `getDeviceID()` function to get it. Very confusing.
I assume we need to send the `getDeviceValidation()` output.

Before any renaming or implementation, it maybe makes sense to have more information about them. Which is set by the uCU manufacturer, and which is set by the Pine64 factory?

River-Mochi, 2023-03-24:
> If so, the FW already sends it via BLE

we are not currently seeing serial number in PineSAM 
@builder555 could you please check this?

River-Mochi, 2023-03-24:
Pinesam  app is here if you want to check it out
https://github.com/builder555/PineSAM

gamelaster, 2023-03-24:
@ew1abz Serial Number is set by chip manufacturer. Device validation is set by Pine Store factory.

Ralim, 2023-03-24:
> What is a Serial number, and what is ID? 

Serial number is the ID burned by manufacturer that is the device MAC.
ID is a unique key Pine burns at the factory and records in their db.

Serial number is printed using `getDeviceID()`; https://github.com/Ralim/IronOS/blob/11bad84ce7a44c9d2ef0141459a2dc175e5b671a/source/Core/Threads/OperatingModes/DebugMenu.cpp#L22

Meanwhile on the PinecilV2 we show the extra validation code from `getDeviceValidation()` Which returns the validation code in OTP.

> According to https://pinecil.pine64.org/, the Serial number is the second, long string from the Debug menu. If so, the FW already sends it via BLE [...]

Happy for us to add `getDeviceValidation` to the BLE data to allow for this.

River-Mochi, 2023-03-24:
> Serial number is the ID burned by manufacturer that is the device MAC.

- I can get my  mac  address  from  pinesam script terminal which looks something like a MAC  should look like  
 C4:D7:--:--:--:--

- Using this MAC converter with my full real MAC here:   https://www.vultr.com/resources/mac-converter/
   nothing here looks exactly like Serial number on the Pinecil. 

- HEX  example looks _close_  and is all the same letters and numbers I see on Pinecil, but  it's like the Pinecil Serial number is a hex conversion of the MAC  that is then a bit scrambled from any common conversions.

River-Mochi, 2023-03-24:
I see it now,  my serial number is padded  with  4 zeros, and then  it's the reverse of my mac address, in sets of 2 characters.

ew1abz, 2023-03-24:
Thanks, everybody. That reaffirms my suspicions about misnaming things.
Usually, `serial number` refers to a **product** SN. Here, on the device validation page and BLE GUI, it is used for naming chip unique ID. In the FW, though, all naming is OK.
Anyway, the output of the `getDeviceValidation()` function should be exposed via BLE.

River-Mochi, 2023-03-24:
> Anyway, the output of the `getDeviceValidation()` function should be exposed via BLE.

@ew1abz 
do you mean getDeviceValidation() = serial number as seen on Pinecil screen   is already  available over BLE?  
or will you submit PR for this?
will it stay this misname or get a new name?  thanks

ew1abz, 2023-03-25:
Let's stick with @Ralim terminology:

> Serial number is the ID burned by manufacturer that is the device MAC. It is printed using getDeviceID() 
> ID is a unique key Pine burns at the factory and records in their db. It is printed using getDeviceValidation()

> do you mean getDeviceValidation() = serial number exactly as seen on Pinecil screen

No

> is already available over BLE?

Yes, the `Serial number` is available over BLE. The `ID` is not.

> or will you submit PR for this?

Yes, that's my intention.

> will it stay this misname or get a new name?

Unfortunately, we have to keep the old naming. Otherwise, we need to change FW and `Pinecil V2 Authenticity Checker`. It'll not be backward compatible and will cause even more confusion at the user level.

PR #1666 (2023-04-16): Add validation ID to BLE
### **Please check if the PR fulfills these requirements**

- [x] The changes have been tested locally (see the attached picture)
- [x] There are no breaking changes

### **What kind of change does this PR introduce?**
    
Resolves #1609 

### **What is the current behavior?**

Device validation ID is not available via BLE.

### **What is the new behavior (if this is a feature change)?**

The serial number and validation ID are now both exposed via BLE.
Clarifies naming in the source code. The detailed explanation is in the comments for #1609.

[image]
closed 2023-04-16 by discip