Ralim/IronOS #1613: Bugs with MHP30

RWDZ, 2023-03-09:
Device: MHP30

Something strange is happening with the sleep mode settings. The sleep mode is activated significantly earlier than set in the settings. When I set the sleep mode for 3 minutes, sleep started after a minute. When I set the sleep mode for 10 minutes, sleep started after a 3 minute. After that, the heater refused to exit sleep mode at all. The display constantly blinked despite attempts to press different buttons. I turn on to show sleep mode as blinked display.

The heater always turns on with a red backlight. The heater has a tri-color LED installed, which changes color depending on the temperature on the original firmware. On the last Iros OS firmware, the color is always red.

discip, 2023-03-10:
@RWDZ
Maybe this is related to #1570?

RWDZ, 2023-03-11:
Perhaps this is a manifestation of the same problem. But that does not change the essence of the problem. The stock firmware turned out to be even better in this case. I do not recommend anyone to use Iron OS with MHP30 now.

discip, 2023-03-11:
Can you confirm that your post could be declared as a duplicate of the mentioned one?
If so, would you mind if we close this issue, mark it as a duplicate and reference it in #1570, so there is less noise. 😊

closed 2023-06-29 by discip