Ralim/IronOS #1614: Fnirsi HS-01

cvetaevvitaliy, 2023-03-09:
**Is your feature request related to a problem? Please describe.**
Add support Fnirsi HS-01

**Describe the solution you'd like**
Create custom firmware for Fnirsi HS-01

**Describe alternatives you've considered**
There is none

**Additional context**

[image]

PCB board

[image]

River-Mochi, 2023-03-14:
- is this already open source, do you have a link  to the source code?
- is there a  schematic available, also link  or attach it - this helps a lot or contact Vendor and ask for schematic.
- which exact main MCU chip is used in this? 
- currently IRonOS  works on BL706, GD32VF103TB, Stm32f103  MCU chips  (easier to port IronOS if same chips).

Syncmasterklop, 2023-03-20:
I received this soldering iron, it looks and works very well.
Manufacturer - Fnirsi, I wrote them a letter about the circuit and source code, but there is no answer yet (most likely there will be no answer).
The main chip is CH571F, you can see it in the photos.
There are no specifications from the manufacturer, there is an instruction where you can see something.
[HS-01 Intelligent Electric Soldering Iron User Manual V-02.5.pdf](https://github.com/Ralim/IronOS/files/11013440/HS-01.Intelligent.Electric.Soldering.Iron.User.Manual.V-02.5.pdf)
[CH573DS1.PDF](https://github.com/Ralim/IronOS/files/11013442/CH573DS1.PDF)

Ralim, 2023-03-24:
Until manufacturer is willing to provide a pinmap or schematic support isnt going to happen sadly
That MCU can definitely run IronOS from a specifications perspective but without schematics its just too much work to reverse engineer at the moment.
You are welcome to try and get in touch with the manufacturer and find out if they are interested. If they are willing to help with providing information for porting then this can be considered. (If you get an email contact, feel free to include me in the email if they are ok with it).

0x4C4A, 2023-03-24:
I've also sent a message to Fnirsi outlining why IronOS support would be beneficial and the fact they don't have to share the entire schematic, but just the relevant parts. Here's to hoping they realise the marketing potential.
The hotend cap is a really standout feature for me.

Careless, 2023-04-22:
> 

slightly off topic, but can anyone measure the HS01 iron without the hot-end cap?

The measurements i see online are 184mm x 20mm diameter from end to end with the cap installed.

i don't have a need for the cap, but my case I use for storing all my equipment has 170mm column slots. if it fits without the cap, that'd be great.

also if anyone can measure the LxWxH of the included PD65 watt charger, that would be much appreciated. Not much info on these units yet.

0x4C4A, 2023-04-23:
Forgot to update - Fnirsi wrote me back, unfortunately they were not receptive to my reasoning that releasing the schematic would be beneficial. 

> Hi, dear friend,
> Thank you for your support to FNIRSI and your suggestions,
> It's a pity that the schematic is the engineer's secret, he didn't share it with us, we can't help you.
> Thank you for your understanding, 

Perhaps more messages may make them reconsider, but for now, it looks like reverse engineering is the only way to go. 
 

@Careless 
The length depends on the tip used, the iron with my default tip (HS01-B2) is 176mm, with the longest - the thin tip (HS01-ILS), it goes up to 180mm.
I have the EU power brick, and its dimensions are ~90.1mm from prongs to top, ~53mm for just the brick from bottom to top (not counting the prongs) and  35.8mm squared for the brick outer dimensions. [Here's a picture, so it's easier to understand.](https://photos.app.goo.gl/rtL4u8BrsaEsHVg37)

Careless, 2023-04-23:
that's a shame that they don't want to supply the schematic, but maybe they'll change their mind if we bug them.

damn, 176mm is 6mm longer than i was hoping. i wonder if there is a shorter one I can keep in the iron for storage in my carrying cases I use for my equipment.

i have one coming on the way tomorrow, maybe I won't open it just yet. i was hoping i could get something to replace my ts100 that could be powered from a 100w power bank in the future for field repairs where no power is available.

thanks for measuring.

ssharkula, 2023-04-25:
CH573DS1.PDF в конце документа нашел ссылку на сайт. Очень много информации. Будьте внимательны.
At the end of the document I found a link to the site.  A lot of information.  be careful

Careless, 2023-04-26:
I just got the iron yesterday. It feels like a good value due to the charger and the cord and the full range of tips. the entire ensemble fits nicely in my field-repair boxes. 

really hoping we can somehow get IronOS to work on this thing. It seems ok as is, but I noticed it immediately starts heating when you press the left most button and i didn't know what temp it was getting to, so i had to throttle it back.

also navigating the UI is a little cumbersome until you get used to it.

maybe I should pester them to reconsider their engineers opinion on third party firmware.

RogWilco, 2023-05-02:
> It's a pity that the schematic is the engineer's secret, he didn't share it with us, we can't help you.

This sounds more like a _can't_ rather than a _won't_. Are we sure that Fnirsi is the OEM and not simply the first of many to brand a white-labelled product? I know they claim to be an OEM but it wouldn't be the first time.

I'm interested in this iron too so I'll do a little digging to see if I can find anything.

Careless, 2023-05-02:
> 

I emailed them as well through their website form and have not received any reply whatsoever. 

In my message I didn't pressure them or anything of the sort, I just asked them to consider allowing the users of their products extend the life of their purchase by allowing themselves to offer longer term support publicly as it would be beneficial to their sales because right now TS101 and Pinecil V2 are the most popular and the HS-01 is probably not selling as well as it could because of its closed software.

maybe they got the message, maybe not. i can't tell as of yet.

rzabarazesh, 2023-05-02:
+1
This iron looks very promising. I love the construction.

sasuta, 2023-06-09:
Is FNIRSI HS-01 compatible with TS100 tips

Careless, 2023-06-09:
> Is FNIRSI HS-01 compatible with TS100 tips

no. it uses it's own type of tip that i think is the same as the GVDA irons, and is supposedly just a shorter version of the T12 style tips.

sasuta, 2023-06-09:
Thanks bro.

Careless, 2023-06-09:
> Thanks bro.

no problem.

They are referred to as "FM65"-XX tips, the XX being the same two digit identifier for the tip style as used to describe tips for the TS100 as far as I've seen.

limyz, 2023-06-10:
Holding the left button while inserting the USB-C pushes the iron into bootloader mode.
Holding the right button while inserting the USB-C pushes the iron into programming (?) mode (see below for unknown device VID: 4348 PID: 55E0).

Attached is firmware update from their site, http://www.fnirsi.com.cn/download/weld

[image]
[HS01-20221212.zip](https://github.com/Ralim/IronOS/files/11711854/HS01-20221212.zip)

Jonas-fr, 2023-06-17:
Maintaining the **left** button while plugging it in (power it on) :
On the iron LCD : BootLoader
On the computer :

```
$ dmesg
[...]
[14385.166629] usb 2-1.2: new full-speed USB device number 5 using ehci-pci
[14385.278038] usb 2-1.2: New USB device found, idVendor=1a86, idProduct=5722, bcdDevice= 1.00
[14385.278058] usb 2-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[14385.278065] usb 2-1.2: Product: QPUC0
[14385.278069] usb 2-1.2: Manufacturer: QPLINK
[14385.278073] usb 2-1.2: SerialNumber: 2017-2-25
[14385.335846] cdc_acm 2-1.2:1.0: ttyACM0: USB ACM device
[14385.335884] usbcore: registered new interface driver cdc_acm
[14385.335887] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
```

Remarks : Stays in this mode as long as the iron is unplugged (powered off).

- - -

Maintaining the **right** button while plugging it in (power it on) :
On the iron LCD : (empty LCD)
On the computer :

```
$ dmesg
[...]
[14597.822870] usb 2-1.2: new full-speed USB device number 6 using ehci-pci
[14597.931688] usb 2-1.2: New USB device found, idVendor=4348, idProduct=55e0, bcdDevice=29.00
[14597.931708] usb 2-1.2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
```

Remarks : Stays in this mode for about 7-8 seconds then boot normally.

Ralim, 2023-06-18:
Well those usb ID's match wch-ic.com chips as expected. Good to see we have a usb bootloader there.

For next steps in IronOS at a _minimum_ we need pinout information and general iron architecture information.
Basically this comment https://github.com/Ralim/IronOS/issues/1420#issuecomment-1574356410 through to https://github.com/Ralim/IronOS/issues/1420#issuecomment-1574690791 worth of information.

At the moment I dont have one of these units (or much time) to do this tracing.
If someone with the iron can do a rough pinout I can look at scheduling support.

Jonas-fr, 2023-06-18:
> PCB board
> 
> [image]

@cvetaevvitaliy Did you disassemble yourself your unit to tkae this picture? If yes would you mind describing the steps you've taken to do so? I'm trying to open mine and wouldn't want to break anything if possible. Thanks

StevenSeifried, 2023-06-24:
Yesterday, I also bought a Fnirsi HS-01. For 26 EUR/~29 USD, I couldn't say "No" to it ^^
I have now also written an email to Fnirsi regarding the support of IronOS, I am waiting to see what they will answer.

KaMyKaSii, 2023-08-07:
Kaiweets is selling the same soldering iron under the name KETS02. Maybe someone wants to try their luck regarding the scheme with them.

StevenSeifried, 2023-08-07:
@KaMyKaSii I have written to Kaiweets. Let's see what Kaiweets answers ^^

Jonas-fr, 2023-08-17:
> > PCB board
> > [image]
> 
> @cvetaevvitaliy Did you disassemble your unit to take this picture? If yes would you mind describing the steps you've taken to do so? I'm trying to open mine to get the pinout and wouldn't want to break anything if possible. Thanks

Hi @cvetaevvitaliy I guess you have more important things to do right now and that's OK, just don't forget to get back to us regarding this question whenever you'll have the time.

Klbgr, 2023-08-18:
> > > PCB board
> > > [image]
> > 
> > 
> > @cvetaevvitaliy Did you disassemble your unit to take this picture? If yes would you mind describing the steps you've taken to do so? I'm trying to open mine to get the pinout and wouldn't want to break anything if possible. Thanks
> 
> Hi @cvetaevvitaliy I guess you have more important things to do right now and that's OK, just **please** don't forget to get back to us regarding this question whenever you'll have the time.

Hi, I just got mine but it came broken so I contacted Fnirsi on What's App and they provided a video showing how to disassemble it.
Here's the video:
https://github.com/Ralim/IronOS/assets/43754408/a1d946df-8a6d-4245-b54d-c85efa1321ae

ggaljoen, 2023-10-15:
Got my hands on this beauty: now I can start to gather the required information to get IronOS on it
[image]

Careless, 2023-10-15:
excited!

I was using it yesterday and came across an issue where the pick-up re-heat sensitivity is way too high and the weight of the included USB-C silicon cable causes the device to roll around ever so slightly in the tool holder / iron rest on the desk and starts the heating process again or never lets it idle. every time i looked at the screen it was at 300+ degrees, barely turning off. i don't think the elements will last long this way.

Also, upon plugging in, pressing the left-side button to start heating when prompted immediately heats up to the last set temperature without telling you what it actually is before you press it. you press the button and it instantly becomes a lightsaber.

Those are the only two things I found issues with, otherwise this iron is perfect for its size and power.

I was using a PD65W capable battery bank and the iron heats up to 380C in about 3-4 seconds from rest temperature.

really solid iron and worth the investigation.

ggaljoen, 2023-10-15:
More details:
[image]
[image]

Danimal4326, 2023-11-09:
> Got my hands on this beauty: now I can start to gather the required information to get IronOS on it [image]

Any updates?

ggaljoen, 2023-11-09:
Got not enough time in the past weeks; next periode should be better. More information/spare time is on it's way...

ggaljoen, 2023-11-19:
[image]
[image]
[image]
[image]

ggaljoen, 2023-11-19:
Identified parts:
GBG3N		S-8261
S2VH		ME6211
U1			CH224K
U2			CH571F
8831		??? LMV331
Q1			1AM.
Q2			7409
Q3			1AM.
LCD			LT1316P02C JY VG-2832TSWG02 i2c oled 128*32
X160		??? fuse
7409		AON7409

vovogod, 2023-11-25:
Hello!!! When the heating is turned on, the readings jump. Is there a solution to this problem? https://youtube.com/shorts/ClH-eTRLFMA

Careless, 2023-11-25:
> Hello!!! When the heating is turned on, the readings jump. Is there a solution to this problem? https://youtube.com/shorts/ClH-eTRLFMA

why are you heating it up with the cover on?

try cleaning the contact rings on the soldering tip you are using with some iso alcohol.

vovogod, 2023-11-25:
The lid does not affect the heating. That just did not do anything does not help.

Careless, 2023-11-25:
> The lid does not affect the heating. That just did not do anything does not help.

this is not a HS-01 tech support thread.

this is a thread to discuss custom firmware for the iron. 

if your iron is not operating correctly, contact who you purchased it from and ask them what to do.

ggaljoen, 2023-11-25:
First try for the schematic form:
[image]
Remarks very welcome!

ggaljoen, 2023-11-26:
KiCad schematic 80% done: 
[image]

ggaljoen, 2023-11-26:
Firmware (loader) for the Kaiweets KETS02 on G-drive: https://drive.google.com/drive/folders/1qK0RvTpXdFLVqxrZ_p9_Jd7tAUvruVWU

Does anyone have reverse engineering skills to figure out the temperature calculations?

ggaljoen, 2023-11-26:
Temperature calibration (encrypted Firmware?) and Firmware Upgrade Tool on G-drive: https://drive.google.com/drive/folders/1qK0RvTpXdFLVqxrZ_p9_Jd7tAUvruVWU

HS-01 won't "Burn" right away (brand/ID/PID/VID checks?)...

Does anyone have reverse engineering skills to figure out the temperature calculations?

ecsv, 2023-11-27:
> Temperature calibration (encrypted Firmware?) and Upgrade Tool on G-Drive:

Looks like just XORed with 'V'.

Can be easily "decrypted" in python via:

```
#! /usr/bin/env python3

encrypted = bytearray(open('KETS02 temperature calibration_V0.1.s19', 'rb').read())
decrypted = bytearray(b ^ 0x56 for b in encrypted)
open('decrypted', 'wb').write(decrypted)

```

Barracuda72, 2023-11-27:
> Looks like just XORed with 'V'.

Same applies to the original HS-01 firmware, it's just XORed with 'n' (0x6e). First 0x20 bytes in the file is some kind of a header, firmware follows next.

ggaljoen, 2023-11-27:
> Same applies to the original HS-01 firmware, it's just XORed with 'n' (0x6e). First 0x20 bytes in the file is some kind of a header, firmware follows next.

Stupid me, looked over that in this topic. 
Is there a code decipherer that could figure out the temperature calculation/control values?

Rados51, 2023-12-04:
> Hello!!! When the heating is turned on, the readings jump. Is there a solution to this problem? https://youtube.com/shorts/ClH-eTRLFMA

You need to `burn` the tip. Leave it on for 5-10 minutes, and it should fix itself.

Careless, 2023-12-07:
what do you mean it's quicker? at doing what, specifically?

how did you install this new firmware?

AlexandarSteler, 2023-12-07:
It's faster-stable when making / going with temperature from scratch . It doesnt oscillate etc
Except that i downloaded from this link everything https://kaiweets.com/products/kets02

Careless, 2023-12-07:
interesting... i didn't really see much oscilating on the FNIRSI firmware. 

is the interface exactly the same as FNIRSI or are the graphics and menu different too?

AlexandarSteler, 2023-12-07:
Everything is the same except loading name 
And now I am trying in the code to change name to my name hehe .
[image]

AlexandarSteler, 2023-12-07:
[image]

enestr74, 2023-12-23:
Hi I'm new to this forum. I have a problem with my soldering iron a little off topic. As a result of the wrong voltage, the GBG3N component on the board has failed. So I bought one and when I removed the GBG3N on it and installed it on the defective one, the soldering iron started working again. I want to order GNG3N and use both of my soldering irons, but there is no component with this name on the internet. I would appreciate it if you could help what do you think this component could be or what could be used instead of it I apologize in advance for keeping the forum unnecessarily busy.

Axweal, 2023-12-24:
> Привет, я новичок на этом форуме. У меня проблема с паяльником немного не по теме. В результате неправильного напряжения вышел из строя компонент GBG3N на плате. Итак, я купил один, и когда я снял с него GBG3N и установил его на неисправный, паяльник снова заработал. Хочу заказать GNG3N и использовать оба своих паяльника, но в интернете нет компонента с таким названием. Буду признателен, если вы подскажете, что, по вашему мнению, может быть этим компонентом или что можно использовать вместо него. Заранее прошу прощения за чрезмерную загруженность форума.

"GBG3N"  =>  "S-8261"

Axweal, 2024-01-18:
> Схема KiCad готова на 80%: ![изображение](https://private-user-images.githubusercontent.com/7492159/285669399-7475cd0e-d0d1-4e29-8c25-7dd64afa8d2d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU1ODcxNjEsIm5iZiI6MTcwNTU4Njg2MSwicGF0aCI6Ii83NDkyMTU5LzI4NTY2OTM5OS03NDc1Y2QwZS1kMGQxLTRlMjktOGMyNS03ZGQ2NGFmYThkMmQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDExOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMThUMTQwNzQxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZTc4NjBkODdiZWQ4MDUwZGFlMjBiNDc3Mzc4MzFkMmIzYzZkNTdhYjhjMzJhMjA1MjBmMWE3ZGRjOGUyNDZiYyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.L31pYu_OqrFYabQD9fykQLGVxtH2c-rJnRRWMkZAIGA)

What's up

Axweal, 2024-01-18:
Fnirsi replied that they will not optimize the firmware on the HS-01 as it is not relevant. They are already developing another soldering iron. We were abandoned like puppies

Mrcrysis555, 2024-02-11:
Hi, I’m new in this forum. I have fnirsi hs-01 When connecting the power to the adapter, there was a small short circuit, now the soldering iron does not turn on and consumes 0.012amper and that's it all. What could have happened and how to fix it?

Ralim, 2024-02-11:
Please contact the manufacturer, this is not a support forum for their devices unfortunately

bybarbaros, 2024-02-24:
Hi friends. I bought the Fınıs 01 soldering iron set. I tried it for testing purposes. It worked beautifully. When I wanted to turn it on on the second day, the device did not turn on. The screen was turning on and freezing. I contacted the company I purchased it from. They somehow sent me the bios, I installed the software, but nothing changed. I contacted the company again and explained the situation. According to what they told me, my power supply was not enough. The fact that the machine I used after trusting it did not turn on the next day could not be a power supply problem. Then our correspondence continued. Finally, I tried running it with different power supplies. And finally, I made a mistake and burned the device by applying reverse current to it. I opened the device and found that the "GBG4P" integrated inside was burned. Unfortunately, I could not find this integrated device in electronics vendors in my country. My searches on the internet were fruitless. I'm waiting for your help. Thanks in advance for everything.
[image]
[image]
[image]

trantuananh1996, 2024-03-05:
> Hi friends. I bought the FNIRSI HS-01 soldering iron set. I tried it for testing purposes. It worked beautifully. When I wanted to turn it on on the second day, the device did not turn on. The screen was turning on and freezing. I contacted the company I purchased it from. They somehow sent me the bios, I installed the software, but nothing changed. I contacted the company again and explained the situation. According to what they told me, my power supply was not enough. The fact that the machine I used after trusting it did not turn on the next day could not be a power supply problem. Then our correspondence continued. Finally, I tried running it with different power supplies. And finally, I made a mistake and burned the device by applying reverse current to it. I opened the device and found that the "GBG4P" integrated inside was burned. Unfortunately, I could not find this integrated device in electronics vendors in my country. My searches on the internet were fruitless. I'm waiting for your help. Thanks in advance for everything. ![IMG-20240213-WA0005](https://private-user-images.githubusercontent.com/118777939/307528101-639cb2dc-5b70-4b84-8552-5ddbf845b448.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk2MjUzODMsIm5iZiI6MTcwOTYyNTA4MywicGF0aCI6Ii8xMTg3Nzc5MzkvMzA3NTI4MTAxLTYzOWNiMmRjLTViNzAtNGI4NC04NTUyLTVkZGJmODQ1YjQ0OC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMwNVQwNzUxMjNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01OWYwNDYwZDBjM2FkYzBiZTJkM2JiZTQwYjY1YjU5NjQ5MzEyYzdjNWVkMjgzMTUwNmQ1MTgzNGYyNWNmYWMzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Lv0wmG6gl-3ODry_IDe_DMoj77OcZF5jwI_2i_jAh8U) ![IMG-20240213-WA0009](https://private-user-images.githubusercontent.com/118777939/307528103-3aafa9a2-a6fa-457a-b2f7-b86fec323342.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk2MjUzODMsIm5iZiI6MTcwOTYyNTA4MywicGF0aCI6Ii8xMTg3Nzc5MzkvMzA3NTI4MTAzLTNhYWZhOWEyLWE2ZmEtNDU3YS1iMmY3LWI4NmZlYzMyMzM0Mi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMwNVQwNzUxMjNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kZTJhODVkNDg5ZWEwOWRmOTNmMzc3MzNiMDg1MDY1OGZmM2Q1NDk4NmUwNmQ5NmVmMTQ1YmU5M2Y5YjA3MjViJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ziUKZnpsbUdhzrp6QTNTW-q4dbpAcrY0EA3gbRtxHVo) ![IMG-20240224-WA0001](https://private-user-images.githubusercontent.com/118777939/307528106-cd0ca4fa-03df-4716-8ba5-1168b051136d.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk2MjUzODMsIm5iZiI6MTcwOTYyNTA4MywicGF0aCI6Ii8xMTg3Nzc5MzkvMzA3NTI4MTA2LWNkMGNhNGZhLTAzZGYtNDcxNi04YmE1LTExNjhiMDUxMTM2ZC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMwNVQwNzUxMjNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zMjkwMjllM2IyNzcwYTVhYjIwYjU4ZWQwZmM2NGM5YmZiYjEwYjBlZTZhMjhhOGQ5MmY4MDBmMWFkZWNiNmU3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Ou2nMFyJzJVjuT4usTuBm41OYxQGnJpopevhbb7aWZM)

It should be a SOT23-6 power management ic. RY8310

crt396, 2024-03-28:
Hello. Maybe someone can help

I recently purchased HS-01. When I first connected it (first connected it through a 5v power supply), it showed me that the voltage was low. I connected it to a 9v 2a power supply and it turned on at first and showed an icon that I needed to connect the tip, but after connecting the tip it stopped responding to any current source. Just a black screen and does not respond to any action. Please tell me what can be done and what kind of problem could be.

I look forward to hearing from you.

discip, 2024-03-28:
As stated 3 posts prior:

> Please contact the manufacturer, this is not a support forum for their devices unfortunately

joric, 2024-04-23:
There is HS-02 available. https://aliexpress.com/item/1005006655028971.html Color HD OLED screen, JBC 245/210 tips can't wait writing tetris for this thing. have no idea what CPU it uses probably the same as FNIRSI HS-01 (CH571F) so it may have BLE also.

[image]

[20 more replies]

closed 2026-10-27 by Ralim