Ralim/IronOS #1615: TS80 stuck at 5V  using V2.20,  Is TS80 still supported?

Yagoth, 2023-03-10:
**Describe error.**
My TS80 is stuck at 5V using build 2.20. It will report 9V immediately after being plugged in, but after 1second to 5 minutes it will drop to 5v and get stuck there.

**For reproduction**
Be using V2.20 EN
Plug in TS80
Wait 1second to 5 minutes.
Switches and gets stick in 5v mode

**Details of your device:.**

    Device: TS80
    Release: 2.20.14DAF70 13-12-22 EN
    Power adapter(s) used:  
     1. DN. OX DNQ18U (shipped with device)    QC 3.0  9V-12V  1.5 a
     2. Anker PB-Y86                                          QC 3.0  9V-12V  2 A
     3. Anker A1264                                           QC 3.0  9V-12V  2 A
     all explicit support QC 3.0 and work with my galaxy S8

Extra Notes:
I found an old post relating to V2.10 relating to the same issue. https://github.com/Ralim/IronOS/issues/1107

Any idea on a fix?  I kind of get the feeling that TS80 is not even supported at this point, given that TS80P and Pincil sole is market segment.   
If it is no longer supported what firmware should I use?

I can confirm that I an using V2.20 not the stock firmware.  Images attached.

[image]
[image]

gamelaster, 2023-03-10:
Hello,

TS80 is still supported, and this should not happen.

Please, do you have available any power analysis tool?

Yagoth, 2023-03-10:
I am not entirely sure what you mean by power analysis tool.  I have the USB power meter(pictured below) and a decent multi-meter.  I have also confirmed that the temperature reading is accurate.

Power reading I have obtained:
4 patterns occur:

1. Starts 5v 0.9A, then flutters between about 0.2A and 0.8A while trying to reach set temperature and can't go past about 270c 
2. Starts at 9V 1.8A  reaches temperature (currently 340c) then goes to 5v .95A after a few seconds and slowly cools down to around 270c (I can't get it to return to 9 V, even by letting it go to sleep and letting it reheat).  
3. (Harder to test) It Starts at 9V 1.8 A  and I use it immediacy which seems to keep it on 9V for longer, but if I put it down it come back to find it at 5V.  I think this is because me using it make the it harder for the iron keep the temperature at the target.
4. Starts at 9V then immediacy switches to pattern #1

[image]

gamelaster, 2023-03-10:
@Yagoth yes, multimeter and mostly, USB-C meter. And great, those information  will be very useful.

@Ralim is currently out of the office, so he will take look on this on end of this month.

discip, 2023-09-23:
@Yagoth
Is this issue still present?
Have you tested the latest release?

Yagoth, 2023-09-30:
Issue is still present, thank you for following up.  I will install 2.21 when I return home from my trip.

discip, 2023-09-30:
Please try this one: https://github.com/Ralim/IronOS/releases/tag/v2.22-rc2