Ralim/IronOS #1620: Screen Brightness  collapse 1-10 into  levels 1-5  (eliminate level 1 as it makes some sceens BLACK)

River-Mochi, 2023-03-13:
**Is your feature request related to a problem? Please describe.**

1. people report  that when they bring screen brightness to level "1" the screen goes black. one of the people is builder555 (PineSAM).  Screen should not go black  when you bring Pinecil v2 to "1" but we have seen this on several models now brand new from factory.

2.  one person who had this happen  got a little scared that their new V2 was broken when they were  going through settings menu changing the brightness,  and instead of  pressing (+)  a couple times to get back to a higher brightness  that is visible,  started Rapidly and Randomly pressing  +/-   (fear happens). 
 Flashing did not help them since flashing  keeps all the same settings and we don't have a "blank.bin" kind of solution yet.

3. I had them flash a BLE version of IronOS, and then with **PineSAM**  they were able to adjust their Screen back to 4 so they could see it again. 

** Solution Proposed **

1. Could you let Pine Store know that whatever is going on hardware-wise, level 1 screen brightness is **BLACK** on some of these and people think they have a broken pinecil and since reflashing doesn't fix it, they get scared.
2.  Thanos suggested that that using a different Capacitor under the screen could  fix this and they are cheap too.  best to talk to thanos on this, but he had a  specific idea in mind of what kind they need to change for the screen to not go black at  level "1" and overhall be better quality of  brightness.
3.   Is there is anything that can be done in firmware to  make "1"  not get so low that it's black screeen?  

**Describe the solution you'd like**
In addition, I propose  we only have levels 1-5  on brightness **(collapsing the 10 levels into 5)** since on Pinecil V2, people can not see any difference in many of the middle levels.
we tested with several people and there is no visual difference from levels  2-8, therefore  could firmware be changed so that there are only  5 total levels since all the other levels show no difference in brightness?
I  can see "1" and "10" are different. realistically, even with all the room lights off, the 2-8 had no change to the eye. and "9" gets slightly brighter.

- Whatever 1-2 is , eliminate "1" level since it's too low for some people to see and screen is black. make  the new "1" whatever old 2 was.
- 1-2 (specifically 2) becomes 1
- 3-4  (specifically 4) becomes  2
- 5-6 (specifically 6) becomes 3
- 7-8 (sepcifically 8) becomes 4
- 9-10 (sepcifically 10) becomes 5

In my video, 10  is going to be unusally bright because that is a brand new replacement screen I put on a V1.  
TBH,  my  level "5"  on the new V1 screen is the same brightness as level "10" on my  other V2 pinecil. 

I'm using 3 different pinecils all at the same time in this video so it's not a great control.  it should have been one single pinecil filmed at different levels. 
https://user-images.githubusercontent.com/97197236/224631164-3271838a-47ea-4f55-b9bf-65c093f1c2ca.mp4

discip, 2023-03-13:
@River-Mochi
Please give this a go: [readjusted OLED brightness](https://github.com/discip/IronOS/suites/11530220246/artifacts/596450836)

River-Mochi, 2023-03-14:
thanks
@buider555 are you able to test this  and see if your "1"  is no longer black?

randy408, 2023-03-14:
> @River-Mochi No hardware changes are required as this can be solved by software.
> 
> Please give this a go: [readjusted OLED brightness](https://github.com/Ralim/IronOS/suites/11536568535/artifacts/596897737)

This fixes the issue with the screen going black at level 1 and the brightness steps are more noticeable.

River-Mochi, 2023-03-14:
> This fixes the issue with the screen going black at level 1 and the brightness steps are more noticeable.

Randy,  when you say the steps of brightness are more noticieable do you mean from 2-8  you can see the difference more now?   that is good if yes.

River-Mochi, 2023-03-14:
@discip 

@builder555   confirmed that this fixes  pinecil V2  being black at level "1" 
could this be part of  2.21  release?
[image]

River-Mochi, 2023-03-14:
just tested it myself.  I like the change a lot, I can actually see difference now between steps 1-2-3-4-5.  this is easier to use also than having 10 steps with mostly no viisble difference.

Reviewed, Tested  and good to go!

closed 2023-03-14 by River-Mochi

builder555, 2023-03-14:
confirmed. works on mine.

PR #1622 (2023-03-26): Readjusting brightness (Pinecil v2)
Resolves #1620