Ralim/IronOS #1630: Thank you for bricking my soldering iron

henrikra, 2023-03-19:
**Describe the bug**

Just thanking that now my soldering iron is bricked. Well it was nice to use it at least one time.

**To Reproduce**

**Expected behavior**

**Details of your device:**
 
 - Device: [e.g. TS80/Pinecil etc]
 - Release: [eg 2.15.40087E6]
 - Power adapter being used:
 - If this is an accelerometer related issue, please include its model number here:

**Additional context**

CRImier, 2023-03-19:
As an onlooker, I have to ask - have you considered unbricking it? From what I've seen, that might be closer than you realize, unless you've missed one of the "this will brick your iron" warnings.

gamelaster, 2023-03-19:
It would be handy if you would explain your issue more. For example, which kind of device you have. Also, 95% of times that device is really bricked (and that happens rarely), it is fault of hardware, not software. So give more information, and we might be able to help you to recover your soldering iron.

henrikra, 2023-03-19:
TS100.

After flashing won't wake up with battery. Only thing that still works is the DFU: 3.49 when starting with button A pressed when connecting with micro usb.

gamelaster, 2023-03-19:
@henrikra DFU mode means that chip responds, that's good.
Did you tried older version? Also, are you sure you 100% correctly followed the instructions?
(NOTE: I don't have TS100, so I don't know all the issues and stuff around it, but if DFU works, it means chip is OK.)

henrikra, 2023-03-19:
Yes I have tried flashing back the old version from TS100 firmware page but still no luck. Yes I have checked multiple times that I have followed the instructions

Ralim, 2023-03-20:
Hello,

You have not bricked your device at all, it's completely recoverable.

The problem you are running I to us issues with the Miniware DFU bootloader they ship combined with the clone stm32 chips they are using.

There are other issues already around about this problem, people have generally had to try under Linux or on another machine for it to work.

I would suggest reading other issues around ts100 programming failing. This is the exact reason that it's no longer recommended to purchase the ts100.

I would suggest focusing on flashing the Miniware firmware from their forum back first, to make sure it's working for you. If that fails programming raise a support issue with Miniware. (I dont have the source code to their DFU firmware; I can't fix their bugs or provide support)

henrikra, 2023-03-20:
Alright thx, have to try with another computer. What soldering iron is recommended nowadays if not TS100?

gamelaster, 2023-03-20:
@henrikra Most well supported and maintained iron is Pinecil V2.

henrikra, 2023-03-21:
I tried with desktop computer and then the flashing worked. For some reason the laptop flashing did not work.

closed 2023-03-21 by henrikra