Ralim/IronOS #1639: Windows 11 Can't load .hex file.

pbouna321, 2023-03-31:
Hello all! I am having an issue where I can enter DFU mode, but once I try and send over the .hex file my explorer disconnects the device, and I get...

An unexpected error is keeping your from copying this file...
Error 0x800701B1

When the device shows back up in Explorer it is now called USB Drive and when I hit try again on the file I get the message that the file is too large for the destination file system.

In CMD using the copy function I receive
A device which does not exist was specified.
        0 file(s) copied.

Any ideas?

pbouna321, 2023-03-31:
Update:

Used Ubuntu 18.04 from USB with the try out feature and it worked in .4 seconds lol.

Damn windows.

pbouna321, 2023-03-31:
> what Iron did you have it's not in the info. many irons are supported by IronOS and they all update with a different procedure.
> 
> I'm not sure it was windows, it could have been port you were using. for example my pinecils I can use the Usb-a ports in my laptop but not the USB-C ports. but on my windows PC I can use all the ports and it updates.

Sorry, just updated. I have a TS80P. I tried both a USB-A to USB-C and a USC C to USB C both from back panel of MB with no luck.

Ran Ubuntu from flash and it worked right away.

River-Mochi, 2023-03-31:
> Ran Ubuntu from flash and it worked right away.

good to know.  since you fixed this now  would  you mind closing the ticket?  ;)

closed 2023-03-31 by discip