Ralim/IronOS #1640: Add Bluetooth Icon to Pinecil V2 Firmware

River-Mochi, 2023-03-31:
**Is your feature request related to a problem? Please describe.**

- Pinecil V2  Only. 
- Add this BLE  bluetooth  icon to  the Advanced menu

[image]

**Describe the solution you'd like**
- Just  this icon and the word  Bluetooth   or if that doesn't fit  then this Icon and word BLE.

thanks

Ralim, 2023-03-31:
I'm inclined to lean towards keeping the menus text based; as it's more uniform to the other menus, and it keeps things simpler to avoid having custom draw handling where we can.

What's wrong with having the menu called Bluetooth, and in the help text saying it enables BLE?

discip, 2023-03-31:
I am sorry to disappoint you, but the same applies to the expression itself as to the symbol.
**`Bluetooth`** is unique in all languages.

discip, 2023-09-23:
Closing as the main developer is inclined to 
> keeping the menus text based

closed 2023-09-23 by discip