Ralim/IronOS #1644: Low amperage PSU causes OSD glitching

Jerrk, 2023-04-03:
**Describe the bug**
I was trying to use my old 2A power supply from my TS100, but it was glitching the OSD whenever it would try to heat up. I'm posting this issue here by directions of @River-Mochi 

**To Reproduce**
Using a low amperage PSU (2A) try to heat the pinecil tip

The OSD will start to glitch out

**Expected behavior**
An error message about low amperage could be useful in this situation - if that information can be made available via the firmware.

**Details of your device:**
 - Device: Pinecil V2
 - Release: v2.20.14AF70
 - Power adapter being used: TS100 - KJS60A-2402000

**Additional context**
Video of the issue here:

https://user-images.githubusercontent.com/49324581/229501417-7e636fa2-8c7c-4d29-9ffc-b327eb613e97.mp4

River-Mochi, 2023-04-03:
thank you for sharing this good example.  this has been happening to a few Pinecil V2 owners, trying to use a low amp power supply is causing Glitchy  reboots or  even wild temperature  jumping  when going to set temperature.

- pinecil V2  minimum rating is  3amps (more is ok too so 3A-5A is good).
- don't exceed 24V  if using the dc barrel  jack
- avoid all cheap universal type DC barrel chargers, they are "dumb" bricks and tend to over-spike voltage which can kill pinecil.

**Solution**: swapping to  a USB-C  20V,  **3.25amp**  (PD65)  charger  fixed problem for you and several other Pinecil V2 owners over the past few months.

River-Mochi, 2023-04-03:
Here is another member video (anker 511, Nano 3, 30W charger is not recommended).
https://www.youtube.com/watch?v=26Dq6a11_9E

This was caused by Anker 511 Nano3, 30W  phone charger. it struggles at Set temperature and jumps all over the place because not enough amps. 
[image]

2 days later he got a new USB-C 20V, 3.25 amp (PD65w) charger from Amazon, and everything was good.

closed 2023-07-23 by Ralim