Ralim/IronOS #1655: BLE: is more unique names gone from Pinecil V2?

River-Mochi, 2023-04-07:
**Describe the bug**

BLE "more" unique name is no longer showing up in BLE apps for Pinecil V2.
"None" in this screenshot used to have the "more unique"  name.

[image]

using the newest  2.21 pre-realease beta, and one from yesterday.

Ralim, 2023-04-07:
Names work perfectly fine for me here scanning with laptop or phone. Also there is no way for the C code to generate the name `Pinecil-None`; the `None` makes me highly suspect your looking at a bug in the app your using to connect to the device.

closed 2023-04-07 by Ralim

River-Mochi, 2023-04-07:
> Names work perfectly fine for me here scanning with laptop or phone. Also there is no way for the C code to generate the name `Pinecil-None`; the `None` makes me highly suspect your looking at a bug in the app your using to connect to the device.

none is from our side yes.  Tom is also having issues getting the  unique name in his ESPhome app  so I was wondering it  something changed.

River-Mochi, 2023-04-07:
Closing ticket,  all good now. 
issue on pineSAM  side  is  UUID  was in more than one place in code and one of the hardcoded variables was missed.

[image]