Ralim/IronOS #1656: Quick Start guide bluetooth communication with IronOS.

River-Mochi, 2023-04-07:
I'd like to write a Quick Start Bluetooth LE article to help people get started connecting bluetooth Apps to IronOS. 
basic things they need so they don't waste time hunting around for it.  I would provide things such as a link to a list of all UUID with names that they need, important things like that.  Probably going to be a short stub for now.

I'm not an expert in this  at all, and hopefully you or some of the people working on BLE apps now will add a little more if I start it. 

Goal is to 
(a)  prevent some tickets and questions  to IronOS  if  the basics are covered.
(b) encourage more people to make BLE apps to work with IronOS. 
(c) give some resources if they get stuck.

warlock31415, 2023-04-08:
+1 for guide. Was thinking of making a stand for the Pinecilv2 with a oled display to essentially mirror the on board screen. This can be done over the Debug info being spit out of the uart but I also wanted to add temp control using an encoder. Can't do that over uart without adding a bunch of command parsing functions over uart.

My next thought was PineSam already has control with BLE so maybe can make something with a NRF52. So documentation would be highly appreciated.

River-Mochi, 2023-04-08:
waiting for Ralim to approve before I start it in IronOS  docs.

River-Mochi, 2023-04-08:
> +1 for guide. Was thinking of making a stand for the Pinecilv2 with a oled display to essentially mirror the on board screen. This can be done over the Debug info being spit out of the uart but I [...]

Look into Tom's project also  for this.  
He is sending bLE data from Pinecil to ESP32, and then that is using Home assist. and then  he has a DIY fan  which automatically comes on if Pinecil is "heating up".    And he is now also trying to implement some of the settings features   that are in PineSAM app.  I added all project  to Development in[ Pinecil wiki here](https://wiki.pine64.org/wiki/Pinecil#Development_Projects).

River-Mochi, 2023-04-08:
@warlock31415  - could also right now use an old android/iphone with PineSAM right now and show all pinecil display too?  that is what I do, old phone in phone stand which I can also tap on screen to change Pinecil settings.

-  I personally tested the buttons on phones to make sure pinesam buttons made big enough to work even if you have nitrile gloves on while soldering. much better accessibility for people with vision issues so they don't have to keep switching from magnifying glasses/lens to  reading glasses while soldering.

warlock31415, 2023-04-08:
@River-Mochi It looks like Home assistant does all the heavy lifting on the ESP32 project. I just want something simplistic. Something that mimics weller base stations. I also want to get into learning the nordic dev system

aguilaair, 2023-05-08:
I am working on a companion app for phones currently: https://github.com/aguilaair/IronOS-Companion

Currently connects and visualizes data, next step is sending commands to the iron

[image]

aguilaair, 2023-05-09:
Hey @gamelaster will do! I don't think it's fully ready yet to be pushed to the app stores, still need a couple of weeks to finish it up :)

Ralim, 2023-06-14:
Correct and deliberate for safety at the moment.

gamelaster, 2023-06-14:
@Ralim & @aguilaair I think STOP command would be useful and not that big safety risk, although, START might be, but that might be solved by showing some prompt, so user actually needs to confirm the START with button (but that misses the whole point of having it wireless).

agibson2, 2023-07-27:
This app looks amazing so far.  Was it ever pushed to the apple store?

gamelaster, 2023-07-27:
It will be hopefully @agibson2

PR #1763 (2023-08-01): BLE getting started document

* **What kind of change does this PR introduce?**

Updating Docs for BLE.
Closes  #1656 

* **What is the current behavior?**

* **What is the new behavior (if this is a feature change)?**

* **Other information**:
closed 2023-08-01 by Ralim