Ralim/IronOS #1659: Possibly a typo in 128-bit BLE UUIDs

joric, 2023-04-08:
@Ralim I have updated my GUI tool yesterday and it works but I'm just wondering why settings UUID start from f6d8 but characteristics start from f6d7. is it intentional or is it a typo? Top 4 digits for BULK data and BULK data UUIDs match.

From https://github.com/joric/pinecil (it works with the latest 2.21, but it's kind of strange).
```js
  BT_UUID_SVC_BULK_DATA                        = '9eae1000-9d0d-48c5-aa55-33e27f9bc533'; // bulk service
  BT_UUID_CHAR_BLE_BULK_LIVE_DATA              = '9eae1001-9d0d-48c5-aa55-33e27f9bc533'; // <-- matches service UUID

  BT_UUID_SVC_SETTINGS_DATA                    = 'f6d80000-5a10-4eba-aa55-33e27f9bc533'; // settings service
  BT_UUID_CHAR_BLE_SETTINGS_VALUE_SAVE         = 'f6d7ffff-5a10-4eba-aa55-33e27f9bc533'; // <-- does not start from f6d8
  BT_UUID_CHAR_BLE_SETTINGS_VALUE_SETPOINT     = 'f6d70000-5a10-4eba-aa55-33e27f9bc533'; // <-- does not start from f6d8
```

Ralim, 2023-04-08:
>Note there's also a comment with an old UUID // f6d75f91-5a10-4eba-a233-47d3f26a907f why it's there? Just for history?
yeah, will probably drop that

I split the settings to be in a different base uuid just in case we end up with a larger set of them; as its most likely to grow and get messy. I want to automate the creation of the ble uuid table for the settings too but it hasnt yet worked for me.

joric, 2023-04-08:
@Ralim when it was standard UUDS I just used BluetoothUUID.canonicalUUID(1); to define a characteristic in js so maybe just use a function, those text labels eat an awful lot of space.

closed 2023-04-08 by joric