Ralim/IronOS #1661: Pinecil v2 doesn't work after updating to v2.21

sorgelig, 2023-04-09:
**Describe the bug**
After flashing by blisp to v2.21 my pinecil v2 looks completely dead.

**To Reproduce**
1. Flash v2.21
2. iron looks dead, no display on screen

**Expected behavior**
normal work

**Details of your device:**
 
 - Device: Pinecil v2 bought on pine64 site, so should be original one.
 - Release: i don't know how to see. in flash mode it doesn't show anything on screen.
 - Power adapter being used: tried both Type-C and 5525

**Additional context**
v2.21 it seems doesn't work on my Pinecil v2. Thankfully, flashing v2.20 brings iron to life again.

sorgelig, 2023-04-09:
**Update:** just passed authenticity test (just for sanity). Manufacture date: February 27, 2023.
Not sure if it's ok to post ID and SN here, but should be irrelevant to issue as long as it's original.

gamelaster, 2023-04-09:
No need to post it here. This looks like software issue, as rollback to v2.20 helps.

sorgelig, 2023-04-09:
There is no sign of exactly software/updater issue. Broken firmware also can lead to such state.

River-Mochi, 2023-04-09:
> There is no sign of exactly software/updater issue. Broken firmware also can lead to such state. Ability to flash any version of firmware comes from bootloader, and has nothing to do with software.

What  OS  are you using for the update  and type of computer?

River-Mochi, 2023-04-09:
Is  there any possibility   you might have accidentally not used the BIN  file when you updated?

if for example  you   flash the 2.21  **DFU**  or HEX   file , then you will get black screen also.   I did this before and I had to flash again with BIN  file to fix it. 

try it one more time  to be sure and make sure to follow instructions in Wiki page.   You can't really brick the pinecil doing this so it is okay to try  2-3  times.  

https://wiki.pine64.org/wiki/Pinecil_Firmware#Update_V2:_Windows

gamelaster, 2023-04-09:
@sorgelig by software I mean firmware, but I think it can be also flasher issue.
Older `blisp` version had issue with bigger firmwares (those with Bluetooth), so be sure you have latest blisp.

River-Mochi, 2023-04-09:
Get the **newest** BLISP here.  **https://github.com/pine64/blisp/releases/tag/v0.0.3**

updated  two Pinecil V2, no problems with 2.21  so far.   other people also  in pinecil chat say it's good for them. 
<br>
[image]

River-Mochi, 2023-04-09:
> Older `blisp` version had issue with bigger firmwares (those with Bluetooth), so be sure you have latest blisp.

Thanks for this information, we already have bunch of people updating this morning with different issues. so far all resolved and Success in the end.

sorgelig, 2023-04-09:
I'm using blisp 0.0.3. Windows 11.
Yes, i've tried Pinecilv2_EN.bin and also tried multi-language version bin - both give non-working iron. Also i've tried dev version which i've downloaded several days ago - also black screen. 
Only flashing v2.20 brings iron to life.

I don't know what else i'm missing...

River-Mochi, 2023-04-09:
do you  mean you also  tried the earlier  Beta a few days ago and also got a black screen?

sorgelig, 2023-04-09:
```
blisp.exe write -c bl70x --reset .\Pinecilv2_EN.bin
Sending a handshake...
Handshake successful!
Getting chip info...
BootROM version 1.0.2.7, ChipID: 00003DF5221017A8
0b / 59200b (0.00%)
4092b / 59200b (6.91%)
8184b / 59200b (13.82%)
12276b / 59200b (20.74%)
16368b / 59200b (27.65%)
20460b / 59200b (34.56%)
24552b / 59200b (41.47%)
28644b / 59200b (48.39%)
32736b / 59200b (55.30%)
36828b / 59200b (62.21%)
40920b / 59200b (69.12%)
45012b / 59200b (76.03%)
49104b / 59200b (82.95%)
53196b / 59200b (89.86%)
57288b / 59200b (96.77%)
59200b / 59200b (100.00%)
Sending a handshake...
Handshake with eflash_loader successful.
Erasing flash, this might take a while...
Flashing boot header...
Flashing the firmware...
0b / 189960b (0.00%)
2052b / 189960b (1.08%)
4104b / 189960b (2.16%)
6156b / 189960b (3.24%)
8208b / 189960b (4.32%)
10260b / 189960b (5.40%)
12312b / 189960b (6.48%)
14364b / 189960b (7.56%)
16416b / 189960b (8.64%)
18468b / 189960b (9.72%)
20520b / 189960b (10.80%)
22572b / 189960b (11.88%)
24624b / 189960b (12.96%)
26676b / 189960b (14.04%)
28728b / 189960b (15.12%)
30780b / 189960b (16.20%)
32832b / 189960b (17.28%)
34884b / 189960b (18.36%)
36936b / 189960b (19.44%)
38988b / 189960b (20.52%)
41040b / 189960b (21.60%)
43092b / 189960b (22.68%)
45144b / 189960b (23.77%)
47196b / 189960b (24.85%)
49248b / 189960b (25.93%)
51300b / 189960b (27.01%)
53352b / 189960b (28.09%)
55404b / 189960b (29.17%)
57456b / 189960b (30.25%)
59508b / 189960b (31.33%)
61560b / 189960b (32.41%)
63612b / 189960b (33.49%)
65664b / 189960b (34.57%)
67716b / 189960b (35.65%)
69768b / 189960b (36.73%)
71820b / 189960b (37.81%)
73872b / 189960b (38.89%)
75924b / 189960b (39.97%)
77976b / 189960b (41.05%)
80028b / 189960b (42.13%)
82080b / 189960b (43.21%)
84132b / 189960b (44.29%)
86184b / 189960b (45.37%)
88236b / 189960b (46.45%)
90288b / 189960b (47.53%)
92340b / 189960b (48.61%)
94392b / 189960b (49.69%)
96444b / 189960b (50.77%)
98496b / 189960b (51.85%)
100548b / 189960b (52.93%)
102600b / 189960b (54.01%)
104652b / 189960b (55.09%)
106704b / 189960b (56.17%)
108756b / 189960b (57.25%)
110808b / 189960b (58.33%)
112860b / 189960b (59.41%)
114912b / 189960b (60.49%)
116964b / 189960b (61.57%)
119016b / 189960b (62.65%)
121068b / 189960b (63.73%)
123120b / 189960b (64.81%)
125172b / 189960b (65.89%)
127224b / 189960b (66.97%)
129276b / 189960b (68.05%)
131328b / 189960b (69.13%)
133380b / 189960b (70.21%)
135432b / 189960b (71.30%)
137484b / 189960b (72.38%)
139536b / 189960b (73.46%)
141588b / 189960b (74.54%)
143640b / 189960b (75.62%)
145692b / 189960b (76.70%)
147744b / 189960b (77.78%)
149796b / 189960b (78.86%)
151848b / 189960b (79.94%)
153900b / 189960b (81.02%)
155952b / 189960b (82.10%)
158004b / 189960b (83.18%)
160056b / 189960b (84.26%)
162108b / 189960b (85.34%)
164160b / 189960b (86.42%)
166212b / 189960b (87.50%)
168264b / 189960b (88.58%)
170316b / 189960b (89.66%)
172368b / 189960b (90.74%)
174420b / 189960b (91.82%)
176472b / 189960b (92.90%)
178524b / 189960b (93.98%)
180576b / 189960b (95.06%)
182628b / 189960b (96.14%)
184680b / 189960b (97.22%)
186732b / 189960b (98.30%)
188784b / 189960b (99.38%)
189960b / 189960b (100.00%)
Checking program...
Program OK!
Resetting the chip.
Flash complete!
```

sorgelig, 2023-04-09:
> do you mean you also tried the earlier Beta a few days ago and also got a black screen?

yes

River-Mochi, 2023-04-09:
Could you  try this  old   Beta BIN   with Bluetooth from   Nov  2022 that  Joric still has on his website.

https://github.com/joric/pinecil/wiki

sorgelig, 2023-04-09:
> Could you try this old Beta BIN with Bluetooth from Nov 2022 that Joric still has on his website.
> This is pre-2.20 stable but has bluetooth.

doesn't work too.

River-Mochi, 2023-04-09:
Now we know you tried  Pre-2.20  with bluetooth(nov 2022 beta),  some of the 2.20xxxx  with bluetooth and  2.21 with bluetooth all cause  black screen.  And only  2.20 with No bluetooth on it works for you. 

That is the extent of my  knoweledge of what to try. need to wait for Ralim and Gamiee on this.

River-Mochi, 2023-04-09:
One more thing to Try...
I'm insterested if you install   2.21  black screen is fine.   then open PineSAM  on browser,  do you see information and  Debug stuff?  

2.21  all all those beta's should have Bluetooth on by default 

Pinesam here:   https://github.com/builder555/PineSAM   Download  the Premade =binary for windows and follow instructions under Premade Binaries.

River-Mochi, 2023-04-09:
watch the terminal windows when you  connect  to Pinesam. has important messages like if it Finds your pinecil  and that you can post screen shot .

sorgelig, 2023-04-09:
btw, about  Joric's pre-2.20 - it's not really pre.. it's pointed to releases/latest which currently is 2.21.

sorgelig, 2023-04-09:
I discovered, on v2.21, if i press button+ while connecting USB then sometimes iron starts (with debug screen) and then it works normally. It only happens sometimes, not every time. So i have to try several times to make it work. Without button, it never starts.
So, for me it looks like some booting process is buggy, it doesn't wait enough for HW to initialize and then get stuck.

sorgelig, 2023-04-09:
> Other thing is if you can get Pinesam up, Reset all Settings to Defaults. do that in 2.20 also.

i've already tried resetting before. It doesn't help.
Not sure if i want to mess with pineSAM. It involves additional variables, i will need to find a working BT dongle, etc... I don't see how it may help.

sorgelig, 2023-04-09:
Partial success! Once i've got it working through button+ i've disabled BT. Now it starts without problems and without button.
Basically i don't need bluetooth but i think this bug should be fixed. I don't know if my iron has broken BT or startup time is too long to make booting process hang.

sorgelig, 2023-04-09:
Some more info (may be useful): if i enable BT and then after several tries able to boot into debug mode, then my phone sees Pinecil BT device, so BLE is working on my iron. Just startup process somewhere hangs (when it doesn't start).

River-Mochi, 2023-04-09:
> i've already tried resetting before. It doesn't help. Not sure if i want to mess with pineSAM. It involves additional variables, i will need to find a working BT dongle, etc... I don't see how it [...]

Actually Pinesam doesn't  you can be up and running in 1 min
you just install that Binary which opens a browser window and you are done.  it runs from browser.

River-Mochi, 2023-04-09:
The difference with PineSam  is it shows  you all of the Information all the Debug, allows all setting changes too.  Even if your  screen  is buggy, it will  let you control Pinecil.  
if you press (+)  to heat up,   then   you can see on pineSAM it heats up. 

the premade binaries,  it's  1  min  to execute it and get it going with   ./start script.   on some windows PC with extra security  you have to do command in Powershell to allow  Powershell to run scripts.   

Powershell runs on MacOS, Linux and Windows but Only windows  locks it down from doing script until  you personally set the permission  which is odd since if I'm opening powershell, I want to run a script.

River-Mochi, 2023-04-09:
> btw, about Joric's pre-2.20 - it's not really pre.. it's pointed to releases/latest which currently is 2.21.

you are right he updated it recently  to  29-03-23 version.

sorgelig, 2023-04-09:
PineSAM won't help. Please read my posts above. Iron really stuck in booting somewhere in BT initialization, so BT doesn't work, buttons don't work (no heat-up on button press). Disabling BT make my iron boot and work normally.

River-Mochi, 2023-04-09:
> 

oic  you don't have any  Bluetooth in your PC or laptop already then .    nice thing about Pinesam is  simply  if you have a black screen it tells us if the rest of the Pinecil is still working.

we have one person who messed up his  screen and is still using Pinecil completely from PineSAM on his phone.

River-Mochi, 2023-04-09:
> PineSAM won't help. Please read my posts above. Iron really stuck in booting somewhere in BT initialization, so BT doesn't work, buttons don't work (no heat-up on button press). Disabling BT make my iron boot and work normally.

Since you are able to partially get  2.21 started by pressing the (+)  button  go to Advanced menu, and turn off BLuetooth. and then see if it boots normally.

sorgelig, 2023-04-09:
> Since you are able to partially get 2.21 started by pressing the (+) button go to Advanced menu, and turn off BLuetooth. and then see if it boots normally.

please read my posts above. It's already done and working.

River-Mochi, 2023-04-09:
do you think it's the firmware or hardware? if you are inside warranty period perhaps they will exchange it.

since it's working for people on bluetooth even for the betas over past several months.

sorgelig, 2023-04-09:
I think it's a bug in firmware. I'm also HW and SW developer, and for me it looks like initialization process doesn't wait for some states.

River-Mochi, 2023-04-09:
I'm on a small screen right now outside with doggies.  

Hopefully Ralim has work around for this .  if it's defective they should honor a replacement  imho since this is first stable firmware that exposes BLE.

and first case  I've seen in with beta bLE over months of people using the  Beta BLE  where screen is black or partially unstable when ble enabled.   could  poor soldering  cold joint  of some part internally caused this unstable behavior. 

would be interesting to see over next few weeks if other people report on this or this is unique.

chiefos, 2023-04-09:
Yeah, have the same issue, reconnecting the cable sometimes helps to start Pinecil , sometimes not. My opinion  - it isn`t caused by BLE, maybe PPS/PD negotiating issue? Hmm..

PSandro, 2023-04-09:
I can confirm this for my pinecil v2. I just upgraded to the latest firmware. Sometimes when I plug in the USB-C cable, a black screen is shown and nothing happens. After replugging the cable, everything works fine usually. Sometimes it needs another replug. 

It also works in the opposite way: I plug in the usb-c cable - everything works fine; I replug the cable and the bug appears again.

River-Mochi, 2023-04-09:
> My opinion - it isn`t caused by BLE, maybe PPS/PD negotiating issue? Hmm..

there was a PPS  change in 2.21.    have you  used  any of the other bluetooth betas before this one?

River-Mochi, 2023-04-09:
I just had a weird issue. 
 Pinepower Portible charger PD20V/65w (newer model from ~Sept 2022).  I plug in cable and screen is black.  I unplug  and plug in again and it's fine the full 20V.  I don't know if it's the charger or the firmware.   
 The charger was plugged in the power strip for a couple hours and unused when I  tried it, don't know if this gan II charger goes into a low energy mode so that it doesn't drain a lot of electricity while sitting unused.
- however,  trying to unplug and plug in again 10 times  I can not repeat the black screen.

chiefos, 2023-04-09:
> 

I tried some betas with ble before. Had the same issue with them. 
I`m using Baseus Blade Laptop Power Bank 100W and Baseus 100W GaN USB Charger Desktop Power Strip with Baseus E-marker chip cables and also with Pine cables from the pine64.com

River-Mochi, 2023-04-09:
Tested my  2nd  V2,  and got a Reboot  on plugging it in and now it's normal. that is unusual, the reboot.  Same 20V PD  Pinepower portable charger.

River-Mochi, 2023-04-09:
also  notice  that when it does work, I can see it  going from 5V increments to 20V  on detailed screen.  Normally I'm used to it being a lot faster jumping to 20V -  so fast I can't normally see all the little incremental steps on the screen from 5v.1v  and all the little decimals up to  19. 9V.

sorgelig, 2023-04-09:
Nothing to do with PPS. I test with USB from PC. So it's pure 5V-only.

River-Mochi, 2023-04-09:
Another test unplugging and plugging into   Pinepower desktop 120W several times,  it reboots a couple of the times  (not solid black like my other test but did reboot  with enough repeats). 
keep volume up to know which time i'm just unplugging versus which time  I'm  getting a reboot. 

https://user-images.githubusercontent.com/97197236/230795420-44fbf637-b480-4bd3-8352-09b14eb86155.mov

River-Mochi, 2023-04-09:
I turned **off**  BLuetooth, and  did same test, and I _still get the Reboots_ after so many unplug+ plug in.  using same charger Pinepower desktop 120w (to be consisitant).

sorgelig, 2023-04-09:
It doesn't look like reboot. It's power cut while plugging. You can see even your power meter turns off. It can be just behavior of your PSU's MCU. MCU in PSU handles power distribution among sockets. When you plug/unplug quickly, the MCU in PSU may get some spikes and unfinished PD/QC negotiation which may lead MCU to reset the circuit and cut the power temporarily.

River-Mochi, 2023-04-09:
> It doesn't look like reboot. It's power cut while plugging. You can see even your power meter turns off. It can be just behavior of your PSU's MCU. MCU in PSU handles power distribution among [...]

 even if I just plug in first time sometimes and leave it pluged it  it reboots.  I only did the Video  a bunch of times to show it more.  I could just do a video showing it one time also. 
Example.  leaving Pinecil unpluged for  10 min.  then plugging it in first time can also reboot.

PSandro, 2023-04-09:
> even if I just plug in first time sometimes and leave it pluged it it reboots. I only did the Video a bunch of times to show it more. I could just do a video showing it one time also. Example. [...]

that sounds like a different issue to me. I'm only experiencing the blackscreen

River-Mochi, 2023-04-09:
> that sounds like a different issue to me. I'm only experiencing the blackscreen

I also get the **solid  black screen** , see my photo above , pressing buttons does not do anything.   this doesn't happen as often on testing. and   Rebooting  seems to fix it.    That photo I took after just leaving it like that for over 3 min. to see if anything would change.  but it just stayed like that.

River-Mochi, 2023-04-09:
I'll try to find my 3rd V2 downstairs and see if there is a difference.

PSandro, 2023-04-09:
> @PSandro
> 
> 1. have you tried testing with bluetooth off?
> 2. have you tried   any of the earlier  beta with bluetooth, do they also cause black screen?
I can confirm that turning bluetooth of in the settings fixes the issue. After that I didn't get any blackscreens and the pinecilv2 functioned normally. 

I haven;t tried any earlier beta versions. Before v2.21 I was running v2.20. I didn't experience any issues there.

Ralim, 2023-04-10:
Hmm, 
By any chance do either of you have the breakout board?
Being able to see what the uart says would be super helpful here. I'm concerned it's a chip hard crash, which can be messy to debug without the uart output. As I have basically zero support on the BLE stack I would not be overly surprised if it behaves badly with "something".

I have not yet been able to reproduce this on any of my power sources or my test irons.

I can try making a build without Bluetooth to try and narrow it down ? Open to other ideas of how to debug without asking you to flash and test all the builds during 2.21.

sorgelig, 2023-04-10:
@Ralim 
I don't think i want to open iron, i have bad experience in opening TS100... But i'm ready to try as many builds as you want. 
If you have a zip file with set of builds, then i'm ready to go through that.

PSandro, 2023-04-10:
> Hmm,\nBy any chance do either of you have the breakout board?

I was also thinking about the UART. I ordered the Breakout Board, yesterday but I could take some time to arrive.

Ralim, 2023-04-10:
No requirement to open the iron, just access to some form of USB C breakout. To get the uart lines.
I'll try and make some firmware to test when I can

sorgelig, 2023-04-10:
> No requirement to open the iron, just access to some form of USB C breakout. To get the uart lines. I'll try and make some firmware to test when I can

Please provide link to pinout and connection parameters. I think i can get breakout board today or tomorrow.

PSandro, 2023-04-10:
> Please provide link to pinout and connection parameters. I think i can get breakout board today or tomorrow.

There's information for the Pinecil Breakout board in the [pine64 wiki](https://wiki.pine64.org/wiki/Pinecil#Pinecil_Breakout_board) including a [board schematic](https://files.pine64.org/doc/Pinecil/Pinecil_Breakout_Board_Schematic_v1.0_20201005.pdf) and [PDB layout](https://files.pine64.org/doc/Pinecil/Pinecil_Breakout_Board_PCB_layout_v1.0_20201005.pdf).

There's also an image on how to connect the breakout board to an UART ttl converter [image]

sorgelig, 2023-04-10:
I didn't mean that full-featured board. What i want to do is to use simple type-c male to pcb breakout board where all type-c signals are output. Then i plan to connect DC5525 and uart pins to that board. As i see from schematics B3 and B2 pins are for UART. So i hope i just need to use those pins without messing with other pins. Not sure if pinecil swaps type-c sides depending on cable sense or they are fixed to specific sides.

sorgelig, 2023-04-10:
do i need to use a special firmware to get UART log? I don't want to setup development environment to compile firmware myself.

Ralim, 2023-04-10:
All firmware builds have the uart enabled.

Just reading off the pins is all you need; no flipping of the pins though they are always the same with cable orientation

sorgelig, 2023-04-10:
ok, i've made USB-C pass-through adapter with B2 pin connected to RX of FT232. Connected by putty with 2000000pbs 8N1. 
I don't see any log in console...

alteiis, 2023-04-10:
Same problem with v2.21:
 - official Pinecil v2 bought at pine64 store
 - flashed with the last blisp

 Ok if Bluetooth off (need to start in debug for first boot to disable Bluetooth)

jonatan-ivanov, 2023-04-10:
Would it make sense to add a warning with the link to this issue to the release notes so people who check it before flashing the new version can decide if they want to go ahead or wait?

[60 more replies]

PR #1730 (2023-07-05): Try side-port of different linker and ram for PinecilV2

* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [] There are no breaking changes

* **What kind of change does this PR introduce?**

A different tact to try and fix #1661 

* **What is the current behavior?**

#1661 
* **What is the new behavior (if this is a feature change)?**
Uses their custom ram allocator  & different linker layout
* **Other information**:
PR #1756 (2023-07-23): BLE crash fixing (Part ?)

* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally (Via @gamelaster )
- [x] There are no breaking changes (AFAIK)

* **What kind of change does this PR introduce?**
Continued work on #1661.

@gamelaster tracked down the issue for the crash was the args to printf being bad causing invalid args to `malloc`. This occurs in the binary blob for BLE, so not trivial to fix or patch.
Luckily by remapping `printf` at link time to redirect to the bfl implementation it lines up.
Now this appears that on the failing unit it will p [... cut]
closed 2023-07-23 by Ralim